lvgl/lvgl #1254: undefined reference to lv_init linker problem

ilijamr, 2019-11-05:
I am having a trouble linking lvgl 6 and esp-idf 4.1. I am calling just lv_init to make a connection, can you tell me where it is located, in which dir/file?

ilijamr, 2019-11-05:
Found it. src/lv_core/lv_obj.c

closed 2019-11-05 by kisvegabor

ilijamr, 2019-11-12:
May I reopen this issue? I am having trouble linking lvgl.

kisvegabor, 2019-11-12:
If your question is related to ESP I suggest asking it either in the
- Forum: https://forum.littlevgl.com/
- ESP32 repo: https://github.com/littlevgl/lv_port_esp32_ili9341

ilijamr, 2019-11-12:
I have a solution finally. Should I write it here?

kisvegabor, 2019-11-13:
Sure!

ilijamr, 2019-11-13:
I followed instructions and put LVGL in my project folder. Then I tried to link the files using GNU MAKE what should be pretty straightforward but it did not work. Finally, I added LVGL to the IDF's components directory. I made directory called lvgl-gui, put everything in there, and added just one component.mk file with the following content:
COMPONENT_SRCDIRS := .
COMPONENT_ADD_INCLUDEDIRS = .
COMPONENT_ADD_INCLUDEDIRS += \
	lvgl/src/lv_core \
	lvgl/src/lv_hal \
	lvgl/src/lv_objx \
	lvgl/src/lv_font \
	lvgl/src/lv_misc \
	lvgl/src/lv_themes \
	lvgl/src/lv_draw	

COMPONENT_SRCDIRS += \
	lvgl/src/lv_core \
	lvgl/src/lv_hal \
	lvgl/src/lv_objx \
	lvgl/src/lv_font \
	lvgl/src/lv_misc \
	lvgl/src/lv_themes \
	lvgl/src/lv_draw

kisvegabor, 2019-11-13:
Thanks.
There were quite a lot of changes in ESP32 repo in the past weeks, maybe it'd worth to try it again with it.

ilijamr, 2019-11-18:
https://github.com/ilijamr/esp-idf/tree/master/examples/ili9488_lvgl_example

C47D, 2019-11-18:
@ilijamr, Is the project of your link working? Last week we have added a way to configure the [lv_port_esp32_ili9341](https://github.com/littlevgl/lv_port_esp32_ili9341) example project using the esp-idf menuconfig method.

@kisvegabor Maybe we can add the `lv_port_esp32_ili9341` project to the official esp-idf repo?

kisvegabor, 2019-11-19:
@C47D 
> Maybe we can add the lv_port_esp32_ili9341 project to the official esp-idf repo?

Does it mean a Pull request to the esp-idf repo?

ilijamr, 2019-11-19:
@C47D yes it's working.

C47D, 2019-11-19:
@kisvegabor Yes, some sort of hello world project and another project with more features enabled.

@ilijamr Great, my kit have the ili9341 controller, but i will check your source code.

kisvegabor, 2019-11-21:
> @kisvegabor Yes, some sort of hello world project and another project with more features enabled.

Great, I'm absolutely familiar with their workflow and requirements. Could you manage this?

C47D, 2019-11-21:
@kisvegabor I will ask on the esp-idf repo what are the requirements to add example projects and add it if possible. I will let you know when i got their reply.

kisvegabor, 2019-11-21:
Great, thanks!