lvgl/lvgl #1256: lv_txt_get_next_line handles \n incorrectly

kisvegabor, 2019-11-08:
**Describe the bug**

If you move the cursor left/right in a text area around a `\n` it works incorrectly in `dev-6.1`.
I've found that it's because `lv_txt_get_next_line` returns the next line position without the `\n` (`\n` should belong to the current line, not the next one).

Probably it's related to the new `lv_txt_get_next_line` introduced by @BrianPugh.

**To Reproduce**

I've used `lv_test_kb_1` like this (in `dev-6.1`)
[image]

**Expected behavior**
It worked like this in `v6.0`

[image]

@BrianPugh Could you take a look at it?

BrianPugh, 2019-11-11:
I'll take a look this week

BrianPugh, 2019-11-11:
I have a fix when navigating by cursor, but clicking is wrong when you click to the right of the last line; so i need to fix that edge case in `lv_label_get_letter_on`.

BrianPugh, 2019-11-11:
Should mark this as closed. Might want to also merge #1259 into either dev-7.0 or the bidi branch

embeddedt, 2019-11-11:
> Might want to also merge #1259 into either dev-7.0

`master`, `dev-6.1`, and `dev-7.0` are periodically merged (from left to right). So this fix will get applied to `dev-7.0` the next time the merge happens.

kisvegabor, 2019-11-11:
@BrianPugh I've just merged it to the [PR for bidi](https://github.com/littlevgl/lvgl/pull/1253). If @amirgon says OK then I'll merge `bidi` to `dev-6.1` and then `dev-6.1` to `dev-7.0`.

---

A partially related question to @BrianPugh: in one of my projects I need to display a long (40 chars) string and I'd like to disable `LV_TXT_LINE_BREAK_LONG_...`. The current behavior is this with
```c

 /* If a character is at least this long, will break wherever "prettiest" */
 #define LV_TXT_LINE_BREAK_LONG_LEN          12

 /* Minimum number of characters of a word to put on a line before a break */
 #define LV_TXT_LINE_BREAK_LONG_PRE_MIN_LEN  3

 /* Minimum number of characters of a word to put on a line after a break */
 #define LV_TXT_LINE_BREAK_LONG_POST_MIN_LEN 3
```

[image]

I just want to wrap the line when it reached the max-width. Can you recommend a configuration?

BrianPugh, 2019-11-11:
Setting `LV_TXT_LINE_BREAK_LONG_LEN` arbitrarily large will make the code using `LV_TXT_LINE_BREAK_LONG_PRE_MIN_LEN` and `LV_TXT_LINE_BREAK_LONG_POST_MIN_LEN` effectively disabled.

A cleaner solution would be to disable these features by changing these lines:

https://github.com/littlevgl/lvgl/blob/dev-6.1/src/lv_misc/lv_txt.c#L241

to something like

```
#if LV_TXT_LINE_BREAK_LONG_LEN > 0
    if(word_len < LV_TXT_LINE_BREAK_LONG_LEN)
#endif
    {
        if(word_w_ptr != NULL) *word_w_ptr = 0;
        return 0;
    }
```

So that if `LV_TXT_LINE_BREAK_LONG_LEN` is <=0 , the "pretty break for long words" is unreachable. To be more explicit, the remaining portions of that function could also be macro'd out, but wouldn't haven't any actual impact.

kisvegabor, 2019-11-11:
@BrianPugh 
Thank you for the suggestion. Actually I've already tried it but got this result:
```c
LV_TXT_LINE_BREAK_LONG_LEN          120
```
[image]

BrianPugh, 2019-11-11:
that is certainly undesired behavior. I'll try and have a PR in tonight!

kisvegabor, 2019-11-13:
#1261 Is merged do I close this issue.

closed 2019-11-13 by kisvegabor