lvgl/lvgl #1257: dev-6.1: cpicker build fails

amirgon, 2019-11-09:
> # Important: issues that don't use this template will be ignored/closed.

Build fails when `LV_COLOR_16_SWAP` set to 1 and `LV_COLOR_DEPTH` set to 16

The problem is that cpicker is trying to access `color.ch.green` explicitly.
It's incorrect in this case (since `lv_color_t` struct is different).

```
|| ../../lib/lv_bindings/lvgl/src/lv_objx/lv_cpicker.c: In function 'lv_cpicker_set_color':
/home/amirgon/esp/projects/lv_mpy/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_objx/lv_cpicker.c|278 col 82| error: 'struct <anonymous>' has no member named 'green'
||      return lv_cpicker_set_hsv(cpicker, lv_color_rgb_to_hsv(color.ch.red, color.ch.green, color.ch.blue));
||                                                                                   ^
/home/amirgon/esp/projects/lv_mpy/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_objx/lv_cpicker.c|279 col 1| error: control reaches end of non-void function [-Werror=return-type]
||  }
||  ^
|| cc1: all warnings being treated as errors
py/mkrules.mk|47| recipe for target 'build-GENERIC_SPIRAM/lib/lv_bindings/lvgl/src/lv_objx/lv_cpicker.o' failed
```

**To Reproduce**

Set `LV_COLOR_16_SWAP`  to 1 and `LV_COLOR_DEPTH` to 16 and build lvgl

**Additional context**

Needed for ESP32 build with pure micropython ILI9341 display driver

kisvegabor, 2019-11-10:
Fixed it 6f762bb.
Please verify if it works now.

amirgon, 2019-11-10:
> Fixed it [6f762bb](https://github.com/littlevgl/lvgl/commit/6f762bb7c776894ca49f8143cf28dade4099d988).

@kisvegabor Does it make sense to add an `#if` on every line of code that access r/g/b?
Why not create accessor macros instead? (`GET_R(color)`, `GET_G(color)`, `GET_B(color)`)   
This way the ugly `#if`s and lo/hi shifts arithmetic would be contained in the accessors implementation instead of duplicated and spread out for each r/g/b access.  
Also it would make the code forward compatible if you ever change `lv_color_t` struct.

kisvegabor, 2019-11-11:
> Does it make sense to add an #if on every line of code that access r/g/b?

It really doesn't make sense. I always forget the case of `LV_COLOR_16_SWAP`.

@embeddedt 
It might affect the [fixing of warnings](#1213) in `color.h`, right?
Probably it would be worth to add the macros or inline functions suggested by @amirgon before fixing the warnings.

embeddedt, 2019-11-11:
> Probably it would be worth to add the macros or inline functions suggested by @amirgon before fixing the warnings. If you agree I'll add them.

I agree. That should simplify things.

kisvegabor, 2019-11-12:
I've ad the macros to `dev-6.1`. Now `LV_COLOR_16_SWAP` used in `color.h` in some special cases.

amirgon, 2019-11-12:
> I've ad the macros to `dev-6.1`. Now `LV_COLOR_16_SWAP` is used only in `color.h` in some special cases.

Looks good! I verified this on ESP32.

Since `LV_COLOR_GET/SET` are macros, they are not available on Micropython.
I'm not sure how important that is, we can always add some static inline function if we want them on Micropython.

kisvegabor, 2019-11-13:
> Looks good! I verified this on ESP32.

Great!

> Since LV_COLOR_GET/SET are macros, they are not available on Micropython.

IMO it's not a problem if these low-level things are missing because they are shouldn't be used on the user directly. What do you think?

amirgon, 2019-11-13:
> IMO it's not a problem if these low-level things are missing because they are shouldn't be used on the user directly. What do you think?

That's fine with me.  
And if we ever decide otherwise, we can always add inline functions.

commit 6f762bb7 (2019-11-10): fix lv_cpicker_set_color with 16 bit color depth and byte swap
Fixes #1257
closed 2019-11-13 by amirgon