lvgl/lvgl #1262: Add a "rotation" feature to the line meter

Maldus512, 2019-11-12:
A few weeks ago I started designing a GUI where I wanted to have a couple of line meters displaying parameters like speed, pressure and/or temperature. To better organize the information I would like to dispose said meters as 90 degrees arcs in the four corners of the rectangular display.

Unfortunately the line meter object does not expose any way to achieve such result. I submitted this request to the forum (https://forum.littlevgl.com/t/how-to-display-a-rotated-line-meter/1064/2) and I've been told to bring it here.

As I suspected the feature is extremely easy to implement. All I had to do was add a `rotation` field to the `lv_lmeter_ext_t`, set it with an ad-hoc function and then take it into account when drawing the lines:

```c
diff --git a/src/lv_objx/lv_lmeter.c b/src/lv_objx/lv_lmeter.c
index 78f7a8e6..b7721100 100644
--- a/src/lv_objx/lv_lmeter.c
+++ b/src/lv_objx/lv_lmeter.c
@@ -168,6 +168,21 @@ void lv_lmeter_set_scale(lv_obj_t * lmeter, uint16_t angle, uint8_t line_cnt)
     lv_obj_invalidate(lmeter);
 }
 
+/**
+ * Set the rotation settings of a line meter
+ * @param lmeter pointer to a line meter object
+ * @param angle angle of rotation (relative to the northen axis)
+ */
+void lv_lmeter_set_rotation(lv_obj_t * lmeter, uint16_t angle)
+{
+    lv_lmeter_ext_t * ext = lv_obj_get_ext_attr(lmeter);
+    if(ext->rotation == angle) return;
+
+    ext->rotation = angle;
+
+    lv_obj_invalidate(lmeter);
+}
+
 /*=====================
  * Getter functions
  *====================*/
@@ -268,7 +283,7 @@ static bool lv_lmeter_design(lv_obj_t * lmeter, const lv_area_t * mask, lv_desig
 
         lv_coord_t x_ofs  = lv_obj_get_width(lmeter) / 2 + lmeter->coords.x1;
         lv_coord_t y_ofs  = lv_obj_get_height(lmeter) / 2 + lmeter->coords.y1;
-        int16_t angle_ofs = 90 + (360 - ext->scale_angle) / 2;
+        int16_t angle_ofs = ext->rotation + 90 + (360 - ext->scale_angle) / 2;
         int16_t level =
             (int32_t)((int32_t)(ext->cur_value - ext->min_value) * ext->line_cnt) / (ext->max_value - ext->min_value);
         uint8_t i;
```

This way I can have two line meters side by side and displaying different values, like in this screenshot:
[image]

It's a very simple feature, and by adding a new function (`lv_lmeter_set_rotation`) it does not even break the API. Would it be possible to add it?

kisvegabor, 2019-11-12:
Hi,

Thank you for this feature! Can you send it in a [Pull Request](https://blog.littlevgl.com/2018-12-06/contributing#how-can-i-send-fixes-and-improvements)?

Maldus512, 2019-11-12:
Certainly!

kisvegabor, 2019-11-13:
#1264 is merged so I close this issue.

Thank you for your contribution!

closed 2019-11-13 by kisvegabor