lvgl/lvgl #1266: lv_obj_get_screen endless loop

zeile218, 2019-11-13:
The following code triggers an endless loop in lv_obj_get_screen:

	lv_obj_t *scr_page = lv_page_create(lv_scr_act(),NULL);

	lv_obj_t * cont_systeme[8];
	lv_obj_t * label_systeme[8];

	for(int i=0; i<8; i++){
		cont_systeme[i] = lv_cont_create(scr_page,NULL);
		label_systeme[i] = lv_label_create(cont_systeme[i],NULL);
	}

	lv_obj_clean(scr_page);

	cont_systeme[0] = lv_cont_create(scr_page,NULL); //triggers the bug

The problem is that the child and parent in lv_page_signal have the same address. lv_obj_get_ext_attr returns the address of the parent and lv_obj_set_parent now has the address of the parent for tmp and ext->scrl. This triggers the loop in lv_obj_get_screen because par is always != NULL. 

I'm using the lvgl master from github.

kisvegabor, 2019-11-13:
Hi,

Thanks for the clear and simple code example! :+1: 

I've tried it but found that the problem is something different. To clean a page you should use `lv_page_clean(scr_page);` It will clean the children only from the scrollable part of the page. 
`lv_obj_clean()` deletes the scrollable part of the page too but its an essential part of the page and if it's not there LittlevGL crashes.

zeile218, 2019-11-13:
Thanks for the fast feedback! It works now with `lv_page_clean(scr_page);` Thanks!

closed 2019-11-13 by zeile218