lvgl/lvgl #1270: lv_cb_get_style of LV_CB_STYLE_BG returns NULL

CURTLab, 2019-11-14:
I found a small bug where get style of the checkbox widget returns NULL if the type is LV_CB_STYLE_BG. It seems to me since checkbox is inherited from the button widget that the background style is not correctly saved within the private data of the widget. (see lv_cb.c at line 209 to 215)

I'm using the dev-6.1 branch.

embeddedt, 2019-11-14:
Nice catch!

https://github.com/littlevgl/lvgl/blob/bbf2c8247ccca1fabee135feab52c9c7f58d76f4/src/lv_objx/lv_cb.c#L209-L216

I think the simplest solution would be to just retrieve any one of the ancestor button's styles.

```c
case LV_CB_STYLE_BG: style = lv_btn_get_style(cb, LV_BTN_STYLE_REL); break;
```

This should work because whenever the background style is set, [all of the button styles are changed](https://github.com/littlevgl/lvgl/blob/bbf2c8247ccca1fabee135feab52c9c7f58d76f4/src/lv_objx/lv_cb.c#L165-L169).

embeddedt, 2019-11-15:
@CURTLab Could you try adding the line I suggested in `lv_cb.c` and see if it works for you? If it does I will apply the patch to `master` (provided that @kisvegabor approves of it).

kisvegabor, 2019-11-15:
Really nice catch!
I've tested @embeddedt's proposed solution and it worked well.

kisvegabor, 2019-11-19:
I've just fixed it on both `master` and `dev-6.1` as @embeddedt suggested.

closed 2019-11-19 by kisvegabor