lvgl/lvgl #1272: Pointer events don't go through masked parts of objects

embeddedt, 2019-11-15:
I was playing with `dev-7.0` today and I noticed that the color picker now seems to work with a transparent center, which was great to see!

The problem is that I cannot click through the center of the color picker. I'm sure that this is because input events are ignoring whatever is masking out the color picker's center.

I'd like to add this functionality, but I'm struggling to figure out where the color picker masking is being done. Does anyone know?

You can see this behavior on a 480x272 screen with the following code:

```c
   lv_test_theme_1(lv_theme_get_current());
   lv_coord_t pickerSize = 200;

    static lv_style_t styleMain;
    lv_style_copy(&styleMain, &lv_style_plain);
    styleMain.line.width = 30;
    styleMain.body.radius = 5;
	styleMain.body.main_color = styleMain.body.grad_color = LV_COLOR_WHITE;

    static lv_style_t styleIndicator;
    lv_style_copy(&styleIndicator, &lv_style_pretty);
    styleIndicator.body.border.color = LV_COLOR_WHITE;
    styleIndicator.body.opa = LV_OPA_COVER;
    styleIndicator.body.border.opa = LV_OPA_COVER;

    lv_obj_t * scr = lv_scr_act();

    lv_obj_t * colorPicker = lv_cpicker_create(scr, NULL);

    lv_obj_set_size(colorPicker, pickerSize, pickerSize);
    lv_cpicker_set_type(colorPicker, LV_CPICKER_TYPE_DISC);
    lv_obj_align(colorPicker, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_cpicker_set_style(colorPicker, LV_CPICKER_STYLE_MAIN, &styleMain);
    lv_cpicker_set_style(colorPicker, LV_CPICKER_STYLE_INDICATOR, &styleIndicator);
    lv_cpicker_set_indic_colored(colorPicker, true);
```

kisvegabor, 2019-11-15:
> The problem is that I cannot click through the center of the color picker. I'm sure that this is because input events are ignoring whatever is masking out the color picker's center.

You need to long-press the center. As I'm thinking about it the long press is really not required in case of disc type.

> I'd like to add this functionality, but I'm struggling to figure out where the color picker masking is being done. Does anyone know?

In `v7.0` I've found that drawing lines are faster and result in better quality than drawing triangles.
The lines are simply drawn between the outer and the inner radius so the "hole" i nthe middle is produced naturally.

embeddedt, 2019-11-15:
> You need to long-press the center. As I'm thinking about it the long press is really not required in case of disc type.

I think we could enable long-pressing the center if the preview circle is also drawn. Otherwise I think it should be disabled, because it's unintuitive to press another object to change something on the color picker.

Could we add a new signal, then, that would allow objects to become partially clickable?

> The lines are simply drawn between the outer and the inner radius so the "hole" in the middle is produced naturally.

Okay, that makes sense. I was wondering what had happened to the triangle drawing. :slightly_smiling_face: 

I still think we need a way to tell the input system that there is a hole in the middle of an object.

Perhaps, when the preview circle is disabled, we could add a mask to the center of the color picker (even though it wouldn't actually be masking anything) and then have the input system check whether the point is within the object's mask(s) as well.

kisvegabor, 2019-11-15:
> I think we could enable long-pressing the center if the preview circle is also drawn. Otherwise I think it should be disabled, because it's unintuitive to press another object to change something on the color picker.

I mean, in my opinion the long press is not required here at all because there is no action for simple click either. So why not use the simple click for the only user input we have with cpicker?

> I still think we need a way to tell the input system that there is a hole in the middle of an object.

Actually the input system does not process the masking (i.e. the draw masks in v7.0). E.g. you can still click a button out of the rounded corners. 

Or do you mean to process/skip an area manually in the signal function of the cpicker?

Anyway, I agree, it's really unintuitive to make the user to click the "hole".

embeddedt, 2019-11-15:
Basically, I think we need to find some way (either manually in the object's signal function or automatically in the input system) of allowing users to click through part of an object.

> I mean, in my opinion the long press is not required here at all because there is no action for simple click either. So why not use the simple click for the only user input we have with cpicker?

I think we should leave it as a long-press because otherwise it will increase the frequency of accidentally switching modes (e.g. if the user taps a part of the color picker away from the knob to move the knob there).

kisvegabor, 2019-11-16:
So, if I understand correctly:
- if no preview, ignore user inputs on the hole and use long-press of the knob to change mode
- if there is preview then use long-press on the preview to change mode. (an maybe still on the knob too)

Is this your idea?

embeddedt, 2019-11-16:
@kisvegabor Yes; that's my thinking.

kisvegabor, 2019-11-18:
I've implemented it. Could you test it?

embeddedt, 2019-11-18:
(For some reason your comment got duplicated so I deleted the duplicate one.)

I will test it and report back in a few minutes.

embeddedt, 2019-11-18:
Now it doesn't change the color on a long press in the transparent section, however, I still can't click on objects behind the color picker.

kisvegabor, 2019-11-19:
> I still can't click on objects behind the color picker.

My goal was only to not click the object behind the cpicker. It'd be quite difficult and hacky to extend the "search the objects on a clicked point" algorithm to the objects' level.

IMO the current approach is simple and avoids confusion. 
I'm not sure, even if we could easily find the clicked object behind the cpicker, it would be the right way. I think it would be also confusing.

embeddedt, 2019-11-19:
I agree. Let's leave it as it is for now.

The general confusion of long pressing the transparent area has been resolved, so this issue can be closed.

PR #1318 (2019-12-22): Implement custom, opt-in hit-testing handlers for objects
Fixes #1272 in a more generic manner. Advanced hit-testing must be explicitly enabled by the user for the appropriate objects, so there is 0% performance overhead by default.

Test code:

```c
    const lv_coord_t pickerSize = 200;

	/* Set the style of the color ring */
    static lv_style_t styleMain;
    lv_style_copy(&styleMain, &lv_style_plain);
    styleMain.line.width = 30;
	/* Make the background white */
    styleMain.body.main_color = styleMain.body.grad_color = LV_COLOR_WHITE;

	/* Set the style of the knob */
    static lv_s [... cut]
closed 2019-11-19 by embeddedt