lvgl/lvgl #1274: Unicode Bidirectional Open Issues

amirgon, 2019-11-15:
This issue documents and tracks the progress of open issues related to Unicode Bidirectional support on LittlevGL, as [discussed](https://github.com/littlevgl/lvgl/issues/899#issuecomment-554234385) on #899 

The following issues are visible when `LV_USE_BIDI` and `LV_LABEL_TEXT_SEL` are defined as `1` on `lv_conf.h`, and a Label / Text Area contains RTL (Right-To-Left) text (for example Hebrew or Arabic characters). Some issues may only visible when the object's base direction is `RTL`. Some issues may only be visible when text is multi-line.

- [ ] Can't select the first character
- [ ] Text suddenly scrolls all the way down when selecting backwards towards the beginning of the line (rightmost)
- [ ] Cursor and selection area edge are, at some cases, away from each other
- [ ] Correctly support different cursor types
- [ ] When RTL in RTL is selected (not LTR in RTL!) sometimes a reversed selection happens.
This "reversed" behavior should happen only when selecting a mix of RTL and LTR runs. 
- [ ] slider/bar/sw should have RTL directionality and behavior when its base dir is RTL. For example, a slider should be "mirrored" such that low values would be on the right and high values on the left.

More details about each of these issues are available on [this](https://github.com/littlevgl/lvgl/issues/899#issuecomment-552622018) and [this](https://github.com/littlevgl/lvgl/issues/899#issuecomment-553100665) posts, including animated GIFs that illustrate the problems.

kisvegabor, 2019-11-15:
Thank you for the summary!

embeddedt, 2020-03-08:
@amirgon I'm working on making the slider, bar, and switch respect their base direction. Do I need to mirror them when they're vertical, or only if they're horizontal?

amirgon, 2020-03-08:
> Do I need to mirror them when they're vertical, or only if they're horizontal?

@embeddedt only when they are horizontal.  
Thanks.

embeddedt, 2020-03-14:
> slider/bar/sw should have RTL directionality and behavior when its base dir is RTL

This should work now with the changes in https://github.com/littlevgl/lvgl/commit/d399dec0c76ffdb874ae60a073da290747d398eb.

kisvegabor, 2020-03-16:
Thank you @embeddedt.

I've just tested it and found that the changes were not applied to `lv_swich`. Could you take a look?

embeddedt, 2020-03-16:
Thanks for the feedback. I thought the switch was using `lv_slider` under the hood, but it appears that that is no longer the case (or never was :wink:). I made it work in https://github.com/littlevgl/lvgl/commit/ab10bdaa033a0166bb863055b4f197e75d2ac752.

kisvegabor, 2020-03-17:
Perfect, thank you!

The switch was based on the Slider in v6 but I found too many differences (although they seem quite the same) and derived the Switch directly from Bar in v7.

kisvegabor, 2020-07-08:
I close this issue in lack of activity. 

We can reopen it any time if this topic becomes active again.

closed 2020-07-08 by kisvegabor