lvgl/lvgl #1276: Crash when delete a task in its callback.

dulm-today, 2019-11-18:
**Describe the bug**

In function **lv_task_handler**,  **task_interrupter** is set after **lv_task_exec**. If I delete the 
task in its callback, then **task_interrupter** will be a wild pointer, and it will be accessed in next loop.

```c
else if(task_interrupter) {
    if(((lv_task_t *)LV_GC_ROOT(_lv_task_act))->prio > task_interrupter->prio) {     /* read wild pinter here! */
        if(lv_task_exec(LV_GC_ROOT(_lv_task_act))) {
            task_interrupter =
                LV_GC_ROOT(_lv_task_act); /*Check all tasks again from the highest priority */
            end_flag = false;
            break;
        }
    }
}
```

kisvegabor, 2019-11-18:
I've just fixed it. 
Can you verify if it's work now?

dulm-today, 2019-11-18:
Yes, it work now! Thank you!

closed 2019-11-18 by dulm-today