lvgl/lvgl #1288: Memory leak during object creation

MaJerle, 2019-11-26:
> # Important: issues that don't use this template will be ignored/closed.

If extended parameters for object fail to allocate memory area, there is a memory leak:

https://github.com/littlevgl/lvgl/blob/ec18ea42fed7a4a3337c2b932d7871d8349aacd1/src/lv_objx/lv_cpicker.c#L93-L109

If extended attributes memory allocation fail, clean initial object before returning NULL. See line 107.

embeddedt, 2019-11-26:
Thanks for the bug report. This is definitely an issue. It probably applies to most objects since they all use a similar creation pattern.

MaJerle, 2019-11-27:
I was always wondering why Lv_obj_create doesnt also allocate extended memory. New param would be necessary indeed for that. Then you check against object being null only.

kisvegabor, 2019-11-28:
That's true, and it can be fixed by calling `lv_obj_del` is the allocation of `ext` fails. Although I think you should never out of memory so it's better to stop at `LV_ASSERT_MEM` to see where the problem has happened. 

> I was always wondering why Lv_obj_create doesnt also allocate extended memory.

For example if you create a button it calls `lv_cont_create` which calls `lv_obj_create`. When `lv_obj_create` is ready `lv_cont` adds it's own `ext` and don't know anything about it will be button later. When `cont` is ready `btn` reallocates `ext` to add the button specific parts too.

To make `lv_obj_create` allocate `ext` we needed to pass the `ext size` through the whole `btn->cont->obj` chain. So all `create` functions should have an  `ext_size` parameter too which would be quite ugly.

kisvegabor, 2019-12-03:
I've fixed this issue in c7374948.

closed 2019-12-03 by kisvegabor