lvgl/lvgl #1291: Border part type not behaving as expected

Guillaumebeuzeboc, 2019-12-02:
**Describe the bug**
The border part of the cell type doesn't seem to act correctly.
The types LV_BORDER_FULL, LV_BORDER_TOP and LV_BORDER_LEFT seems to work as expected but not the types LV_BORDER_BOTTOM, LV_BORDER_RIGHT, LV_BORDER_INTERNAL. When I use these types no borders appear at all on my table.

**To Reproduce**
I took the example from the documentation and changed the fields for border.part
In this example I used LV_BORDER_INTERNAL and we see that no border appears at all.
I am using a stm32F429I-DISC1 and I have the same behavior with the simulator.

```
  static lv_style_t style_cell1;
  lv_style_copy(&style_cell1, &lv_style_plain);
  style_cell1.body.border.width = 1;
  style_cell1.body.border.color = LV_COLOR_BLACK;
  style_cell1.body.border.part = LV_BORDER_INTERNAL;

  /*Crealte a header cell style*/
  static lv_style_t style_cell2;
  lv_style_copy(&style_cell2, &lv_style_plain);
  style_cell2.body.border.width = 1;
  style_cell2.body.border.color = LV_COLOR_BLACK;
  style_cell2.body.main_color = LV_COLOR_SILVER;
  style_cell2.body.grad_color = LV_COLOR_SILVER;
  style_cell2.body.border.part = LV_BORDER_INTERNAL;

  lv_obj_t *table = lv_table_create(lv_scr_act(), NULL);
  lv_table_set_style(table, LV_TABLE_STYLE_CELL1, &style_cell1);
  lv_table_set_style(table, LV_TABLE_STYLE_CELL2, &style_cell2);
  lv_table_set_style(table, LV_TABLE_STYLE_BG, &lv_style_transp_tight);
  lv_table_set_col_cnt(table, 2);
  lv_table_set_row_cnt(table, 4);
  lv_obj_align(table, NULL, LV_ALIGN_CENTER, 0, 0);

  /*Make the cells of the first row center aligned */
  lv_table_set_cell_align(table, 0, 0, LV_LABEL_ALIGN_CENTER);
  lv_table_set_cell_align(table, 0, 1, LV_LABEL_ALIGN_CENTER);

  /*Make the cells of the first row TYPE = 2 (use `style_cell2`) */
  lv_table_set_cell_type(table, 0, 0, 2);
  lv_table_set_cell_type(table, 0, 1, 2);

  /*Fill the first column*/
  lv_table_set_cell_value(table, 0, 0, "Name");
  lv_table_set_cell_value(table, 1, 0, "Apple");
  lv_table_set_cell_value(table, 2, 0, "Banana");
  lv_table_set_cell_value(table, 3, 0, "Citron");

  /*Fill the second column*/
  lv_table_set_cell_value(table, 0, 1, "Price");
  lv_table_set_cell_value(table, 1, 1, "$7");
  lv_table_set_cell_value(table, 2, 1, "$4");
  lv_table_set_cell_value(table, 3, 1, "$6");
```
[image]

**Expected behavior**

I was expecting to see internals borders of the table for  LV_BORDER_INTERNAL, bottom borders for LV_BORDER_BOTTOM, etc.

**Additional context**
I am working under ubuntu 18.04
littlevGL V6.1

kisvegabor, 2019-12-02:
I'l ltake a look today or tomorrow.

kisvegabor, 2019-12-03:
I've examined it and found that the issue is that there is 1 px overlap in the neighbor cells which covers the right and bottom borders of the neighbor cells.

If I fix it the internal borders would look as they have 2 px border (1 px form a cell and 1 px from the neighbor)
[image]

I'm still thinking about an appropriate approach.

Guillaumebeuzeboc, 2019-12-25:
This should be labeled as a bug no? (So it doesn't get closed)

kisvegabor, 2019-12-27:
@embeddedt just added the labels. 
I'm [reworking the style system](https://github.com/littlevgl/lvgl/issues/1311) of lvgl, and probably I'll fix it during the rework.

Unfortunately, it will take some time but probably it would be difficult to merge the fix to the new style system.

kisvegabor, 2020-02-20:
It should be solved in `dev-7.0`.
A lot of new code was merged, please see [this comment](https://github.com/littlevgl/lvgl/issues/1243#issuecomment-586579395) for the details.

If it's still not working correctly please comment here and reopen the issue.

closed 2020-02-20 by kisvegabor