lvgl/lvgl #1292: Collection of minor issues

xennex22, 2019-12-02:
I'd rather someone else did these changes than making a PR.

lvgl\src\lv_draw\lv_draw_blend.c(54)
static inline lv_color_t color_blend_true_color_additive(lv_color_t bg, lv_color_t fg, lv_opa_t opa);
should be
static inline lv_color_t color_blend_true_color_additive(lv_color_t fg, lv_color_t bg, lv_opa_t opa);

lvgl\src\lv_font\lv_font_fmt_txt.c(83)
if(!gid) return false;
should be
if(!gid) return NULL;

lvgl\src\lv_objx\lv_line.c(249)
lv_coord_t r_corr = style->line.width & 1 ? 0 : 1;
some compiles prefer this (none will complain)
lv_coord_t r_corr = (style->line.width & 1) ? 0 : 1;

lvgl\src\lv_themes\lv_theme_material.c(684)
indic.body.padding.inner = LV_DPI / 20;
indic.body.padding.inner = LV_DPI / 20;

lvgl\src\lv_themes\lv_theme_nemo.c(622)
list_pr.body.opa        = LV_OPA_TRANSP;
list_pr.body.opa        = LV_OPA_COVER;

lvgl\src\lv_objx\lv_ta.c(1710)
style_res->body.opa            = LV_OPA_COVER;
style_res->body.opa            = LV_OPA_COVER;

lvgl\src\lv_objx\lv_sw.h(150)
lv_sw_ext_t * ext = lv_obj_get_ext_attr(sw);
should be
lv_sw_ext_t * ext = (lv_sw_ext_t *)lv_obj_get_ext_attr(sw);

Finally there are a lot of functions which have a dead code return to prevent a warning.
eg lvgl\src\lv_objx\lv_roller.c(320)
const lv_style_t * lv_roller_get_style(const lv_obj_t * roller, lv_roller_style_t type)
{
    LV_ASSERT_OBJ(roller, LV_OBJX_NAME);

    switch(type) {
        case LV_ROLLER_STYLE_BG: return lv_obj_get_style(roller);
        case LV_ROLLER_STYLE_SEL: return lv_ddlist_get_style(roller, LV_DDLIST_STYLE_SEL);
        default: return NULL;
    }

    /*To avoid warning*/
    return NULL;
}

Where some compilers will give a warning because they recognize the dead code.  I'm not sure of the best approach - the only one I can think of is to use the pre-processor.

kisvegabor, 2019-12-02:
Thank you, I'll examine them!

kisvegabor, 2019-12-03:
I've fixed all the listed issues except the one:

> lvgl\src\lv_objx\lv_sw.h(150)
lv_sw_ext_t * ext = lv_obj_get_ext_attr(sw);
should be
lv_sw_ext_t * ext = (lv_sw_ext_t *)lv_obj_get_ext_attr(sw);

Why is it a problem? It's used without casting everywhere. Because it's in a header file?

embeddedt, 2019-12-03:
> Why is it a problem? It's used without casting everywhere. Because it's in a header file?

Not sure what the original reason is, but anyone using a C++ file to include the LittlevGL headers will get a warning about that cast, because C++ forbids implicit casts from `void *` to another type.

kisvegabor, 2019-12-03:
> Not sure what the original reason is, but anyone using a C++ file to include the LittlevGL headers will get a warning about that cast, because C++ forbids implicit casts from void * to another type.

I see. I've fixed it too.

xennex22, 2019-12-05:
Thanks for looking at this.

kisvegabor, 2019-12-05:
> Thanks for looking at this.

I also thank you for the report.

I close this issue. 
Feel free to reopen if you found similar issues.

closed 2019-12-05 by kisvegabor