lvgl/lvgl #1302: use unitialized variable

ImmortalBuka, 2019-12-05:
During lv_init, then try to do a lv_img_cache_set_size, variable _lv_img_cache_array used first time.
But it is unitialized.

    if(LV_GC_ROOT(_lv_img_cache_array) != NULL) {
        /*Clean the cache before free it*/
        lv_img_cache_invalidate_src(NULL);
        lv_mem_free(LV_GC_ROOT(_lv_img_cache_array)); <------ HERE
    }
[image]

kisvegabor, 2019-12-05:
The `if(LV_GC_ROOT(_lv_img_cache_array) != NULL) ` should make sure to enter there if `_lv_img_cache_array` is initialized.

xennex22, 2019-12-06:
How does _lv_img_cache_array get set to NULL?  It's not static and I see it in the .data section of my code, so it seems uninitialized.  It might depend on the custom memory settings.

kisvegabor, 2019-12-06:
It's a little bit tricky. LittlevGL works with the Garbage collector of Micropython, therefore, there are some variables (roots of the garbage collector) declared in a special way. Take a look at `lv_misc/lv_gc.h`. `_lv_img_cache_array` is declared in `LV_GC_ROOTS` macro which is used in `lv_gc.c` to declare the variables as global.

xennex22, 2019-12-16:
You are correct.  _lv_img_cache_array is a global variable and therefore initialized to zero.  I think the issue can be closed.

embeddedt, 2019-12-16:
Let's wait a few more days and if we don't hear from @ImmortalBuka then I'll close it.

ImmortalBuka, 2019-12-16:
It was no so clear for me to catch it. Thanks for help!

closed 2019-12-16 by ImmortalBuka