lvgl/lvgl #1303: lv_obj_del() makes "Segmentation fault (core dumped)"

magailee, 2019-12-06:
**Describe the bug**

    app_root_scr = lv_scr_act(); // get the currently active screen
    app_init(app_root_scr); // init app lvgl-objs
    ......
    // my env
    //
    // root@localhost:~/works/gui# uname -a
    // Linux localhost.localdomain 4.4.167 #1 SMP Fri Sep 20 13:29:17 CST 2019 aarch64 GNU/Linux
    //
    // this makes "Segmentation fault (core dumped)"  in ubuntu
    lv_obj_del(app_root_scr); // <== crash here when i use Tabview obj in app
    exit(0);

**Additional context**
   
I used gdb to trace what's going on, and I figured out it crashed at lv_mem_free() with lv_del_obj().

I made some changes in lv_mem_free() in lv_mem.c, and it worked normally.
I have no idea what the real problem is, and what's the best solution.

Here the codes I modified:

    void lv_mem_free(const void * data)
    {
        if(data == &zero_mem) return;
        if(data == NULL) return;

    #if LV_MEM_ADD_JUNK
        memset((void *)data, 0xbb, lv_mem_get_size(data));
    #endif

    #if LV_ENABLE_GC == 0
        /*e points to the header*/
        lv_mem_ent_t * e = (lv_mem_ent_t *)((uint8_t *)data - sizeof(lv_mem_header_t));
        
    ///////////////////////////////////////////////////////////////////////////////////
    /////// printf("mem-free %d: %d %d\n", e, e->header.s.used, e->header.s.d_size);
    /////// .s.used and .s.d_size must not be zero 
    if (e->header.header) 
    //////////////////////////////////////////////////////////////////////////////////        
        
        e->header.s.used = 0;
    #endif

    #if LV_MEM_CUSTOM == 0
    #if LV_MEM_AUTO_DEFRAG
        /* Make a simple defrag.
        * Join the following free entries after this*/
    /////////////////////////////////////////////
    if (e->header.header) { 
    ////////// printf("--defrag %p\n", e);
    ////////////////////////////////////////////
        lv_mem_ent_t * e_next;
        e_next = ent_get_next(e);
        while(e_next != NULL) {
            if(e_next->header.s.used == 0) {
                e->header.s.d_size += e_next->header.s.d_size + sizeof(e->header);
            } else {
                break;
            }
            e_next = ent_get_next(e_next);
        }
    //////////////////////////////////////////////
    } 
    ///////////////////////////////////////////////
    #endif
    #else /*Use custom, user defined free function*/
    #if LV_ENABLE_GC == 0
        LV_MEM_CUSTOM_FREE(e);
    #else
        LV_MEM_CUSTOM_FREE((void *)data);
    #endif /*LV_ENABLE_GC*/
    #endif
    }

kisvegabor, 2019-12-06:
Hi, 
you send a simple code snippet to reproduce?

Do you delete the active (loaded) screen? If so I could be a problem. You should load an another screen first to always have an active screen.

magailee, 2019-12-06:
> 
> 
> Hi,
> you send a simple code snippet to reproduce?
> 
> Do you delete the active (loaded) screen? If so I could be a problem. You should load another screen first to always have an active screen.

Hi, thanks your reply

The real dead location is below I traced out when the last child lv_obj of root active screen deleting:

    (lv_obj.c: line 2570)
    static void delete_children(lv_obj_t * obj)
    {
        ......
        ......
        /* Clean up the object specific data*/
        obj->signal_cb(obj, LV_SIGNAL_CLEANUP, NULL); <==== process dead "after" here (signal_cb() returned)
        ......
    }

I do delete the active (loaded) screen when my app finished that I don't need to load another active screen always I think, or maybe I should do nothing at app end.

here the main.c code snippet:

    static lv_obj_t *app_scr = NULL; // main root screen
    static volatile bool app_break = false;

    // signal handler for mainloop breaking
    void signal_handler (int sig) 
    { 
        printf("\napp mainloop break with signal %d\n", sig);
        app_break = true;    
    } 

    void *app_timer_thread (void *data)
    {
        (void)data;	// suppress warnings

        while (!app_break)
        {
            usleep(PERIOD_TIMER_ISR * 1000);   // sleep for 10 millisecond
            // must call lv_tick_inc() periodically to inform LVGL about how much time has elapsed
            lv_tick_inc(PERIOD_TIMER_ISR); // tell lvgl 10 ms were elapsed
        }

        pthread_exit(NULL);
    }

    int main (void)
    {
        lv_init(); // lvgl init
        lv_fs_if_init(); // lvgl pc file system init

        init_display();
        init_input_device();

        app_scr = lv_scr_act(); // get the currently active screen
        app_init(app_scr);
        
        //---------------------
        // lvgl tasks mainloop
        //---------------------

        //// timer-isr thread ////
        pthread_t timer_thread;
        pthread_create(&timer_thread, NULL, app_timer_thread, NULL);

        //// general break-handling setup //// 
        signal(SIGINT, signal_handler);  // ctrl-c break 
        signal(SIGTERM, signal_handler); // termination from others (mostly system)

        //// app main loop ////
        while (!app_break)
        {
            lv_task_handler();
            usleep(PERIOD_TASK_MAIN * 1000);
        }

        app_end(); 

        ///////////////////////////////////////////////////////////////////////////
        lv_obj_del(app_scr); // delete ‘app_scr and all of its children <=== !!!
        ///////////////////////////////////////////////////////////////////////////

        pthread_join(timer_thread, NULL);
        fbdev_exit();

        return 0;
    }

kisvegabor, 2019-12-06:
> obj->signal_cb(obj, LV_SIGNAL_CLEANUP, NULL);

It just asks the object to clean it's own allocated data. It's still valid after that.
Why you think the pointer is dead after calling the signal callback?

> here the main.c code snippet:

I've tried it and it worked for me.

embeddedt, 2019-12-08:
> I do delete the active (loaded) screen when my app finished that I don't need to load another active screen always I think, or maybe I should do nothing at app end.

Once you're done running LittlevGL, you shouldn't delete the active screen. Your application should just exit naturally.

Deleting the active screen, as @kisvegabor mentioned above, is likely to cause problems.

magailee, 2019-12-09:
> Once you're done running LittlevGL, you shouldn't delete the active screen. 
> Your application should just exit naturally.
> Deleting the active screen, as @kisvegabor mentioned above, is likely to cause problems.

Okay, now I see, thanks.

closed 2019-12-09 by magailee