lvgl/lvgl #1305: Reconsider release policy

kisvegabor, 2019-12-09:
The release policy of lvgl should be reconsidered to avoid immediate hotfix releases after a minor/major release.

My current idea:
1. Before minor/major release, release an `rc` (release candidate) version directly into `master`. E.g. `x.y-rc`
2. Allow 1 week to get feedback and fix warnings, forgotten things, etc.  
3. After 1 week remove the `rc` tag and make a normal release. E.g. `x.y`.
4. After the normal release change the version to `x.y.1-dev` to indicate that the next bugfix version is under development in `master`
5. Push bugfixes to `master`
6. Every month (near to 1st) make a bugfix release.  E.g. `x.y.1`
7. After bug fix releases add `dev` tag to the version. E.g. `x.y.2-dev`

TODO list before release:
- create a demo with all object types and try it
- test groups: keyboard, encoder
- run stress test
- test with 1,8,16,16 swapped, 32 bit color depth
- test with disabled animations and/or groups
- test with -O0, -Os and -O3 optimization.

It'd be nice to set up an automatic test environment (Travis?) for the tests above and see if at least it can be compiled and run successfully (even without a display dirver). 
Writing tests for the whole library is out of scope but we can test some core modules' functionality (e.g. lv_mem, lv_ll, lv_task, lv_color).

Ideas are very welcome. So please let me know what you think.

amirgon, 2019-12-09:
> It'd be nice to set up an automatic test environment (Travis?) for the tests above and see if at least it can be compiled and run successfully (even without a display dirver).

Consider [Azure Pipelines](https://azure.microsoft.com/en-us/services/devops/pipelines/) (azp) instead of Travis.  
It's free like Travis but more powerful. I noticed a trend of repos migrating from Travis to azp.

amirgon, 2019-12-09:
> TODO list before release:

In several occasions changes in lvgl caused `lv_micropython` build errors.  
For example, a function declaration (prototype) without definition is legal on C but will fail on `lv_micropython`.  
I suggest adding `lv_micropython` to the list (build unix port and esp32 port, run some basic test on the unix port). This can be automated when we have a CI tool.

kisvegabor, 2019-12-09:
I've just tried Azure Pipelines and integrated it into my lvgl fork. See: https://github.com/kisvegabor/lvgl/pull/1
It just `echo`s some test lines and not builds anything but at least it's integrated. 

> I suggest adding lv_micropython to the list (build unix port and esp32 port, run some basic test on the unix port). This can be automated when we have a CI tool.

Good idea. I've added it.

puzrin, 2019-12-09:
I see no problem with current hotfix releases, but if you wish to play with RC - usually pre-release names are: x.y.z-beta1, x.y.z-beta2, ..., x.y.z-rc1, x.y.z-rc2...

As far as i know, PIO does not support pre-release names. Probably, it will ignore those (i think, not a problem).

> Ideas are very welcome.

Add test to check versions in all configs are in sync. PIO config update was missed several times.

> It's free like Travis but more powerful. I noticed a trend of repos migrating from Travis to azp.

AFAIK azp is used by projects which need testing in windows env. Travis support for win is limited https://docs.travis-ci.com/user/reference/windows/.

If there are more advantages, could you point out where to read about?

The only really needed feature is sub-dependencies support - running lvgl repo tests should run lv_micropython repo tests on success.

puzrin, 2019-12-10:
> It'd be nice to set up an automatic test environment (Travis?) for the tests above and see if at least it can be compiled and run successfully (even without a display dirver).

CI just run something that programmer can do local from CLI, it's config is trivial. The real blocker for this is lack of "official" CLI scripting (`make` or something else) for developpers. You need to make a choice. Without CLI scripting, tests and benchmarks can not be added in useable way.

kisvegabor, 2019-12-10:
> I see no problem with current hotfix releases, but if you wish to play with RC - usually pre-release names are: x.y.z-beta1, x.y.z-beta2, ..., x.y.z-rc1, x.y.z-rc2...

Right now I think we need to deal with numbering beta/rc releases. Having 1 rc or beta should work for as (for now at least)

 > Add test to check versions in all configs are in sync. PIO config update was missed several times.

I've updated the first comment

---

I've read some docs but something is still not clear for me. Probably you know it. When a pull request arrives to a repo how can I clone the branch with the updates in the pull request? I found a way only to simply clone from GitHub.

I was able to clone a repo with submodules and build it (actually fails because SDL is missing):
config file: https://github.com/kisvegabor/lvgl/blob/master/azure-pipelines.yml
build log: https://dev.azure.com/kisvegabor/test/_build/results?buildId=14

puzrin, 2019-12-10:
> When a pull request arrives to a repo how can I clone the branch with the updates in the pull request? I found a way only to simply clone from GitHub.

https://stackoverflow.com/questions/1783405/how-do-i-check-out-a-remote-git-branch

`git fetch` + `git checkout branch_name`. I this what you asked about?

> I was able to clone a repo with submodules and build it (actually fails because SDL is missing):
> config file: [kisvegabor/lvgl:azure-pipelines.yml@`master`](https://github.com/kisvegabor/lvgl/blob/master/azure-pipelines.yml) [](/kisvegabor/lvgl/blob/HEAD@{2019-12-10T14:44:19Z}/azure-pipelines.yml)

I strongly recommend to NOT use scripting at vendor side. Devs should be able to run scripts locally. CI should call the same scripts. Something like `make dev_deps` + `make`.

Second - in CI session you already are in repo's folder with appropriate commit. To need to clone manually, except very specific cases.

kisvegabor, 2019-12-11:
> I strongly recommend to NOT use scripting at vendor side. Devs should be able to run scripts locally. CI should call the same scripts. Something like make dev_deps + make test.

That sound quite reasonable, I'll go this way.

>  in CI session you already are in repo's folder with appropriate commit. No need to clone manually, except very specific cases.

Ahh, that was the missing piece for *"how can I clone the branch with the updates in the pull request"*.

My plan is to create a `test` folder next to `src` with a Makefile and `lv_test.c`. In `lv_test.c` there will be a `main()` function but the whole file will be guarded with `LV_BUILD_TEST` (which will be set in the Makefile).

puzrin, 2019-12-11:
> My plan is to create a `test` folder next to `src` with a Makefile and `lv_test.c`. In `lv_test.c` there will be a `main()` function but the whole file will be guarded with `LV_BUILD_TEST` (which will be set in the Makefile).

With high probability, in addition to Makefile you may need this things:

- Test suite (assertion helpers, multiple tests runner, reporters). I use [Unity via PIO](https://github.com/puzrin/dispenser/blob/dev/.travis.yml#L19-L21). But you may wish to run Unity native, somehow (or any other test suite)
- The same, but + coverage reports (very useful, as paranoid warnings - cheap detector of expensive errors)
- QEMU - i know people use it to run tests. Never tried myself, but hope it can be used for precise ARM benchmarking. Тhere should be some benchmarking suites (~ as test suites, but for benchmarks)

kisvegabor, 2019-12-11:
In the first step, coverage is not really important. We just want to test some core modules functionality. So seeing 2% because only 3 files are tested is not that useful.

I'm thinking about writing a Python build script instead of Makefile. Do you know anything against it?

amirgon, 2019-12-11:
> I'm thinking about writing a Python build script instead of Makefile. Do you know anything against it?

How would you control file dependencies? Do you give-up upfront on incremental builds?
Did you consider CMake?

kisvegabor, 2019-12-11:
> How would you control file dependencies? Do you give-up upfront on incremental builds?

I'm not a big expert of build systems, so maybe I misunderstand the questions. 
My approach is we need to test with different global configurations in `lv_conf.h` so the whole project needs to be rebuilt anyway. 
So to test various configurations to see there are no build errors and warnings:
1. collect C files
2. set the new configs with `-D`
3. build object files
4. link them 
5. maybe run the program to see if it really can run  

To run some tests is almost the same just run the executable to perform the tests. So it's my naive approach to get started. I really don't have much experience with these things so let me know if I'm on the wrong way :slightly_smiling_face: 

> Did you consider CMake?

I think we will have more freedom with a full-featured language. I saw a lot of projects where python scripts are used to build the project. (e.g. ESP)

embeddedt, 2019-12-11:
Since we already have `.mk` files in all of the subdirectories, wouldn't it make more sense to just use a `Makefile` in `tests` that includes those `.mk` files? It also lets us test the `.mk` files to ensure that they include everything necessary.

amirgon, 2019-12-11:
> I think we will have more freedom with a full-featured language. I saw a lot of projects where python scripts are used to build the project. (e.g. ESP)

You can separate it to two tasks: 

 - Prepare the build (create the configuration files, set up build parameters and variables)
- The build itself.

My opinion is that the build itself should be done with a well known build system such as gnu make, CMake or something similar.
But for preparing the build and driving the build - your could use a Python script.

kisvegabor, 2019-12-11:
@embeddedt @amirgon Okay, I agree. Let's use python script to execute `make`s with different configs.

puzrin, 2019-12-11:
> I'm thinking about writing a Python build script instead of Makefile. Do you know anything against it?

I don't know python. But probably then [SCons](https://github.com/SCons/scons) will be good choice for you.

> To run some tests is almost the same just run the executable to perform the tests. So it's my naive approach to get started. I really don't have much experience with these things so let me know if I'm on the wrong way slightly_smiling_face

Test runner is usually part of test suite (Unity has such runner in ruby lang). It can also exist in modern build systems (SCons has something for that).

> In the first step, coverage is not really important. We just want to test some core modules functionality. So seeing 2% because only 3 files are tested is not that useful.

I don't suggest to make 100% coverage. I suggest to create separate build target with coverage reports enabled. Or, at least design your runners such way to keep this possibility.

puzrin, 2019-12-11:
> Since we already have `.mk` files in all of the subdirectories, wouldn't it make more sense to just use a `Makefile` in `tests` that includes those `.mk` files? It also lets us test the `.mk` files to ensure that they include everything necessary.

IMO it's not principal where to place Makefile. But it worth study existing test suites docs to understand how to organize `/test` content. For example, PIO requires separate subfolder for each test.

kisvegabor, 2019-12-12:
Thanks for the suggestions. As a newbie in this filed I want to start with something simple to get some experience.

Here is what I've done so far:
1. Python script + Makefile in [tests](https://github.com/kisvegabor/lvgl/tree/master/tests) folder 
2. 3 build configs in [Python](https://github.com/kisvegabor/lvgl/blob/master/tests/build.py) 
3. Return error is the build or the program fails
4. Config Azure to run these tests

And it seems working. See the 2 wrong and 1 ok PRs:
https://github.com/kisvegabor/lvgl/pulls

puzrin, 2019-12-12:
IMO looks like coding "just for fun". It does not answer questions "how to organize & run multiple tests". Also, most of py files are config generator. Those can be just located in separate folders as .h files. Dynamic configs do not simplify anything.

amirgon, 2019-12-12:
I think that having [lists of defines](https://github.com/kisvegabor/lvgl/blob/edd3c180332a727cd0967b92273349af9e48b239/tests/build.py#L49-L316) hard-coded in a python script is not very useful, not pretty and hard to maintain.
A script should contain functionality, not data.
If you want to represent data, use XML, YAML, CSV or whatever format you like and load it from your script.   
In this case a CSV could be useful as the information you have is a 2D matrix (*defines* x *build-types* (minimal_monochrome/all_obj_minimal_features/all_obj_all_features/...))

At least you could use a "base" build-type with common defines that all other build types inherit from (each of them only specifies what defines are different), instead of repeating the same defines names and values over again and again.  

(or.... if every define on `lv_conf.h` was guarded by `#ifndef`, you would only need to specify those defines which are different from the default, but we already had that discussion.)

kisvegabor, 2019-12-16:
> IMO looks like coding "just for fun". It does not answer questions "how to organize & run multiple tests".

It will be the next step :slightly_smiling_face: 
Contribution would be welcome in it.

> Also, most of py files are config generator. Those can be just located in separate folders as .h files. Dynamic configs do not simplify anything.

@amirgon suggested CSV, XML etc files. Do you also think `py` would be equally good as well as @puzrin suggested?

puzrin, 2019-12-16:
> Contribution would be welcome in it.

I have no resources to become lvgl contributor. But all "my" repos in your org have CI configured. This one includes some simple C testing https://github.com/littlevgl/lv_i18n/blob/master/.travis.yml.

kisvegabor, 2019-12-16:
@puzrin Thanks for the hint. Unity seems quite straightforward to use.

puzrin, 2019-12-17:
I like Unity too, but it's C code needs "runner" to execute multiple tests.

- Built-in is written in Ruby - you may not like that.
- PIO has built-in runner - you may not like lock to concrete "ide"
- You may wish to write your own runner (need to read docs about existing approaches & select preferred language)

Runner usually solves 2 main "problems":

1. Do multiple builds, because tests are usually splitted to multiple files or subfolders. That includes wildcards filters, grouping and so on.
2. Autogenerate `main` content, because manually writing [such things](https://github.com/littlevgl/lv_i18n/blob/master/test/c/test.c#L245-L273) is not convenient and prone to human errors (people can add new test but forget include it into `main`).

I also tried to learn how to create coverage reports for C, but not sure about code quality.

kisvegabor, 2019-12-26:
Unfortunately, I didn't have time to immerse deeper in Unity and polishing the tests. So I've added the current build tests to `dev-7.0`. 

The followings still should be added later:
- Move the test configurations into separate files
- Functional test: color, mem, txt, linked list, math etc modules, create/delete/set/get objects
- Integrate a test environment (e.g. Unity)

puzrin, 2019-12-26:
> 6. Every month (near to 1st) make a bugfix release.  E.g. `x.y.1`

IMO that's to rare. I'd declare 1 month as max possible gap, when accumulated patches must be landed to release. But not make it fixed interval.

There can be urgent bugfixes or something significant. Especially after new major releases. For example, warnings fix is significant improvement IMO.

I guess, you may see no difference because C has lack of package managers (those exists but are not common practice). When PM actively used, it's normal to make release every week.

kisvegabor, 2019-12-27:
> IMO that's to rare. I'd declare 1 month as max possible gap, when accumulated patches must be landed to release with no more delay. But not make it fixed interval.

@embeddedt What do you think?

> I guess, you may see no difference because C has lack of package managers (those exists but are not common practice). When PM actively used, it's normal to make release every week.

Thinking about package managers it'd also reasonable to have a "moving" tag e.g. `latest` or `stable` which always points to the latest release. Maybe it's a branch not a tag, I don't what is the practice. Or to keep backward compatibility: `v6-latest`, `v7-latest`.

embeddedt, 2019-12-27:
Perhaps we should dispose of the `master` branch altogether, and only have `v6.0`, `v6.1`, `dev-7.0` (renamed to `v7.0` upon release) branches. That way, someone could safely setup their package manager to use a certain branch knowing that it would never update to a new release. It wouldn't really change our workflow that much, and we could set the default branch on GitHub to be whatever the latest release was (in our case, `v6.1`).

kisvegabor, 2020-01-02:
Interesting idea. 
I'd still keep `master` because hotfix commits can also introduce errors. What about making patch releases time to time which are merged to the appropriate minor release branch, as you suggested?

amirgon, 2020-01-02:
> Perhaps we should dispose of the `master` branch altogether

My opinion is that you should keep master as the main trunk, because this is the most widely accepted convention and this is what users expect.   

Imagine a new user who is browsing or cloning the repo. Making some other branch the "default" branch could be confusing for them.  
99% of the users would try to browse/clone the master branch, and assume the other branches are development branches, future versions etc.

puzrin, 2020-01-02:
https://lucamezzalira.com/2014/03/10/git-flow-vs-github-flow/ - i already mentioned somewhere. Try to google for "git flow" and use one of well known approaches.

embeddedt, 2020-01-02:
The disadvantage of using `master` in that manner is that it will never be guaranteed to contain a certain version. I'm trying to avoid introducing extra tasks into the release workflow. Currently, patches to a released version don't require anything beyond a `git push`. I'd like to keep it that way if possible.

Another option is to keep `master`, while having a `v6.1` branch as well. The issue is that it would require periodic merging of the former into the latter, which introduces another task (unless we automated it so a new patch release was generated every week).

puzrin, 2020-01-02:
> Currently, patches to a released version don't require anything beyond a `git push`. I'd like to keep it that way if possible.

That's true only until you don't care about results :). Even in nodejs, where you can relax dependencies to 1 or 2 last numbers, people invented `package.lock` to avoid problems in big projects, and strict locking is actively forced.

True rolling deploy requires very high test coverage. Also, you can quick-fix website, but usually can't quick-upgrade flash. And, when problem reported, you should be able to reproduce setup exactly as user have (install exactly the same deps).

I'd suggest to keep aside arguments like "just a push", because they says nothing about final goal. Final goal is to provide possibility to manage deps in comfotrable way on user side, with low risks.

kisvegabor, 2020-01-04:
> i already mentioned somewhere. Try to google for "git flow" and use one of well known approaches.

I've checked some git flows but I'm still puzzled. It seems the common practice is to have only tagged versions in `master`. So we can do 2 things:
- create a `hotfix` branch where the simple fixes are committed and release from it periodically
- make a release after every commit in `master`

From tagged released in `master` `release/xxx` branches are created. But if we do a patch release every week we will have 100 release branches in 2 years.

So my vision is:
- as the API of the minor and patch releases should be backward compatible create `release/v5`, `release/v6` branches to let the user select the major version. 
- bugfixes still can go to `master` but tag them every week or every two weeks and merge them to the related `release/vx` branch.
- e.g. `release/v5` can have "local" releases which are not merged back to `master` because that code might not exist there already
- use `dev` branches for new minor and major versions
- when a version is considered ready, merge it to `master` and tag it as `rc/v6.1.0`.
- in summary  `master` will contain the freshest things, but stable versions are always available via:
  - tags in `master`
  - `HEAD` from `release/xxx`

What do you think?

amirgon, 2020-01-04:
> It seems the common practice is to have only tagged versions in `master`

Where did you read that this is the common practice?

> in summary `master` will contain the freshest things, but stable versions are always available via:
> * tags in `master`
> * `HEAD` from `release/xxx`

Makes sense to me.  
I think this is the common practice, and what most users would expect.

puzrin, 2020-01-04:
https://github.com/nodejs/node IMO that's a common examples of project which has to support LTS versions. Branches are created for major releases, then tags inside for minor releases and patches. Master always contains edge working code.

kisvegabor, 2020-01-04:
@amirgon 
> Where did you read that this is the common practice?

I've searched for [git flow images](https://www.google.com/search?q=git+flow&tbm=isch&ved=2ahUKEwiZjdCn5OrmAhXNOlAKHaeKAFgQ2-cCegQIABAA&oq=git+flow&gs_l=img.3...0.0..32434...0.0..0.0.0.......0......gws-wiz-img.dNBY_or57kQ&ei=bfUQXtmiNs31wAKnlYLABQ&bih=949&biw=1920) and almost all diagrams showed it.

@puzrin 
> https://github.com/nodejs/node IMO that's a common examples of project which has to support LTS versions. Branches are created for major releases, then tags inside for minor releases and patches. Master always contains edge working code.

That's quite the same that I've described. Good to know that it already exists.

@embeddedt What do you think?

puzrin, 2020-01-04:
> That's quite the same that I've described. Good to know that it already exists.

Because number of flows is very limited :). There are only 2 major flows - with parallel versions development (git flow), and with single "latest" version (github flow). Everything else are minor variations (latest + staging, and so on).

kisvegabor, 2020-01-05:
Seemingly I've suggested GitHub flow which encourages the use `master` not only for hotfixes but for new features too. (Instead of merged new features to `dev-x.y` as we do now).

I wonder how versioning happens in GitHub flow. How do they call the version after `v2.1` with two big features (which really makes it different from `v2.1`)? I've found suggestions only to use `git describe` and make regular releases.

puzrin, 2020-01-05:
> I wonder how versioning happens in GitHub flow.

They develop website, it has no versions at all :). That's named "rolling releases", "continuous deploy" etc. That will not be suitable for your users, because embedded world is very conservative.

I'd suggest you very simple thing. Until you have lot of commercial clients - use git flow but reduce time frame for previous versions support.

embeddedt, 2020-01-05:
> the API of the minor and patch releases should be backward compatible

Not entirely. The API itself is generally backwards/forwards-compatible but we do have a tendency to add/remove/move source files, as well as making other minor changes. For example, applying a 6.1 Makefile to version 6.0 isn't going to work (as there are new source files) and vice versa.

If we do adopt that approach, we would have to be *very* strict about not changing much in minor releases (otherwise the problems mentioned above appear).

One option would be to make more frequent major releases. That way there is less of a shock when a major version releases.

----------

I'll need to read some more about git flow, as I don't entirely understand how `master` works with two major releases at a time. For example, if `master` contains what is now `dev-7.0`, and we make a bugfix that also applies to version 6.1, how do we merge the bugfix into an older `release/v6` without bringing all of the new 7.0 changes as well?

puzrin, 2020-01-05:
> For example, applying a 6.1 Makefile to version 6.0 isn't going to work (as there are new source files) and vice versa.

And it's better to avoid such things :). As example - pio demo "expired" because configs are not compatible too.

> I'll need to read some more about git flow, as I don't entirely understand how master works with two major releases at a time. For example, if master contains what is now dev-7.0, and we make a [...]

Usually people don't actively develop 2 versions in parallel. They accept all non-breaking things to master, and postpone breaking things unmerged as "features". Master is merged to active release branch when it's time to create minor release. When authors decide to make breaking changes, they stop propagate master to release branch, until new major release happens. From this moment release branch become independant and accepts fixes only.

> how do we merge the bugfix into an older release/v6 without bringing all of the new 7.0 changes as well?

cherry pick + manual conflicts resolve when required. That's boring, yes. But you can backport only when "there is no chance to avoid that" (critical things like security fixes). For everything else you can advice upgrades to new major version.

embeddedt, 2020-01-05:
> cherry pick + manual conflicts resolve when required. That's boring, yes. But you can backport only when "there is no chance to avoid that" (critical things like security fixes). For everything else you can advice upgrades to new major version.

Essentially, it would be a similar workflow to what we use now (where fixes are applied to `master` and then merged to `dev-7.0` so that they are available on both sides).

I think this workflow will work, then.

puzrin, 2020-01-05:
I'd like to clarify: instead of master, in "git flow" you surggest users to use tags, produced via latest "release XX" branch. More tags, more releases, no "fetch latest".

kisvegabor, 2020-01-05:
@embeddedt 
> One option would be to make more frequent major releases. That way there is less of a shock when a major version releases.

So we need to choose between smaller API breaks in every ~2 months or huge API breaks in every 6 months. It seems to me more often major releases would make the API fragile and annoy the users. E.g. if we make two major release while they are developing even a simple project, they need to refactor 2 times during the development. However, I'm not sure I see it well, so of course, we can discuss it.

> I'll need to read some more about git flow, as I don't entirely understand how master works with two major releases at a time. For example, if master contains what is now dev-7.0, and we make a [...]

We can apply the fix to `release-x` first and merge `release-x` back to `master` if the fix still makes sense in `master`.

@puzrin 
> I'd like to clarify: instead of master, in "git flow" you surggest users to use tags, produced via latest "release XX" branch. More tags, more releases, no "fetch latest commit".

As I see the changes so far are:
- create `release-x` branches
- make more often patch releases

In addition, we can consider if we really need a development branch for minor releases or just use feature branches. If there are 2-3 parallelly developed features we can merge them at the same time and create a minor release. It's mean more often minor releases too.
I'd also help to keep the documentation updated because only a few parts should be updated on every minor release and it means less chance to forget to document something. 
What do you think?

puzrin, 2020-01-05:
> In addition, we can consider if we really need a development branch for minor releases or just use feature branches. If there are 2-3 parallelly developed features we can merge them at the same time and create a minor release. It's mean more often minor releases too.

Don't mix new features and bugfixes. Bugfixes can be landed directly, with tests, and there is nothing to stage. New features land to master first.

> I'd also help to keep the documentation updated because only a few parts should be updated on every minor release and it means less chance to forget to document something.

I'd suggest you to support [changelog](https://github.com/littlevgl/lv_font_conv/blob/master/CHANGELOG.md) file and update it on each notable change (feature, bugfix, breaking). It's convenient both for users and for devs.

puzrin, 2020-01-06:
> So we need to choose between smaller API breaks in every ~2 months or huge API breaks in every 6 months. It seems to me more often major releases would make the API fragile and annoy the users.

There are no universal recipe. While lvgl architecture changes fast, it's better to have shorter cycles for major releases + may be select some "LTS" version. Later, when architecture become more solid, major releases will happen more rare.

Consider 6 months not as fixed period but as max deadline. In "ideal worlds", current 6.0 release could be "temporary", and 6.1 could be 7.0.

I understand your wish to provide extra stability, but that's difficult in period while architecture & development process drift significantly. Possible, but require more efforts than you could spend for free. 

> E.g. if we make two major release while they are developing even a simple project, they need to refactor 2 times during the development. However, I'm not sure I see it well, so of course, we can discuss it.

From my personal experience, real issues were not because of API change, but because config become broken. Twice(!), with 6.0+...6.1+.

When i bulk upgrade outdated packages in nodejs, i usually inspect changelogs & see commit logs when unsure. That's usually quick.

embeddedt, 2020-01-06:
> real issues were not because of API change, but because config become broken. Twice(!), with 6.0+...6.1+.

Usually, whenever lvgl is updated, you should compare your copy of `lv_conf.h` to `lv_conf_template.h` and add any new settings.

kisvegabor, 2020-01-06:
> Don't mix new features and bugfixes. Bugfixes can be landed directly, with tests, and there is nothing to stage. New features land to master first.

I meant stage merging features only, and merge fixes immediately. 

> I'd suggest you to support changelog file and update it on each notable change (feature, bugfix, breaking). It's convenient both for users and for devs.

That's a good idea.

---

An advantage of more frequent major releases came to my mind. In v7.0 will have 2 new changes: new drawing and style system. After the new drawing system, I have rewritten some object types to add new features, improve them, or to simplify them. However, these are based on `dev-7.0` so I can't merge them into `master` because the new style system also meat to be part of `v7.0`. 

If we had a major release with the new drawing system only, some other features would be already available in `master`.

Besides with a release with fewer changes probably have less critical issues. I mean if we rewrite the half library (like in v7) probably there will be a lot of issues which will turn out only after release. 

So I tend to agree with more frequent major releases.

puzrin, 2020-01-06:
> Usually, whenever lvgl is updated, you should compare your copy of `lv_conf.h` to `lv_conf_template.h` and add any new settings.

That may be true for 6.x => 7.x upgrade (actually, i used `meld` to refresh). But unexpected for 6.0 => 6.1 and especially unexpected for 6.1.0 => 6.1.+

You explain how to fix problem, but i explain some kind of problems should not happen at all :). And IMO that's possible with reasonable low additional efforts.

---

@kisvegabor https://github.com/nodeca/js-yaml/blob/master/CHANGELOG.md - that's a real example + spec ref to one of possible changelog format. Advantage of unified format is in additional processing tools. For example, to automate export into github releases page.

kisvegabor, 2020-01-09:
@embeddedt 
Could you comment on my proposal on getting rid of dev branches for minor releases?
> In addition, we can consider if we really need a development branch for minor releases or just use feature branches. If there are 2-3 parallelly developed features we can merge them at the same time and create a minor release. It's mean more often minor releases too.
I'd also help to keep the documentation updated because only a few parts should be updated on every minor release and it means less chance to forget to document something.
What do you think?

---

The more I think about tests I see them less and less evil. :slightly_smiling_face: 
I still believe pursuing coverage would mean a huge extra burden, but we can start to write tests for new features. That is if somebody adds a new feature he needs to add tests to verify the feature's functionality. The same would apply for bugfixes: if a corner case situation is resolved, there should be a test to be sure the bug is not introduced again. Slowly we can add tests to the existing modules too.
What do think?

embeddedt, 2020-01-09:
@kisvegabor It seems reasonable to me. The smaller the difference between releases, the faster we can drop support for older versions.

> That is if somebody adds a new feature he needs to add tests to verify the feature's functionality.

For example, I would have to add an automated test that ensures advanced hit-testing works as expected?

puzrin, 2020-01-09:
> For example, I would have to add an automated test that ensures advanced hit-testing works as expected?

It would be nice, but not deadly critical. Critical is to write code such way to be testable, when really needed.

I'd suggest you to inspect issues for last XX months, and decide yourself where it's time to add tests.

kisvegabor, 2020-01-10:
>  It seems reasonable to me. The smaller the difference between releases, the faster we can drop support for older versions.

Great!

> For example, I would have to add an automated test that ensures advanced hit-testing works as expected?

If you agree, yes. I don't mean anything super-deep but something like you could use anyway to verify the feature works. 
- If the feature is "algorithm only" then we can simply test it. (e.g. `lv_area_is_point_on` with radius)
- Rendering related things can be verified via screenshots.
- We need to find a good way to verify input device-related features (like hit test). The question is how to send "virtual" clicks/releases (drags?).

If you are not against this testing I can put together a  minimal engine for it.

puzrin, 2020-01-10:
> * We need to find a good way to verify input device-related features (like hit test). The question is how to send "virtual" clicks/releases (drags?).

In general, search for "text mocks" and "text spies". I see no problems to create testing indev with additional method, accepting array of recorded `{ action, delay }` as param.

kisvegabor, 2020-01-13:
> But testing such way would be an "integration test". In most cases you need "unit test" instead. I mean, you compile only ONE file, simulate input calls and see output reaction. Principle is simple - if each separate file is ok, they will be ok together.

It seems to me it'd be much easier to maintain a functional test compared to unit tests. Writing tests for every line is tedious and very time-consuming.  I see the advantages though. 
With our limit resources, we need to make compromises.

puzrin, 2020-01-13:
It's not correct to compare different kinds of tests, intended for different needs.

Functional tests are usually not suitable for 100% coverage and precise branch control.

kisvegabor, 2020-01-14:
> Functional tests are usually not suitable for 100% coverage

As I've mentioned, in lack of resources (dedicated developers) keeping up a constant high coverage would be too much work.

Besides I'd be more interested if features are really working or not. Maybe it's only dumb logic: "Working or not? Don't care how" :smile: But as I see that's what we can do now. It should much much better than doing nothing (regarding tests).

puzrin, 2020-01-15:
I don't say what you have to do, i only explain how testing is usually done. It's only your choice, how to use this info.

Asked at local forum about c/cpp testing:

1. Most use cmake/ctest to build/run
2. Most prefer CPP frameworks (even for C code). Most frequently used are:
   - catch2 (or catch)
   - gtest+gmock

kisvegabor, 2020-01-15:
> I don't say what you have to do, i only explain how testing is usually done. It's only your choice, how to use this info.

Got it! :slightly_smiling_face: 

> Asked at local forum about c/cpp testing...

Thank you. I'll take a look on them.

[33 more replies]

closed 2020-05-28 by kisvegabor