lvgl/lvgl #1306: Label text missing after decimal point when style's letter_space is nagtive

magailee, 2019-12-10:
**Describe the bug**

I updated the lvgl this week from github , and my app made of lvgl got a problem as mentioned above. I wrote a test as below as my issue. 

**To Reproduce**

test src:

    #include "lvgl/lvgl.h"
    #include "lv_drivers/display/fbdev.h"
    #define DRAW_BUF_SIZE (LV_HOR_RES_MAX * LV_VER_RES_MAX / 4)
    int main ()
    {
        lv_color_t draw_buf0[DRAW_BUF_SIZE];
        lv_disp_buf_t disp_buf;
        lv_disp_drv_t disp_drv;

        lv_init(); // lvgl init
        fbdev_init();
        lv_disp_drv_init(&disp_drv);
        lv_disp_buf_init(&disp_buf, draw_buf0, NULL, DRAW_BUF_SIZE);
        disp_drv.buffer = &disp_buf;
        disp_drv.flush_cb = fbdev_flush;
        lv_disp_drv_register(&disp_drv);

        lv_style_t sty;
        lv_style_copy(&sty, &lv_style_plain);

        sty.text.letter_space = -1; // !!! nagtive is ok before !!!

        lv_obj_t *lab, *labs[3];
        for (int i = 0; i < 3; i++)
        {
            lab = lv_label_create(lv_scr_act(), NULL);
            lv_label_set_style(lab, LV_LABEL_STYLE_MAIN, &sty);
            lv_obj_set_pos(lab, 40, (i + 1) * 30);
            labs[i] = lab;
        }

        lv_label_set_text(labs[0], "0123456789."); // decimal point missing in output
        lv_label_set_text(labs[1], "012345.6789"); // missing tail after decimal point
        lv_label_set_text(labs[2], "012.3456789");

        lv_task_handler();
        return 0;
    }

output in my device (linux framebuffer):

    0123456789
    012345.
    012.

embeddedt, 2019-12-10:
What font are you using?

magailee, 2019-12-11:
> 
> 
> What font are you using?

By this example, it's LV_FONT_DEFAULT (lv_font_roboto_16). And I've tried most of fonts like as lv_font_roboto_22, lv_font_roboto_28 ... and other fonts made by myself. All I got the same result.

kisvegabor, 2019-12-11:
Hi,

I've fixed it in `master`. Can you confirm if it's working now?

magailee, 2019-12-11:
> I've fixed it in `master`. Can you confirm if it's working now?

I'm sorry about I can not update anything fresh from `mater` to confirm, could you tell me which part of source fixed I should git it ?

kisvegabor, 2019-12-11:
Upps, sorry. It seems I haven't pushed it.
It should be here: https://github.com/littlevgl/lvgl/commit/f54ecc470cde598c56488809bac0e42f1fc5b58b

magailee, 2019-12-11:
> 
> Upps, sorry. It seems I haven't pushed it.
> It should be here: [f54ecc4](https://github.com/littlevgl/lvgl/commit/f54ecc470cde598c56488809bac0e42f1fc5b58b)

It's working OK now 👍.
Thank you for early bug fixing :-)

closed 2019-12-11 by magailee

kisvegabor, 2019-12-11:
Great! 

Thank you for the report! :slightly_smiling_face: