lvgl/lvgl #1308: set_obj_size and alignment

crnchbng, 2019-12-10:
**Describe the bug**

When an parent object is resized the aligned children objects aren't updated in relation to the new size of the parent object. The flag ```auto_realign``` on the parent and children is set.

**To Reproduce**
```C
        // create parent container with default size
        lv_obj_t* parent = lv_cont_create(lv_scr_act(), NULL);
        lv_obj_set_auto_realign(parent, true);
        
        // create child and align on the top middle of the parent
        lv_obj_t* child = lv_cont_create(parent, NULL);
        lv_obj_set_auto_realign(child, true);
        lv_obj_align(child, parent, LV_ALIGN_IN_TOP_MID, 0, 0);
        
        // resize parent
        lv_obj_set_size(parent, LV_HOR_RES, 60);
```

**Expected behavior**

The children's alignments should update after the size of object ```parent``` is changed. 

**Additional context**

I have seen that ```lv_obj_set_size``` spawns a signal ```LV_SIGNAL_PARENT_SIZE_CHG``` which is not caught (implemented) yet in the ```signal_cb```. 

Next there is a call on the children of ```parent``` through ```lv_obj_realign```, ```lv_obj_align``` or ```lv_obj_align_origo```, ```lv_obj_set_pos``` and finally ```refresh_children_position```, that only take into account to change the children's positions according to the old position and not the alignment. 

**Just a quick fix which works for my case**
For now, I help my self with the following code added in the ```lv_obj_signal``` function:
```C
#if LV_USE_OBJ_REALIGN
    else if(sign == LV_SIGNAL_PARENT_SIZE_CHG) {
        if(obj->realign.auto_realign) {
            lv_obj_realign(obj);
        }
    }
#endif
```

**Other remarks:** 
Here I used parent and base as the same object, but the ```lv_obj_align``` or ```lv_obj_align_origo``` allows to clearly distinguish between a parent and a base object on which the object will align to. Unfortunately the call of ```refresh_children_position``` in  ```lv_obj_set_pos``` on a parent object might lead to unexpected behavior where the child is aligned to another base object than parent.

When looking at a function like ```lv_obj_align```, I'm not sure which way are you going in the future: The signal way or the ```refresh_children_position``` way.  Either way, it would be better to call a function on each child and let them decide whats their base for alignment and positioning is. 

Where there is still a issue arising, when the base's position or size is changed and how to signal the elements which are aligned to that base object.

kisvegabor, 2019-12-12:
Hi,

Thanks for the report.

My intention with this feature was to perform the realignment if the object itself changes.

The really proper solution would be to check all the other objects if an object's size/position is changed and see if any of the other objects are using it as `base_obj` and perform the realignment if necessary. However, it'd be quite slow. 

As a semi-solution - which would cover the majority of the cases - we can use you proposal with `LV_SIGNAL_PARENT_SIZE_CHG`.

kisvegabor, 2020-01-06:
@crnchbng 
I've added your initials proposal to `dev-7.0`.

Thank you for the report and the suggestion!

closed 2020-01-06 by kisvegabor

fpw, 2020-01-25:
This patch is very useful for applications that support resizing. Could it please be applied to the master as well so we can use it in our 6.1.1 applications?

embeddedt, 2020-01-25:
We generally don't add new features to `master` after a release, but you can apply the patch locally.

commit 1c46d69e (2020-01-06): apply lv_obj_realign on parent size change. Close #1308