lvgl/lvgl #1311: Rework styles

kisvegabor, 2019-12-13:
For a long time, I'm considering to improve the style system of LittlevGL.

**The issues I see**
- the styles are to complex: you need to create a style to change only one parameter, e.g. label color
- there is no way to easily change something locally for only one object (you need to create a whole new style)
- it's difficult to describe uncommon properties: e.g. the line length on a line meter comes from a padding (in lack of better property) 

**So the goals would be**
- support local (per object) style change too
- still support style inheritance
- make the styles smaller

**The idea**
My idea is inspired by CSS. 
- Have a list of enums such as `LV_STYLE_RADIUS`, `LV_STYLE_BORDER_COLOR`
- In `lv_style_t` create some kind of a lists where property names and their values are stored
- Create this list dynamically. I.e. add a new element only if it's explicitly set.
- When a property (e.g. radius) is queried, check if it's set in the object and if so return it. If not check the parent, and so on. 
- Object types can also have custom properties. E.g. `LV_LMETER_STYLE_LINE_LENGTH`.
- In `lv_obj_t` have `lv_style_class_t` where styles can be chained. E.g. 
```c
lv_obj_add_style_class(obj, &btn);     //Button-like appearnce
lv_obj_add_style_class(obj, &btn_lg) //Overwrite only paddings
lv_obj_add_style_class(obj, &btn_red) //Overvrite only colors
```  
- Besides `lv_style_class_t` have a simple `lv_style_t *` in `lv_obj_t` where the object's local style modifications are storder. E.g.
```c
lv_obj_set_style_property(obj, LV_STYLE_RADIUS, 20); //Overwrite radius from classes
```

**What I have so far**
- I made an initial, very minimal implementation and it's working
```c
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_color(&style, LV_STYLE_BG_COLOR, LV_COLOR_RED);
    lv_style_set_value(&style, LV_STYLE_BG_RADIUS, 5);

    lv_obj_t * obj = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_pos(obj, 100, 30);
    obj->style_p = &style;
```
- Now there are 4 set/get functions for opa, value (int16_t), color and pointer types. It's ok but there could be `lv_style_set(&style, TYPE, value)` macro which can call the corrent function based on `TYPE`. However, the macros can't be used from Micropython so I don't know what to do with this part. @amirgon ?

It's a huge rework, but I think it's worth it because finally, it will be much easier to use the styles and they will be smaller.

Please, let me know what you think about it. I hope I explained it in an understandable way. :slightly_smiling_face:

amirgon, 2019-12-13:
> * It's ok but there could be `lv_style_set(&style, TYPE, value)` macro which can call the corrent function based on `TYPE`. However, the macros can't be used from Micropython so I don't know what to do with this part. @amirgon ?

@kisvegabor As long as you use **inline functions** (could be *static* inline functions, which don't cost more ROM/cycles than macros), and **enums** instead of C macros, it is supported by Micropython.   

Is there any special reason you need to use macros instead of inline functions and enums?

kisvegabor, 2019-12-13:
> Is there any special reason you need to use macros instead of inline functions and enums?

Now I have these 3 functions to set the properties in a style:
```c
void lv_style_set_value(lv_style_t * style, lv_style_property_t prop, lv_style_value_t value);
void lv_style_set_color(lv_style_t * style, lv_style_property_t prop, lv_color_t color);
void lv_style_set_opa(lv_style_t * style, lv_style_property_t prop, lv_opa_t opa);
```
They do the same things but have different parameter type.

With a macro there could be something like:
```c
LV_STYLE_SET(style, prop, value) \ 
       PROP_IS_COLOR(prop) ? lv_style_set_color(style, prop, value) :    \ 
       PROP_IS_VALUE(prop) ? lv_style_set_value(style, prop, value) :   \
       lv_style_set_opa(style, prop, value) 
```

kisvegabor, 2019-12-13:
@amirgon 
I've just realized it won't work because the wrong parameter assignment is still there. E.g. if it's a color, the compiler sees `lv_style_set_opa(style, LV_STYLE_BG_COLOR, LV_COLOR_RED)` which is clearly an error.

embeddedt, 2019-12-13:
When you call `lv_obj_add_style_class` do the passed styles override properties in the order that they are added, or is there a more advanced concept of specificity?

> Now I have these 3 functions to set the properties in a style

Are all style values going to be integers? Can we just pass a `int32_t` and cast to the appropriate type based on the property chosen? Or, could we require passing a pointer to the value instead? Having 3 functions seems a bit cumbersome, particularly with the problems you mentioned with MicroPython.

embeddedt, 2019-12-13:
It seems to me that based on CSS we need at least 3 scopes that styles can be applied to:

* A scope for the object type (themes would use this).
* A scope for classes added to the object (would replace the current use of styles).
* Local modifications to that particular object (would make it easier to change minor things, like the text color).

I think we can leave out using IDs or something similar to style objects; the use of IDs is frowned upon in modern web development anyways.

kisvegabor, 2019-12-13:
@embeddedt 

> When you call lv_obj_add_style_class do the passed styles override properties in the order that they are added, or is there a more advanced concept of specificity?

Yes, I thought this way. It's only interesting if two class defines the same attribute.  
CSS works also like this doesn't it?

> Are all style values going to be integers?

As far as I see 4 types are required:
1. opa: uint8_t
2. color: lv_color_t (could be 32 bit long as well and it's a struct, not a single integer)
3. value: int16_t (padding, width, radius etc)
4. pointer: void * (fonts)

Right now I can't image a concept to make it work with a single function. 

Maybe with `vargs`? Always read only one parameter but with the required type. 
It should work with `set` functions but not helps with `get`.

> A scope for the object type (themes would use this).

I still don't know exactly how to handle object type level styles. E.g. `lv_btn` can have a default class which or a similar from the theme is automatically applied to the object.

> A scope for classes added to the object (would replace the current use of styles).

Yes, but support a "style list" similarly to `class="btn btn-lg btn-red"`

> Local modifications to that particular object (would make it easier to change minor things, like the text color).

Yes.

> I think we can leave out using IDs or something similar to style objects; the use of IDs is frowned upon in modern web development anyways.

I agree.

kisvegabor, 2019-12-13:
I've tested `vargs` and it works well:
```c
    static lv_style_t style_rect;
    lv_style_init(&style_rect);
    lv_style_set_prop(&style_rect, LV_STYLE_RADIUS, 5);
    lv_style_set_prop(&style_rect, LV_STYLE_BG_COLOR, LV_COLOR_RED);
    lv_style_set_prop(&style_rect, LV_STYLE_BG_COLOR, LV_COLOR_GREEN);
    lv_style_set_prop(&style_rect, LV_STYLE_BORDER_OPA, LV_OPA_COVER);

void lv_style_set_prop(lv_style_t * style, lv_style_property_t prop, ...)
{
   va_list valist;
   va_start(valist, prop);

   lv_color_t color;
   lv_opa_t opa;
   lv_style_value_t value;

   switch(prop & LV_STYLE_ATTR_TYPE_MASK) {
   case LV_STYLE_ATTR_TYPE_COLOR:
       color = va_arg(valist, lv_color_t);
       lv_style_set_color(style, prop, color);
       break;
   case LV_STYLE_ATTR_TYPE_OPA:
       opa = (lv_opa_t)va_arg(valist, int);
       lv_style_set_opa(style, prop, opa);
       break;
   case LV_STYLE_ATTR_TYPE_VALUE:
       value = (lv_style_value_t)va_arg(valist, int);
       lv_style_set_value(style, prop, value);
       break;
   }

   /* clean memory reserved for valist */
   va_end(valist);
}
```

embeddedt, 2019-12-13:
I think we need to include some type of logic similar to CSS selectors if we have multiple scopes for styles. We need a central location where styles are registered and applied in order of specificity.

The main issue I see with that is performance. Recalculation of styles is often a performance bottleneck in web browsers; I imagine we will have similar problems if this isn't architected cleverly enough.

I think the logic for handling themes/default classes should be as extensible and generic as possible, similar to how CSS frameworks work. It should be possible to override the theme in much the same manner that a class can be applied to an object.

Perhaps we should remove the concept of themes all together, and replace them with functions that simply add a number of object-level styles to the style registry. It would simplify the duopoly that we have now between styles and themes (i.e. when do we need a theme vs style?). Switching between themes would also be simplified - as it would now simply become changing a style.

kisvegabor, 2019-12-13:
> I think we need to include some type of logic similar to CSS selectors if we have multiple scopes for styles. We need a central location where styles are registered and applied in order of specificity.

Do you mean something like "apply this style only to labels"?

> The main issue I see with that is performance. Recalculation of styles is often a performance bottleneck in web browsers; I imagine we will have similar problems if this isn't architected cleverly enough.

I'm also thinking about it.
 I've already introduced the "inheritable" flag for properties (e.g. border is not inherited). It's also stolen from CSS where only [these properties](https://stackoverflow.com/a/5612360/1999969) are inherited (in our case, it mainly means the text-related properties.) 
So if a property is no inherited we need to search only in the "local style" and in classes. Assuming there is a local style with 5 properties and classes with 20 properties together. At worst case, we need to search amongst 25 properties. The search needs to be done only once per property in the design function. With inheritance, the number of properties easily can be about 100. 
Let's say we have 
- a 100 MHz MCU
- we need to search amongst 50 properties in classes and styles on average (between 25 and 100),  
- there are 10 properties to get in the design function (e.g. radius, border color, etc) 
- there are 20 objects to draw. 

It means 50 x 10 x 20 = 10,000 "search steps". If each of them takes 20 instruction then 200k/100Mhz = 2ms.

In the current implementation, all properties are there in `lv_style_t`. So with 20 objects going to the parent on average, it takes 20x2 = 40 steps (1/250 of 10,000).

So it's really crucial to make the get functions as fast as possible. 

> I think the logic for handling themes/default classes should be as extensible and generic as possible, similar to how CSS frameworks work. It should be possible to override the theme in much the same manner that a class can be applied to an object.

I can't see the exact method yet, but we really have some good options with "classes".

> Perhaps we should remove the concept of themes all together, and replace them with functions that simply add a number of object-level styles to the style registry. It would simplify the duopoly that [...]

Maybe finally the themes can be a collection of "classes" similarly to Bootstrap.

embeddedt, 2019-12-13:
> Do you mean something like "apply this style only to labels"?

Yes. That way there is no need to keep adding, for example, a material class to every button. You could still override the material properties using a class, but they would be available on every button by default (if you loaded the object-scope material style).

> there are 10 properties to get in the design function (e.g. radius, border color, etc)
> It means 50 x 10 x 20 = 10,000 "search steps".

The way most browsers implement it is to compute the whole style at once and cache it. That way the whole property tree only needs to be searched once. The disadvantage for us is that it consumes more memory. In addition to the memory required by the classes/local styles, an `lv_style_t` equivalent also has to be kept in the object structure. It might still be a reasonable tradeoff though.
It eliminates the need to recompute styles repeatedly. Only when a style is changed would everything have to be recalculated (and most objects' styles wouldn't change at all, hopefully).

If we had the memory, we could even implement something more complex like hashing the style combinations and saving them to increase performance further. But I think that's overkill for an embedded library.

amirgon, 2019-12-13:
> I've tested `vargs` and it works well:

Micropython binding does not support variadic arguments, and I don't have a clear idea how to support it.

When you declare a function like `void lv_style_set_prop(lv_style_t * style, lv_style_property_t prop, ...)`, take into account that **this will not be supported in Micropython** (at least, the variadic argument `...` will be omitted).

My personal opinion is that using functions with variadic arguments is error-prone and dangerous in C. The compiler does not protect you from errors related to the type or the number of arguments. Even for `printf`, gcc added a special [`format` attribute compile time mechanisms](https://gcc.gnu.org/onlinedocs/gcc/Common-Function-Attributes.html#Common-Function-Attributes) to protect the user from such problems and validate the variadic arguments on compile time.    

I think it makes more sense and simplify things to use a different function for each type of property you want to set or get.

kisvegabor, 2019-12-14:
I've pushed the current version which:
- works only with `lv_obj`
- supports local styles
- supports class "chaining"

Here are the most important parts:
- Declare the [style property IDs](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_style.h#L78-L102). The IDs are organized into groups, defined by the upper 4 bit of the ID. E.g. `2` for [BORDER](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_style.h#L88-L92).
- In [lv_style_t](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_style.h#L106-L110) in `map` I build a list with property IDs and values in the [set functions](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.c#L132-L149). `size` stores the actual size of the map. `used_groups` will be a bitfield to tell if a style contains a certain type of properties. Each bit represents a group. It makes it fast to decide if a style may contain a property or not. 
- The [get functions](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_style.c#L165-L176) look for the property in a style using [get_property_index](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.c#L280-L312)
- In [lv_obj_t](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_obj.h#L207-L208) there is an `lv_style_t` for the local styles and an [lv_obj_style_chain_t](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_obj.h#L190-L193) (a single linked list of styles) whose [first element points to the local style](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_obj.c#L192-L194).
- The [object level style get functions](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_obj.c#L2085-L2109) checks all styles in the chain until finding the requested property. If not found, and the style can be inherited check the parent too.
- The [object level style set](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_obj.c#L1230-L1233) simply sets the local style
- [Classes](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_obj.c#L1235-L1244) are simple `lv_style_t`s which are inserted after the first element (local style) of the chain. 
- The rectangle draw function is updated (the others will be updated too) to accept an [lv_draw_rect_dsc_t](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_draw/lv_draw_rect.h#L27-L39) instead of a style. 
- The draw dsc can be created from an object with [lv_obj_init_draw_rect_dsc](https://github.com/littlevgl/lvgl/blob/a101e9a3e51956f7880fadeaeddc3f117d02a8f0/src/lv_core/lv_obj.c#L2526-L2541). It makes a minor optimization: don't get irrelevant properties, e.g. border color if border width == 0.

I've tested with this code:
```c
   /*Set style directly on the screen*/
    lv_obj_t * scr = lv_scr_act();
    lv_obj_set_style_color(scr, LV_STYLE_BG_COLOR, LV_COLOR_CYAN);
    lv_obj_set_style_color(scr, LV_STYLE_BORDER_COLOR, LV_COLOR_RED);
    lv_obj_set_style_value(scr, LV_STYLE_BORDER_WIDTH, 5);

    /* Create styles
     * Will be used as classes */
    static lv_style_t style1;
    lv_style_set_color(&style1, LV_STYLE_BG_COLOR, LV_COLOR_YELLOW);

    static lv_style_t style2;
    lv_style_init(&style2);
    lv_style_set_color(&style2, LV_STYLE_BG_COLOR, LV_COLOR_ORANGE);
    lv_style_set_value(&style2, LV_STYLE_BORDER_WIDTH, 3);

    /*Create an object*/
    lv_obj_t * obj = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_pos(obj, 100, 30);

    /*Set some styles directly*/
    lv_obj_set_style_value(obj, LV_STYLE_RADIUS, 5);
    lv_obj_set_style_opa(obj, LV_STYLE_BORDER_OPA, LV_OPA_COVER);
    lv_obj_set_style_color(obj, LV_STYLE_BORDER_COLOR, LV_COLOR_PURPLE);

    /*Add classes*/
    lv_obj_add_style_class(obj, &style1);
    lv_obj_add_style_class(obj, &style2);
```
[image]

kisvegabor, 2019-12-14:
@embeddedt 
> Yes. That way there is no need to keep adding, for example, a material class to every button. You could still override the material properties using a class, but they would be available on every button by default (if you loaded the object-scope material style).

It works similarly to the themes now, doesn't it? If there is an active theme, apply a class on the object when it's created.

> The way most browsers implement it is to compute the whole style at once and cache it.

Let's how fast it will be. At first look, caching all the properties seems a little bit overkill. I've found two ways to optimize :
- `used_groups` in `lv_style_t` to skip most of the searches
- get only the required properties (last point in my previous post)

They should reduce the number of searches significantly. But we will see...
  
> If we had the memory, we could even implement something more complex like hashing the style combinations and saving them to increase performance further. But I think that's overkill for an embedded library.

How do you mean hashing? Hashing the current combinations to see if there is a difference later?

@amirgon 
> I think it makes more sense and simplify things to use a different function for each type of property you want to set or get.

Finally, I agree. Besides, it doesn't work with MicroPython and really error-prone, it has some overhead and performance is crucial in this case. So let's leave the separate functions for now and hopefully, we will find a  way to uniform them later.

embeddedt, 2019-12-15:
> Let's how fast it will be. At first look, caching all the properties seems a little bit overkill.

It does simplify the code and speed it up for certain cases, however. For example, if we have a child object, there is an inheritable property 3 ancestors up the tree, and the child's style is changed, caching means that only the child's style needs recalculation. Without caching, all three ancestors may also need to be checked.

It would reduce the duplication that we'll have between `lv_draw_rect_dsc_t` and the styles, since we could continue to pass regular styles around to lower level functions as usual. With the draw dsc method we would have to keep data in multiple places.

I agree, though; we'll need more objective data in order to decide.

kisvegabor, 2019-12-15:
@embeddedt 
First of all, thank you for the review 🙂 

I can't exactly imagine how caching would work. Do you mean to cache all the required properties from classes of an object and the ancestors to the local style (or a separate `cache_style`) of the object, and keep passing that style to the draw function?

If so we can automatically cache the used properties in  `lv_obj_get_style_...()`. If a style (or class) changes just invalidate the cache and let the  `lv_obj_get_style_...()` function load the properties again.

There is a case where I see `draw_dsc` reasonable: in some design function, the original style is hacked before drawing. For example, `lv_page` draws the background without border first and draw only the border in `LV_DESIGN_DRAW_POST` (over the scrollable area to make the page always "closed"). Or in some cases, "handcrafted" styles are used to draw something special. 
It seems complicated to add these tricks to these complex styles. And ultimately either before the call of the draw function or in the draw function the style needs to be "translated" to plain parameters. Then why not do it before calling the draw function? It gives better separation between the layers and therefore the draw functions can be used in a more flexible way. What do you think?

embeddedt, 2019-12-15:
I mean that we should generate a "classic" `lv_style_t` from the new system, and then handle that in the lower layers as usual. Essentially, we should keep all internal handling of styles the same, and only change how the user deals with it.

kisvegabor, 2019-12-16:
> I mean that we should generate a "classic" lv_style_t from the new system, and then handle that in the lower layers as usual.

Got it, thank for the explanation. 

To summarize, there are still three unclear points for me:

**1. The mentioned style caching and the API of drawing functions**
I think it's not critical right now but good to keep in mind there are questions here.

**2. Announce style change**
Reimplement the previously existing mechanisms. Such as report style changes, notify object on style change, etc.

**3. Find API to set object types style**
The current `lv_bar_set/get_style()` will be more complicated. The following functions would be required:
- `lv_bar_set_style_opa/value/color/...(bar, LV_BAR_STYLE_INDICATOR, LV_STYLE_COLOR_BG, LV_COLOR_RED)`
- `lv_bar_get_style_opa/value/color/...(bar, LV_BAR_STYLE_INDICATOR)`
- `lv_bar_add_style_class(bar, &style)`
- `lv_bar_remove_style_class(bar, &style)`
- `lv_bar_style_class_next(bar, LV_BAR_STYLE_INDICATOR, prev_class)` to list the classes.

I'm thinking about making these functions general like:
- `lv_obj_set_style_opa/value/color/...`
- `lv_obj_add_style_style_class()`
- `lv_obj_remove_style_class()`
- `lv_obj_style_class_next()`

And add functions to object types like:
- `lv_bar_get_style(bar, LV_BAR_STYLE_INDICATOR)`

And it can be used like:
```
//Get a pointer where the local style + classes are available
lv_obj_style_t * s = v_bar_get_style(bar, LV_BAR_STYLE_INDICATOR);

// Set `s` style of the bar. `LV_SIGNAL_STYLE_CHG` will be sent to `bar`
lv_obj_set_style_color(bar, s, LV_STYLE_COLOR_BG, LV_COLOR_RED)
```
What do think about it? Or do you know a better approach?

embeddedt, 2019-12-16:
> The current `lv_bar_set/get_style()` will be more complicated. 

CSS uses pseudo-elements for this, where styles can be applied to the pseudo-elements of an object.

```css
.my-bar::indicator {
    background-color: red;
}
```

I find this approach pretty clean and it would be ideal if we could duplicate its ease-of-use. I think the approach you've suggested is probably the simplest.

puzrin, 2019-12-17:
@embeddedt 

IMO CSS pseudo-elements are for DOM hacking - to inject ADDITIONAL objects via styles. That's certain not what you talk about.

You described default elements styles feature (set default styles for `body`, `a`, `p` and other tags). Note, applying styles directly is considered a bad practice in themes (the same about assign styles by element `id`). Good practice is to assign class name (at least one) OR combination of class names for complex cases, WITHOUT tag name.

For non-trivial projects i'd recommend to read [BEM methodology](https://en.bem.info/methodology/css/). Skip tools description, and pay attention only to naming approach. In short - instead of use nested classes, use convention to apply long-named classes directly to each element.

PS. Instead of separate implementation of "tag" styling, consider each tag has some default class name (example: `.label_default`). 

@kisvegabor 

> **1. The mentioned style caching and the API of drawing functions**
> I think it's not critical right now but good to keep in mind there are questions here.

I'd say everything else is almost trivial :). Main problem with classes is reverse lookup order of combinations:

```css
.container .block {
  __styles__
}
```

because there can be unknown amount of nested childs between `.container` and `.block` (container -> nested1 -> nested2 -> block). BEM allows avoid (or minimize) such cases and use merge mostly (elements are described by multiple direct classes).

Another alternative (not BEM):

```css
// this requires .block to be direct child of .container - scan is cheap, only 1 element up.
.container > .block {
  __styles__
}
```

Personally, i encourage move to CSS direction (not full implementation, but minimize divergence of supported features). Near questions are:

- How to assign multiple classes to single element (= now "widget").
- How to restrict nested classes supported.
- Estimate memory use.
  - Also check if GC needed when dynamic pages created/destroyed
- In C enums are used wide for nice names. Good for classes, but not clear now to generate global unique values by macros. Or is it acceptable to drop const support and generate class ids in runtime?
- Revisit "should support" values (+ types) and scaling factors. CSS use floats (sizes, offcets, opacity, ...) and texts. We need int & enums.
- Define madness level for layout composer. In modern HTML "tables" are bad idea for pages organize. Recommended approach are "flex" and "grid". Tables should be used only to display real table data.

kisvegabor, 2019-12-17:
@embeddedt 

> CSS uses pseudo-elements for this, where styles can be applied to the pseudo-elements of an object.

I haven't realized this similarity with the pseudo-elements. However, CSS doesn't support adding classes to pseudo-elements, right? E.g. adding `.bar_indicator_red` to `::indicator`.

> I think the approach you've suggested is probably the simplest.

Finally, I found and tried an even cleaner approach. 

Now the main style type is `lv_style_dsc_t` which contains a local style and a list to classes.
Every internal style (e.g. `LV_BAR_STYLE_INDIC`) is an `lv_style_dsc_t` so they have local properties and classes.

Similarly to `lv_obj_set_size` (which can be applied to any object type) I've created functions like: 
```c
lv_obj_add_style_class(bar, LV_BAR_STYLE_INDIC, &style1);
lv_obj_set_style_colors(btn, LV_BTN_STYLE_REL, LV_STYLE_COLOR_BG, LV_COLOR_RED)
lv_style_dsc_t * style_dsc = lv_obj_get_style(slider, LV_SLIDER_STYLE_KNOB);
```

I'm still not happy with the names of these 2 types:
- `lv_style_t` simple list of properties (rename to `lv_style_props_t`?)
- `lv_style_dsc_t` a local `lv_style_t` and linked list built from `lv_style_t *` pointers. (rename to `lv_style_t`?)

kisvegabor, 2019-12-17:
@puzrin 

> I'd say everything else is almost trivial :). Main problem with classes is reverse lookup order of combinations:

Actually it's not that difficult. I store classes in a single linked list and try to find the required property from the first class.

> How to assign multiple classes to single element (= now "widget").

See the code snippet above. You can create styles (`lv_style_t`) with the required properties and add their pointer to an object's class list

> How to restrict inheritance & nested classes support.

You can see the list of properties [here](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.h#L78-L103). (not complete yet). By default, the properties can be inherited from the parent but e.g. the entry for `FONT` will look like this:
```c
    LV_STYLE_PROP_INIT(LV_STYLE_FONT,  0x80, LV_STYLE_ATTR_TYPE_PTR | LV_STYLE_ATTR_INHERIT),
```

The "inherit" property is taken into account [here](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_obj.c#L2676).

> Estimate memory use.

**Non styled object**
So far `lv_obj_t` had an `lv_style_t *` which is usually 4 bytes.
Now we have a more complex [lv_style_dsc_t](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.h#L118-L121) which is 12 bytes (on 32 bit MCU). So for a non-styled object we have +8 bytes in `v7`.

**Use 1 style**
Not counting the cost to store the style itself (because it's reused by many objects):
- In `v6` it's still 4 bytes because only a pointer needs to be set.
- In `v7` every added class needs another 12 bytes to store the pointers of classes.  So 12 + 12 x class_num = 24 bytes for one syle.

**Change object's style**
In `v6` an other static `lv_style_t` is required which is ~80 bytes and the 4 byte pointer in the object. So 84 bytes in total to change the object's style. If an other object also needs a custom style then it's also 84 bytes. So 84 bytes/custom styled object 

In `v7` 
- The local style properties are stored like: 2 byte header + 1/2/4 bytes value. Let's say 4 bytes in total on average. There is an initial 4 byte cost when the first property is allocated. So 4 + 4 x property_num. E.g. (1 class) + (3 local property) = (12 + 12 x 1) + (4 + 4 x 3) = 40 bytes (compared to 84 in `v6`)
-  To store 2 class in the object = 12 + 12 x 2 = 36 bytes. 

> In C enums are used wide for nice names. Good for classes, but not clear now to generate global unique values by macros. Or is it acceptable to drop const support and generate class ids in runtime?

Classes are pointers:
```c
lv_style_t class1;
..set properties...

lv_obj_add_style_class(obj, LV_OBJ_STYLE_MAIN, &class1); 
```
> Revisit "should support" values (+ types) and scaling factors. CSS use floats (sizes, offsets, opacity, ...) and texts. We need int & enums.

I've created more types to save RAM
- opacity: 1 byte
- value: int16_t
- pointer: 4 bytes (usually)
- color: depends on `LV_COLOR_DEPTH`
- flags: 1 byte (still considering)

> Define madness level for layout composer. In modern HTML "tables" are bad idea for pages organize. Recommended approach are "flex" and "grid". Tables should be used only to display real table data. Not sure if needed here at all.

Somewhat similar is already supported by the  [layout and fit](https://docs.littlevgl.com/en/html/object-types/cont.html#layout) features

puzrin, 2019-12-17:
> I haven't realized this similarity with the pseudo-elements. However, CSS doesn't support adding classes to pseudo-elements, right? E.g. adding `.bar_indicator_red` to `::indicator`.

I guess here is some confusion due terms misuse. There are pseudo elements, and pseudo-classes. Pseudo-elements are `::before` and `::after` - really add dom content. Pseudo classes are used to reflect states (focus, hover) and define complex selectors ("first word of paragraph")

- https://developer.mozilla.org/en-US/docs/Web/CSS/Pseudo-classes
- https://developer.mozilla.org/en-US/docs/Web/CSS/::before

Both allows assign CSS classes/styles. And IMO both are not needed in lvgl next 5 years :)

> Actually it's not that difficult. I store classes in a single linked list and try to find the required property from the first class.

side note: not from first, but from last class, if you plan be close to CSS.

As far as i understand, that's not enougth. Because missed properties can be inherited from any upper parent. You will have to scan parents too.

With cache, "problem" will be different

- should uid (cache entry) be asigned to element or to each property? (opacity and width can come from different nesting levels).
- uid search is not free.

kisvegabor, 2019-12-17:
> I guess here is some confusion due terms misuse. There are pseudo elements, and pseudo-classes. Pseudo-elements are ::before and ::after - really add dom content. Pseudo classes are used to reflect states (focus, hover) and define complex selectors ("first word of paragraph")

Ahh, thanks for the hint. It just came at the right time. I'm dealing with the buttons now where there are 5 styles for the 5  state. If each has 1 class then it needs 24 x 5 = 120 byte. It's way too much.

If we had the concept of pseudo-classes then 
- we could store in the classes that which state it refers
- when searching the properties the object's state can be taken into account

But what states we have in general?
- normal
- pressed (for any object?)
- toggled (for button, switch, checkbox)
- toggled pressed (for button switch, checkbox)
- disabled/read-only (for any object?)
- ideas? :slightly_smiling_face: 

For different parts of an object (pseudo-elements), we can still have separate styles for each part (e.g. in case of button matrix: bg + buttons)

puzrin, 2019-12-17:
I can not say exact, what is needed for lvgl - only you can decide how to spend your personal time. For my current needs existing features are "a bit unusual but enough". But i can share my CSS/JS/architect experience, if that can help.

AFAIK, pseudo-classes, reflecting interface states, are ~ the same as generic. The only difference is, those are added automatically, instead of manual mode.

Imagine, you have:

```html
<div class="foo">
```

and manually add "foo_focus":

```html
<div class="foo foo_focus">
```

Via pseudo-classes this can be simplified:

```css
.foo {
  font-size: 20px;
  color: black;
}

.foo:focus {
  color: red;
}
```

I think, for states, this can be solved buy special flags in class struct - those will say, if class should be counted or skipped in concrete scan.

Examples: if element not focused, then all `:focused` classes still exists in your list, but should be ignored.

> I'm dealing with the buttons now where there are 5 styles for the 5 state. If each has 1 class then it needs 24 x 5 = 120 byte. It's way too much.

As far as i understand, any class should exist only when in contains some data and data should consists of only required changes.

If you don't need `:disabled` display, it should not take any space at all.

Also, instead of store all possible final states at once, you can reset cache for updated elements and calculate result again.

Or may be i don't understand your problem.

> But what states we have in general?
> 
>  * normal
>  * pressed (for any object?)
>  * toggled (for button, switch, checkbox)
>  * toggled pressed (for button switch, checkbox)
>  * disabled/read-only (for any object?)
>  * ideas?

I'm 100% sure all possible states already exists and battle tested in CSS. I would use names from this page https://developer.mozilla.org/en-US/docs/Web/CSS/Pseudo-classes and avoid any possible deviations.

I mean toggled & toggled_pressed => `:active` and so on.

kisvegabor, 2019-12-17:
> I think, for states, this can be solved by special flags in class struct - those will say, if class should be counted or skipped in concrete scan.

I was thinking in the same way. 

> As far as i understand, any class should exist only when in contains updated properties.

However, a button should have at least a minimal "buttonish" look by default so it needs a default classes for each state. Similarly to `<button>` in HTML. 

With `pseudo-class`es the button still needs 12 + 5x12 = 72 byte RAM. Better but still a lot.
If only normal and pressed classes were added by default then it'd be only 36 bytes which sounds much better. And the user can set toggled, toggle pressed, inactive if he needs. (defaults are provided but not applied automatically)

Or another approach:
It needs so much RAM because it takes 12 bytes to add a class. However, it seems possible to handle states different from CSS. In CSS a class (or a collection of properties) is assigned to a state. However, we could set the state on the property level too. E.g.
```css
.my_btn {
  color:red;
  color.blue!pressed; 
}
``` 
This way only one class would be required for a button which takes 24 bytes. In `v6` 20 bytes were required in button for the 5 styles for the 5 states. And the 5 styles used 400 bytes, while this "class" would require ~100 bytes (depending on the number of properties).

@puzrin  I know you are not a big fan of doing things differently then battle-tested solutions, but I see clear advantages here.

 > I'm 100% sure all possible states already exists and battle tested in CSS. I would use names from this page https://developer.mozilla.org/en-US/docs/Web/CSS/Pseudo-classes and avoid any possible deviations.

Sure, we use the same or at least similar names where applicable.

embeddedt, 2019-12-17:
> In CSS a class (or a collection of properties) is assigned to a state. However, we could set the state on the property level too.

I wish CSS had this feature - I think this is a great solution, since most of the properties likely won't be changing.

> I store classes in a single linked list and try to find the required property from the first class.

I think we need to search in reverse order (i.e. the last class needs to be the first member in the linked list), because we would want classes added later on to override classes added earlier.

> For cache - instead of store all possible final states at once, you can reset cache for updated elements and calculate result again.

Exactly.

puzrin, 2019-12-17:
I don't understand problem with CSS-like classes.

- Each element should contain list of class refs. That's only 4 bytes per class.
- Button usually needs 3 states - :normal, :focus, :active => 12 bytes per each button + class data (once)
- Classes are defined once per multiple elements

> ```css
> .my_btn {
>   color:red;
>   color.blue!pressed; 
> }
> ```

This will produce a lot of garbage for combined states. For example `a:focus:active:visited`.

If you care so much about class list size, you can organize it as separate abstraction layer and share between multiple elements. And this optimization may be transparent for user. But IMO that's overkill.

> @puzrin I know you are not a big fan of doing things differently then battle-tested solutions, but I see clear advantages here.

I have reasons for this :). Because i know costs of risks, when someone tries to reinvent something new. Reusing known solutions is more predictable. Usually it's possible apply existing solutions in non-destructive way. Seems i never suggested blindly follow some abstract ideal patterns, and always provided concrete examples of reuse :)

kisvegabor, 2019-12-17:
@embeddedt 

> I think we need to search in reverse order (i.e. the last class needs to be the first member in the linked list), because we would want classes added later on to override classes added earlier.

I do this way. I insert the new classes to the head, so they will be seen first.

> I wish CSS had this feature - I think this is a great solution, since most of the properties likely won't be changing.

Great! :slightly_smiling_face: 

@puzrin 
> Each element should contain list of class refs. That's only 4 bytes per class.

Actually it's 12 byte/call reference:
- 4 byte for the pointer to the class's style
- 4 byte for a pointer to the next class
- 4 byte because allocation has 4 byte header and the new node is allocated.

However, I've just realized a simple dynamically array can be used as well.  An then the size would be:
- 4 byte to store a pointer to the array
- 4 byte overhead for the allocated array
- 4 byte/class

So 1 class = 12 bytes; 2 class = 16 bytes; 5 classes  = 28 bytes.

puzrin, 2019-12-17:
@kisvegabor thanks for explanations about memory.

There is possibility to use 4 bytes only in transparent way:

- Create "classes list manager", where each entry is a unique combination of classes. Number or records will be limited.
- When user assign new class - search existing combinations. If exists - reuse reference. If not exist - create new record.

Of cause, some care should be spent to reference counters, GC and so on. But general principle works and "read" access is fast. "write" is slower, but should not be critical for our needs.

kisvegabor, 2019-12-19:
@puzrin 
Interesting idea! It's quite complex but really very memory effective. 
Fortunately, the class list handling is quite transparent so we improve it later too.

---

I made some progress with states:
- I defined [object state](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_obj.h#L207-L214) and [style state](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.h#L118-L125)
- To set the state of a style property:
`lv_obj_set_style_value(btn, LV_STYLE_BORDER_WIDTH | LV_STYLE_STATE_PRESSED, 2);`
- The order of states is not arbitrary. A higher value means higher precedence. E.g. on a pressed + hovered button the pressed style will be (shown if it's defined).
- The normal state has the lowest precedence.
- Combinations of states also can have styles (AFAIK it's not possible or not trivial in CSS). E.g:
  - hover: blue
  - pressed: dark blue 
  - hover + pressed: light blue
- The algorithm which decides which value to use, if there are multiple states for property, works like this:
   - only those properties matter which state is related to the current object state. E.g. `obj_state = HOVER+PRESSED`, then styles with `HOVER`, `PRESSED`, `HOVER+PRESSED`, `NORMAL` states are ok, but `HOVER+CHECKED` don't.
  -  from the matching values choose that where the most state matches. Match in higher precedence states has a higher weight. In other words, choose the value where `state_to_get & state_of_property`  is the greatest. 

It might sound complicated but it's quite intuitive in the practice. An examples:
- These styles are applied to a button (I've renamed `toggled` to `checked`):
```
1. BG_COLOR = RED
2. BG_COLOR | PRESSED = DRARK_RED
3. BORDER_WIDTH | CHECKED = 3
4. BG_COLOR | CHECKED | PRESSED = LIGHT_RED
```
- If the button is released (normal state) it will be red, 0 px border (1. matched in BG_COLOR and default 0 px border-width)
- If the button is pressed is will be dark red, 0 px border (1, 2 matched in BG_COLOR, but 2 has higher precedence)
- If the button is checked (toggled) released it will be red, 3 px border (1. matched in BG_COLOR, 3. matched in border-width)
- If the button is checked (toggled) pressed it will be light red, 3 px border (1,2,4 matches in BG_COLOR but 4 has higher precedence, 3. matched in border-width)

[image]

As you could see I've add hover, focus and edit to states too. Hover is still not supported but it could be later so better be prepared.
Focus and edit are related to keypad or encoder navigation (`lv_group`) but by making them generic states they can be used with touchpad too. (The clicked object becomes focused.) E.g. The selected (focused) text area can have a blue shadow which disappears when a new object is selected.
I see some ways to greatly simplify the current "group" system and make it more generic and easier to use. But there are some unclear points here.

Thank you if you've read this through :slightly_smiling_face:

puzrin, 2019-12-19:
> Combinations of states also can have styles (AFAIK it's not possible or not trivial in CSS)

As i said before, everything for this task kinds exists in CSS spec. Simple check is: "if you think something not exists in CSS => something goes wrong, try to search docs again" :). Seriously.

You can combine pseudo classes in CSS as much as you need::

```css
.foo:focus:hover {
  ...
}
```

Also, i'm not sure i understand 100% right your algorythm, so, general note is: please recheck docs how CSS selector priorities calculated. It ~ depends on path length + order of declaration. It's important to create "reduced" implementation, not "completely new" (because more common to users).

For lvgl's simplified case, this rule could be enougth:

- child value overrides parent value
- if child has multiple declarations, LAST has higher priority (overrides previous value)

Of cause, if you support priorities, based on selector path length (number of matched states), that will be nice. But that's not mandatory.

Kick me, is you have any questions about CSS or need help how to shrink it without breaking "css spirit".

kisvegabor, 2019-12-19:
> You can combine pseudo classes in CSS as much as you need::

I didn't know about it, thanks!

> child value overrides parent value
> if child has multiple declarations, LAST has higher priority (overrides previous value)

The's really the super minimal requirement. 

> Also, i'm not sure i understand 100% right your algorythm, so, general note is: please recheck docs how CSS selector priorities calculated. It ~ depends on path length + order of declaration. It's important to create "reduced" implementation, not "completely new" (because more common to users).

I've checked it and I've created this (simplified) algorithm based on that. I've replaced "order of declaration" with "fix precedence" though.

> Of cause, if you support priorities, based on selector path length (number of matched states), that will be nice. But that's not mandatory.

A more simple version would be to match only one state (no combinations) but combined states mean only a minimal overhead in searches but give more flexibility.

puzrin, 2019-12-19:
> I've checked it and I've created this (simplified) algorithm based on that. I've replaced "order of declaration" with "fix precedence" though.

Well, formally we can say "you preferred to drop declaration order weight". That's logically correct. But this combination of features will be very inconvenient for users (IMO).

> A more simple version would be to match only one state (no combinations) but combined states mean only a minimal overhead in searches but give more flexibility.

I mean a bit different. If we wish to make something useable, then combinations MUST be supported. But length-based priorities can be ignored (use declaration-order only)

In CSS:

```css
a:focus:hover {
  color: red;
}

a:focus {
  color: orange;
}
```

link with focus + hover will be always red, independent on declaration order. Because path is more long. But if you drop that in lvgl and leave order-only priority - that will be "good enougth".

kisvegabor, 2019-12-19:
> But if you drop that in lvgl and leave order-only priority - that will be "good enougth".

With "order only" the "focus+hover" link should be still red, shouldn't it? Because it's the style that matches the exact state so it should be preferred over styles that describe less from the given state.

puzrin, 2019-12-19:
Imagine, you have:

```css
.style1 {
  color red;
}

.style2 {
  color: orange
}
```

And set both styles to link. Then `style2` color will be used (apply style1 first, then apply style2). For CSS order = sequence in stylesheets. In LVGL that will be "order or adding class to object".

- "order-based" priority should exist anyway (IMO), for equal "paths".
- for pseudo-classes (path extention) you can implement one more rule OR do nothing.

If you select "do nothing", users will have to describe rules in "natural" way - "from generic to detailed". That's not ideal, but acceptable.

That does not mean something is better or worse. I only describe how to split logic to independent parts and allow progressive implementation, depending on available time & cost.

I have nothing against states priorities. I say if you wish those, they should gently extend order-based rule, not replace it everywhere.

puzrin, 2019-12-19:
Probably i should reread about CSS priorities too :). In CSS you may have:

```css
.parent .child {
}

.parent .child:focus {
}

.parent:focus .child {
}
```

And i don't remember spec priorities for (2) & (3).

kisvegabor, 2019-12-26:
Sorry for the late answer.
> I have nothing against states priorities. I say if you wish those, they should gently extend order-based rule, not replace it everywhere.

Great!

> And i don't remember spec priorities for (2) & (3).

(2) and (3) have the [same specificity](https://specificity.keegan.st/) so the order should matter.

---

In the past days I've dealt with important things:
1. Implement caching (WIP). I got an idea to cache only the [typical values](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.h#L165-L203) in order to save RAM. For example cache the "BG_OPA" on 2 bits:
- COVER
- TRANSP
- OTHER
Or store the "BORDER WIDTH" on 3 bit.
It has the advantage that the cache data is only 12 byte and the properties can be read from the cache most of the time, however, sometimes the property still needs to get normally.
2. I've defines this [built-in-calsses](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.c#L49-L67)

Please let me know what you think.

puzrin, 2019-12-26:
To optimize cache speed/memory, need to know critical chains of rendering process. First rule of any cache is "do we really need to implement it" :).

As far as i understand, inspected time period should be single frame render. It consists of:

- Multiple passes (defined by framebuffer size)
- Search involved objects for each screen region (need to know absolute coords of each object's bounding box)
- Draw filtered objects into screen region. For each object (ordered by Z-level):
  - calculate properties
  - compose to framebuffer

That's very simplified and naive, only to understand candidates for optimization. Fix me if something is deeply wrong. Second rule of cache is - "avoid invalidation dependencies". Let's consider cache valid for 1 frame render only (build it on the fly and drop in the end). If speed will be good enougth - we are lucky. If not - this will be implemented anyway, as part of more complex system (no full rewrite and time loss).

(1) Bounding box coords - first candidate to "cache" (probably can be private part of each object)

To calculate coords (and other properties), we should traverse all parents up to root and merge classes. This should be optimized before cache add (optimized code can be good without cache at all).

For each property, we have to:

- scan parents up and build list to traverse down
- traverse down and merge classes

Here are several possibilities of optimization:

- keep full path for each end node, instead of scan parents up every time (optimize building traverse path)
  - probably, classes merge is much more expensive, skip this opt on first pass
- build multiple properties in parallel (required props can be defined by bitmask)
- optimize class data pickup from class
  - use bitmap to mark if prop exists at all
  - use table for frequent/all properties instead of list

I don't know all details. This are only preliminary thoughts about implementation possibilities. Very simplified, without more complex memory-expensive approaches.

---

>     1. I've defines this [built-in-calsses](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.c#L49-L67) [](/littlevgl/lvgl/blob/HEAD@{2019-12-26T00:40:45Z}/src/lv_core/lv_style.c#L49-L67)

- I'm a bit confused about using term "style". It look like "theme" or "class" in code sometime.
- Modifiers (red, green, ..) looks a bit random. I can't understand logic, why some exists and other skipped. If that's for example only - then no questions.

In general - looks convenient, i like it.

kisvegabor, 2019-12-26:
@puzrin  Thank you for the suggestions!

> As far as i understand, inspected time period should be single frame render. It consists of ...

Exactly!

> Let's consider cache valid for 1 frame render only (build it on the fly and drop in the end).

Style properties are mainly used when drawing happens. So if we'd think in 1 frame lifetime then caching wouldn't help too much. But usually not that's the case. Instead, you set a style/classes and use the UI: update chart, press buttons, type in a text area, move objects etc (which doesn't require style change). However, style can be animated too. Typically to change the opacity or color. In this case, the lifetime is really only ~1 frame.

> keep full path for each end node, instead of scan parents up every time (optimize building traverse path)

Getting the parent is very cheap: `lv_obj_t * parent = obj->parent`. In addition, usually, there is no more than 5 "layers". So we should speed up somewhere else.

> build multiple properties in parallel (required props can be defined by bitmask)

I was also considering it, but haven't implemented yet. 

> use bitmap to mark if prop exists at all

I use `groups` for this purpose. There can be up to 16 property group and each corresponds to a bit in an `uint16_t` bitfield. [Here](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.h#L79-L122) the second parameter is the group of the property. It allows to quickly skip a class or local style if it has no related properties. 

> use table for frequent/all properties instead of list

Now I build an array from elements like this:
- 1 byte property ID: id + group
- 1 byte state
- N (1/2/4) byte data depending on the type of property 

How did you imagine a "list" which can provide faster access to the elements?

> I'm a bit confused about using term "style". It look like "theme" or "class" in code sometime.

I'm not sure about the terms yet. They should be classes (e.g. `lv_class_red`) but the "class" term might not be obvious in this context.

> Modifiers (red, green, ..) looks a bit random. I can't understand logic, why some exists and other skipped. If that's for example only - then no questions.

For the colors the logic was to keep the number of classes minimal but cover the followings:
- danger/error: red
- ok/apply: green
- "normal" color: blue
- light/dark: quite obvious to have

Do you miss any other colors?

---

So far I developed on PC but I'll try it on an STM board soon to see the performance effect of the current caching and the groups. I've already integrated labels and buttons so I can test with a more or less typical UI.

puzrin, 2019-12-26:
> So if we'd think in 1 frame lifetime then caching wouldn't help too much.

I have no benchmarks to trust or not trust such statements. There can be different situations. For example, if cache in 1 frame is not effective, it may be not needed at all.

> However, style can be animated too. Typically to change the opacity or color. In this case, the lifetime is really only ~1 frame.

IMO it's the only real case when performance matters - animation should be smooth. It would be not rationale to design system more complicated than required. There is huge complexity difference between self-invalidated cache, and partial invalidation. I would not recommend to dive into complex invalidations until you have absolutely no alternatives.

To be more precise - webdevs have term "animation frame" (~20ms) - period when state is not changed (=> cache continue stay valid). In your simplified case, "single frame draw" would be more convenient & easy to implement.

> Now I build an array from elements like this:
>
> * 1 byte property ID: id + group
> * 1 byte state
> * N (1/2/4) byte data depending on the type of property
> 
> How did you imagine a "list" which can provide faster access to the elements?

I meaned, initialy list of properties is "map" data type. With appropriate search cost. For short case this can be implemented as list (cost = full scan). Or lookup table with unused cells (cost = memory). Plus some combined approaches, tending to binary trees.

Your choice with 2-dimentional addressing looks good. I don't think it can be improved without significant complexity increase.

> For the colors the logic was to keep the number of classes minimal but cover the followings:
> 
> * danger/error: red
> * ok/apply: green
> * "normal" color: blue
> * light/dark: quite obvious to have
> 
> Do you miss any other colors?

IMO that's to primitive to be useable out of helloworld demo. I'd suggest you to not hardcode such things into core. This could be moved to separate (optional? example?) file.

puzrin, 2019-12-26:
https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_core/lv_style.h#L79-L122

There are some props, not existing in CSS: `LV_STYLE_OPA_SCALE`, `LV_STYLE_PAD_INNER`, `LV_STYLE_SHADOW_OPA`, `LV_STYLE_BLEND_MODE`, `LV_STYLE_IMG_OPA` and others. I did not dived into details, only suggest to check if it's possible make things more close to CSS.

kisvegabor, 2019-12-27:
> IMO that's to primitive to be useable out of helloworld demo. I'd suggest you to not hardcode such things into core. This could be moved to separate (optional? example?) file.

Probably built-in style can be converted to a very simple light weighted theme. Themes will be reworked anyway, so there is room for new things here too. I was planning only 2 themes (light and dark) but "simple" (lightweighted) and "none" make sense too.

> There are some props, not existing in CSS: LV_STYLE_OPA_SCALE, LV_STYLE_PAD_INNER, LV_STYLE_SHADOW_OPA, LV_STYLE_BLEND_MODE, LV_STYLE_IMG_OPA and others. I did not dived into details, only suggest to check if it's possible make things more close to CSS.

In CSS you set RGBA colors, so opacity is included in the color parameters. In lvgl there are RGB colors so an additional OPA parameter is required. 
`PAD_INNER` is a primitive replacement of `margin` in CSS. 
`BLEND_MODE` [exists in CSS too](https://css-tricks.com/basics-css-blend-modes/)

---
I've finished the benchmark. It used 40 buttons with 1 label on each (80 objects). The buttons had autofit to set their size automatically to the label, and the buttons' parent had `LV_LAYOUT_PRETTY` 
to align the buttons in a grid. The autofit and layout makes the creation of the objects more complicated. I used 1/8 screen sized display buffer.

Initially, I tested 3 button designs: 
- plain (no radius, no gradient, no border)
- typical (radius, gradient, border)
- advanced (typical + 50% button opa, 50 % text opa, shadow)

The typical design looked like this:
[image]

The properties were stored in one class, so the buttons had 1 class and labels had 0 class.

- Having only one class the group bitfield had no effect (there was nothing to skip).
- With cache 3% performance loss compared to `dev-7.0` with original style system.
- Without cache 10% performance loss.

At this point, there were 12 properties in the button's class. I added **20 more properties** to have a really large (but still only one) class. In this case: 
- Without caching another 3% performance loss. (13% in total)
- With cache no change (as the relevant values were cached)
- Groups made no difference (still nothing to skip)

Finally, I added **4 dummy classes** (processed before the real button class) with 5 properties in each. It's 20 additional properties like in the previous case. 
- With cache, there was no change
- With no cache but groups enabled another 8 % performance loss. (18% compared to `dev-7.0`)
- With no cache, no groups 10% performance loss (20% in total). It was the only point when groups could help because the irrelevant classes could be skipped, however, it had only a negligible gain.

I've also measured the time to create the objects. With `dev-7.0` it was 34 ms.
- 1 class:
  - With cache ~150% slower. (57 ms)
  - Without cache ~135% slower. (46 ms)
- 5 class:
  - With cache ~300% slower. (93 ms)
  -  Without cache ~200% slower (64 ms)

If I disabled layout and autofit the creation time with 5 classes:
- Without cache ~3% slower (35 ms)
- With cache ~200% slower (75ms)

So in summary:
- With caching there is 3% performance loss in all conditions
- Cache needs 12 byte/object.
- Without caching there is 10-20% performance loss depending on the conditions. 
- It's faster to read one large class then multiple smaller (no surprise)
- Groups have no real benefit so can be removed
- Creation time is significantly larger (especially with caching), but still in a tolerable range. I haven't immersed in this part yet so probably there are some trivial things to optimize here.

puzrin, 2019-12-27:
> In CSS you set RGBA colors, so opacity is included in the color parameters. In lvgl there are RGB colors so an additional OPA parameter is required.

Hm... is it a principal problem to have RGBA in CSS? Many opacities with different names look strange.

You could provide RGBA in public API and extract alpha to separate property on classes merge.

kisvegabor, 2019-12-27:
> Hm... is it a principal problem to have RGBA in lvgl's "CSS"? Many opacities with different names look strange.

I see some advantages and disadvantages for both ways but let's postpone it a little until the more basic questions are clarified.

puzrin, 2019-12-28:
I'd say, cache performance is not impressive. IMO it worth drop it at all, or postpone until architecture become stable. I'm sure it's possible to find better candidates for optimizations, if needed. For example, implement "parallel" extract for properties.

kisvegabor, 2020-01-02:
As I say the cache performance is impressive because it makes the UI as fast as earlier. Instead, I'd say it's not too bad without cache too. :smile: But it's just playing with the words. 
I was thinking about parallel extracting properties but it's more complicated as it seemed at first look. All properties can have different weights in the styles. E.g. let's say there is:
- color:hover, color:hover:focus, color:focus
- radius, radius:active
It's difficult to keep track of which is the current candidate for the property because their highest fitting weight also needs to be known.  

In the past days,
- integrated half of the object types
- added style animations
- updated drawing routines
- added some new style features
   - color overlay: e.g. add 50% orange to all colors.Useful to show focused/edited state. CSS has "only" hue shift, which is more difficult to use IMO.
  - pattern: and image/symbol drawn on rectangles (center or repeat). It can be used to the add "pipe" to the checkbox in checked state or [skew lines to the bar](https://www.cssscript.com/animated-progress-bar-component-with-pure-css/)

puzrin, 2020-01-02:
> I was thinking about parallel extracting properties but it's more complicated as it seemed at first look. All properties can have different weights in the styles.

1. Calculate weight as (node_level * 2 + state_fix). That should work for new "simplified css"
2. Use bitmasks to check in one step which props available in concrete class (that was not effective for 1 prop, but may be ood for group of prips).

> color overlay: e.g. add 50% orange to all colors.Useful to show focused/edited state. CSS has "only" hue shift, which is more difficult to use IMO.

I could not understand what's wrong with CSS again :). Could you explain?

kisvegabor, 2020-01-02:
> Calculate weight as (node_level * 2 + state_fix). That should work for new "simplified css"

Can you explain it in more detail? Maybe an example?

> I could not understand what's wrong with CSS again :). Could you explain?

"Overlay" is similar to [filter](https://www.w3schools.com/cssref/css3_pr_filter.asp) in CSS but in a more simple way. No sepia, contrast, lightness, etc. just a simple recolor. 
Where can it be used:
- make the whole button darker (border, background, children, etc) when pressed with one property 
- add a new color to the whole focused object

So to answer your question, it a simplified (and slightly different) version of an existing CSS proeprty.

puzrin, 2020-01-02:
> Can you explain it in more detail? Maybe an example?

As far as i understand, we should implement this conditions:

- Child's prop always more important than parent's.
- Inside child, state has priority
- If no more difference - use declaration order.

So, we can

- Scan all nodes down from root to current
- Inside each, scan all classes (already ordered)
- For each class - take decision, if we should use it's props or not. Weight from parent does not matters because child is always more important.

So, in `node_level * 2 + state_fix` first part is to "make child more important", second - to make state inside same node more heavy (add 1 if state exists).

You can use 2 ways to merge "current level classes" to previously found (parents):

1. Merge only classes of current level (here weight will be only 0 or 1). Then merge (with override) to previously scanned (from parents).
2. Merge every class to previous result immediately. Then we should know "global weight" to decide if duplicated prop should be overrided or ignored.

Since we have to store "weight" anyway (2) is preferable.

> Where can it be used:
> 
> * make the whole button darker (border, background, children, etc) when pressed with one property
> * add a new color to the whole focused object

I suspect, in real world all solve this at preprocessor level. In your case, something like this:

```c
lv_css_add(class1, NORMAL,  BORDER, RED);
lv_css_add(class1, NORMAL,  COLOR, RED);

lv_css_add(class1, FOCUSED, BORDER, MACROS_DARKEN(RED, 30));
lv_css_add(class1, FOCUSED, COLOR, MACROS_ADD_ORANGE(RED, 70));
```

Because nobody writes CSS directly, all uses SASS, Stylus, LESS and so on.

`filter` is for very complicated non-discrete cases, for example if you wish to transform images on the fly (but it's a real madness, used only in demos). IMO it's not needed here. It's much more rare than CSS Animations (and those are not needed here too).

Also, you can write macros to define multiple props at once, with variants (for example, define button with selected color base).

In short - "states" are enough for recolor.

puzrin, 2020-01-03:
BTW, let's talk about animations design.

How do you see implementation of animations with new class system? In 6.x object contained all properties inside, and you could vary those as you need anytime. With classes object can have zero props (except coords cache) and significantly reduce memory consumption. Where animation will be applied to? Will user have to animate class property?

Also note, CSS spec describes "CSS animations" and "CSS transitions". Not sure they worth full (or partial) implementation. But it would be useful to resort existing animation demands, and check if those can be mapped to CSS-like approach.

At first glance, 99% of demanded effects are related with opacity dim & slide in/out (to show/hide menus and so on). This can be achieved via [CSS transitions](https://developer.mozilla.org/en-US/docs/Web/CSS/transition):

- Add class `.elementXX-in` to show
- Add class `.elementXX-out` to hide AND remove both on animation end (via callback?).

Explanation: we can run animation on class add, but can not do it on class remove (difficult to fetch previous state). So, to animate in reverse order, we should add one more class with "back" transition and then cleanup on the end. For simplified implementation you can skip cleanup, and if class added twice - just popup it to the end with state reset. In this case, classes in-out-in-out-in-out... can be added infinite, without extra memory loss.

---

If you find a time to explain disadvantages of having full RGBA in classes, i could search workarounds. I guess, in 6.x and earlier versions this was a memory problem, but classes should solve this issue. And moving color conversions to drawer/composer should not cause notable performance regression.

But may be i'm missing something important.

kisvegabor, 2020-01-03:
@puzrin  Thank you for the detailed explanation. However, I came across a question that should be understood before moving forward.

While I was integrating some object types and I realized that I don't fully understand how states should be inherited/propagated. I played with CSS to understand how `:active` and `:hover` are propagated. See this gif:
[image]
[index.html.zip](https://github.com/littlevgl/lvgl/files/4020533/index.html.zip)

The hover is clear: all objects under the cursor are hovered.
But active is quite counter-intuitive: all parents become activated? So all objects under the cursor get clicked?

Can you share your thought?

---

**Animations**
I implemented it like this:
- take a pointer to a style (local or class)
- take a "start" and "end" style
- set animation time and other attributes
- start the animation
- it will animate the attributes in "start" to "end" and apply them to the given style. lvgl will refresh all objects which have the given style pointer (as a class or local style)

What do you think?

---
**RGBA**
My concern is related to caching. If we can get rid of caching, this concern is over.
So the disadvantage I see that it's easy to cache the 2 most typical opacity value (cover, transparent -> 1 bit + 1 bit for other opacity). During rendering, it's required to get the opacity to decide which is the first object to draw. E.g. don't draw the wallpaper if only a button's text has changed, but draw only the button. But if the button's opacity != 100% then the wallpaper needs to be drawn too.
Or when getting the properties of the rectangle to draw, it's useful to get the opacity first, and if it's 0%  ignore the rest of the properties. 
If we'd combine colors with opacity, the color should be read all the time which can't be cached so nicely.

puzrin, 2020-01-03:
> The hover is clear: all objects under the cursor are hovered.
> But active is quite counter-intuitive: all parents become activated? So all objects under the cursor get clicked?
> 
> Can you share your thought?

AFAIK, in DOM all events bubble up (child -> parent -> next parent). Technically, you can add event handler and prohibit propagation, but that's considered a bad practice.

Note, events bubble not to "all under cursor", but to parents of top matched object under cursor. I say "matched", because you can define overlay, transparent for events (not sure if this is needed in LVGL). Since in most cases parents are located behind child, this looks like "all objects under cursor".

---

> **Animations**
> I implemented it like this:
>
> * take a pointer to a style (local or class)
> * take a "start" and "end" style
> * set animation time and other attributes
> * start the animation
> * it will [...]

Here i have 2 checkpoints, for critical things in data design:

1. What is "local style" - as far as i understand, in 6.0 this was a main source of memory consumption. Because each object contained too much data inside. Will 7.0 objects be able to exists without "local styles" at all?

2. Animations should keep state somewhere. If object does not have "local style", we need some way to override values, calculated from CSS, but keep object as light as possible.

It would be attractive to reuse the same classes merge principle for all cases, to save memory and keep optimizations simple. But problem is "classes" are shareable, and animations are always specific to object. If you define `.my_class:hover { transition: ... }` for multiple buttons, those should have independent animations.

So, in ideal world we CAN have shared animation description (CSS definition), but should split animation states, and calculate result effectively.

I think, simplified implementation could be good enough:

- animate only last object, where animation was activated (you can't focus 2 objects at once and don't need to animate 2 menus at once)
- if animation become a singleton in scope of class, we can do appropriate class storage extension, and properties extract will be transparent for merge algorithm. For example, add to each property flag "is_animated" to check overrides.
- if user still wish mad case like hundreds of independently animated buttons - he can create multiple unique classes. I think this case should not be reason to caкe about optimizations and make lvgl code more complicated.

That's very preliminary. Need to check if i missed use cases when parallel animation of the same class on multiple objects can be really useful.

PS. I know, currently your animations are independent of CSS, but 99% of demands are covered with "CSS transition". If interface designers could reuse their CSS experience in LVGL, that would be nice.

---

> **RGBA**
> My concern is related to caching. If we can get rid of caching, this concern is over.
> So the disadvantage I see that it's easy to cache the 2 most typical opacity value (cover, transparent [...]

~ understand. I'd like to clarify:

1. RGBA and caches are ortogonal in right data flow design. Those are NOT related anyhow. If they are - something goes wrong. Don't worry about single byte loss, this should not interfere to create optimal design in general. As example - "class list manager" i described (instead of count every bit, more optimal data flow suggested). 
2. It's difficult to believe, that users who wish background images will not use button shadows :). But you still can have accumulated flag "is_transparent" somewhere inside each object. That's internal data format, it should not affect external API.
   - I would check twice, if described opacity optimization will be really high demanded and worth efforts.
   - If problem is in slow decompression of image region - i'd suggest to use raw images OR split background to multiple images. In other words, if you have problem with image decompression - optimize image decompression instead of global hacks.
3. I think, in scope of "single frame draw" it could be useful to store bounding boxes in each object for faster clip region calculations in each pass. May be, with opacity. Key points are:
   - cache should be attached to object directly, to avoid search in HashMap by ID
   - to keep cache simple, it should stay valid only in scope of single frame draw.
   - i'm not sure if cache needed, but if yes, it's possible to cache only merged parent classes (path to current node) instead of everything. When you traverse objects tree, you will need to add only last node to cache for final result. That will make extraction fast enough and keep cache small (because you keep only active path instead of everything).

Summary: if opacity cache is the only principle problem about RGBA, and this cache can not be dropped at all, i think it's possible to implement it good enougth.

kisvegabor, 2020-01-04:
> AFAIK, in DOM all events bubble up (child -> parent -> next parent). Technically, you can add event handler and prohibit propagation, but that's considered a bad practice.

My problem was not about event's but about states. ("Where does the hover/focus/active states are propagated?" ) However, it seems what you described with events can be applied to states too. 
I've just noticed that it does not go only to the parent.  **Seemingly `<p>` gets the state of the direct parent.**

> What is "local style" - as far as i understand, in 6.0 this was a main source of memory consumption. Because each object contained too much data inside. Will 7.0 objects be able to exists without "local styles" at all?

The local style is the same as `style=` in HTML. It allows a quick and dirty way to change something locally. It was one of my goals to allow this simple, unique style modification. Maybe it can be dropped? Regarding the memory usage: every object needs a 4-byte pointer to store the possible local properties.

@embeddedt What do you think? Are the local style properties are required?

---

> Animations should keep state somewhere. If object does not have "local style", we need some way to override values, calculated from CSS, but keep object as light as possible.

Now I keep the animation state in the animation descriptor (count from 0..256 and proportionally apply the values). 

So the next question is: drop class animations, which affects multiple objects? If so I think the cleanest way would be to 
- tell which object to animate 
- set start/end styles, and temporally add an "animation class" where the new properties are applied
- If the object has "end" class earlier in the list (from a previous animation, see below) remove it
- at the end of the animation add "end" class.

How does it sound?

---

> RGBA and caches are ortogonal in right data flow design. Those are NOT related anyhow. If they are - something goes wrong. Don't worry about single byte loss, this should not interfere to create [...]

Okay, let's see what we can do.

> If problem is in slow decompression of image region - i'd suggest to use raw images OR split background to multiple images. In other words, if you have problem with image decompression - optimize image decompression instead of global hacks.

It's not related to images. It's about not to draw what will be overdrawn. A more extreme example when you have 3 screen-sized layers, then it's pointless to draw the 2 lower. (expect the upper has opacity != 100%)

> I think, in scope of "single frame draw" it could be useful to store bounding boxes in each object for faster clip region calculations in each pass

It's already done. Every object stores its absolute coordinates.

> cache should be attached to object direc7z6tly, to avoid search in HashMap by ID

It also works like this. If an object has pseudo-classes (e.g. slider indicator) it also stores its own cached values.

> to keep cache simple, it should stay valid only in scope of single frame draw.

Why? Once it's calculated, why not keep it? It can be invalidated when the object's style changes.

> i'm not sure if cache needed, but if yes, it's possible to cache only merged parent classes (path to current node) instead of everything. 

Do you mean to build a list in a child like: `class_gradparent1, class_grandparent2, class_parent1, class_child1, class_child2`? 

Note that, traversing the tree is fast because the objects store a pointer to their parent. And the parents directly store their children in a linked list, so it's also fast to iterate the children.

Firstly the parent's properties are cached, so it's possible to read the cached values from the parent when the child's cache is built. (now I don't use the parent's cache but it seems possible)

puzrin, 2020-01-04:
> I've just noticed that it does not go only to the parent. **Seemingly `<p>` gets the state of the direct parent.**

That sounds strange. State changes and events are related directly, and events propagate up only, not down.

In CSS you can create "chained" rule:

```
.parent_class:focus p {
  color: red;
}
```

Then `p` color will really be updated on parent's focus. But that's a different story. We don't need that.

> The local style is the same as `style=` in HTML. It allows a quick and dirty way to change something locally. It was one of my goals to allow this simple, unique style modification. Maybe it can be dropped?

Got it. I'd say, in HTML writing `style` attr directly is bad practice. Normally such things used only in JS-driven animations, to set intermediate values. Script set property directly and debugger displays it via style or virtual class (don't remeber). I asked "what is local style" to understand is it done specially for animation, or that's a tail from previous version.

I think, local values will be implemented somehow anyway, in explicit or implicit form. Question is "can those be unified/simplified via classes or not". This is potential place for core simplification.

> It's not related to images. It's about not to draw what will be overdrawn. A more extreme example when you have 3 screen-sized layers, then it's pointless to draw the 2 lower. (expect the upper has opacity != 100%)

I understand. I try to explain 2 statements:

- Effect in real cases does not seems to be significant. For example, if user click button, you don't need to draw it's state with 50FPS.
- Approach looks hacky by design. Expectation "object is not transparent and none of pixel change transparency after state update" is rare in "modern world". May be for OLED-s, but who cares...

Nobody cares how many layers you draw if that's fast. But lvgl code simplicity is important.

> > to keep cache simple, it should stay valid only in scope of single frame draw.
> 
> Why? Once it's calculated, why not keep it? It can be invalidated when the object's style changes.

- Because correct invalidation, properly working in all possible cases is most complicated thing of cache design.
- Because complex invalidations tend to cause tight-coupled code and architecture mistakes (people start rely too much on "magical" cache instead of right data/logic design)

I suggested conditions when cache will not cause harm, but still have chance to help. Though, it would be better to stay without cache at all, if performance is suitable.

> Do you mean to build a list in a child like: `class_gradparent1, class_grandparent2, class_parent1, class_child1, class_child2`?

I mean, when you calculate classes, you traverse parents from the root and merge data. If parent has multiple children you can remember merged result for parents instead of calculate all from the begining for each child.

That usually works, because nodes are drawn in tree traverse order.

embeddedt, 2020-01-04:
> What do you think? Are the local style properties are required?

Some method to easily overwrite class properties without needing to make a new class is necessary, in my opinion. It's useful in the event that, for example, someone wants to dynamically change the color of a label without having to add or remove classes. I do something similar on React websites frequently.

kisvegabor, 2020-01-04:
@puzrin 

> That sounds strange. State changes and events are related directly, and events propagate up only, not down.

I've realized that I misunderstood how properties are inherited. I thought this "conversation" happens:
\- `p`: hey, parent `div`, I'm in hovered state. Can you tell me a hovered text color?

But instead:
\- `p`: hey, parent `div`, can you tell me a text color?
\- `div`: sure, I'm in focused state, so I tell you the focused text color. Here it is.

So the objects tell the properties for their state, not for the state of the asking object.

It explains why was the `p` element changes color.

> Approach looks hacky by design. Expectation "object is not transparent and none of pixel change transparency after state update" is rare in "modern world". May be for OLED-s, but who cares...

Imagine a fast-updating chart that is on the 3rd upper layer. Maybe there are complex lists, other charts, running animations, or who know what on the lower layers. Drawing all of them would make the upper chart slower. So it's really important to start from the very top object to skip the not visible parts. 

> Nobody cares how many layers you draw if that's fast. But lvgl code simplicity is important.

"Fast" depends on the MCU. "Nothing matters" at 500 MHz, but everything matters at 48MHz.

> I suggested conditions when cache will not cause harm, but still have chance to help. Though, it would be better to stay without cache at all, if performance is suitable.

It'd be definitely better to live without cache, however, I've measured 10-20% performance drop without it. :frowning_face: Maybe there are other places for optimization but I can't see it right now. 

> I mean, when you calculate classes, you traverse parents from the root and merge data. If parent has multiple children you can remember merged result for parents instead of calculate all from the begining for each child.

Got it. It's a kind of dynamically built cache to save only the requested properties or a fixed structure with all values?

@embeddedt @puzrin 

>Some method for users to easily overwrite class properties without needing to make a new class is necessary, in my opinion. It's useful in the event that, for example, someone wants to dynamically change the color of a label without having to add or remove classes. I do something similar on React websites frequently.

> Got it. I'd say, in HTML writing style attr directly is bad practice. Normally such things used only in JS-driven animations, to set intermediate values. Script set property directly and debugger [...]
I think, local values will be implemented somehow anyway, in explicit or implicit form. Question is "can those be unified/simplified via classes or not". This is potential place for core simplification.

Local styles are just like classes dedicated to the object. Both are a list of properties, but the classes are shared and local styles have higher priority. 

What about making local styles optional and create a functions like: `lv_obj_add/get/remove_local_style`. Animations could write the local style, which would make it quite simple. (instead of adding an extra class for animation).

puzrin, 2020-01-04:
> It explains why was the `p` element changes color.

Ah, you mean some CSS properties (colors and so on) are inherited down to childs? Yes, that's true. Events buble up, but styles may inherit down.

> Imagine a fast-updating chart that is on the 3rd upper layer. Maybe there are complex lists, other charts, running animations, or who know what on the lower layers. Drawing all of them would make [...]

Hm... do you mean dynamic chart in popup window over main screen content? Looks correct, but a bit artificial.

Another sample - multiple tabs with chart on each. More simple, because hidden tab content can have CSS property like "display: none", and you have direct info about render skip.

Well, i'm not ready to say this optimization is absolutely useless. But i would inspect more carefully to understand real cases for win.

> Got it. It's a kind of dynamically built cache to save only the requested properties or a fixed structure with all values?

Yes. Benefit is, this cache is very "local", only in scope of traversal algorythm. Almost as local variables on stack. No space for mistakes, no outer dependencies. You can add or remove this feature transparently anytime, without risk to fuckup all architecture.

Since cache is local and dies quick, you can optimize data for speed instead of memory. I mean abstract "class" stores propecties, but it can organize those different: `Merge(Class1, Class2) => FatButFastClass3`

> What about making local styles optional and create a functions like: `lv_obj_add/get/remove_local_style`. Animations could write the local style, which would make it quite simple. (instead of adding an extra class for animation).

Yes, i mean something like this. You can create any facade for user, but have hidden class inside. And user just work with properties and don't know what happens under the hood.

The goal of this suggestion - check if it worth unify styles merge. I don't know if it helps or not. In theory, if you have only one plate to optimize - that's much better of optimizing 2 places.

Unique classes spoils effectivity of "class list manager". Probably more flexible storage required. For example instead of "classListRef" each object will have to store tuples: [staticClassListRef, localClass].

kisvegabor, 2020-01-04:
> Ah, you mean some CSS properties (colors and so on) are inherited down to childs? Yes, that's true. Events buble up, but styles may inherit down.

I knew that styles inherited to the children, I just didn't know whose state matters. But it's clear now. :)

> Another sample - multiple tabs with chart on each. More simple, because hidden tab content can have CSS property like "display: none", and you have direct info about render skip

Non-visible tabs are cropped anyway because they are out of their parents. (lvgl works with `overflow:hidden` by default)

> Since cache is local and dies quick, you can optimize data for speed instead of memory. I mean abstract "class" stores propecties, but it can organize those different: Merge(Class1, Class2) => FatButFastClass3

What is the advantage of `FatButFastClass3`? If `Class1` and `Class2` have 10-10 properties `FatButFastClass3` will have 20 properties. (Assuming no repetition). And scanning 20 properties instead of 2x10 is not a big gain.

embeddedt, 2020-01-05:
> What is the advantage of FatButFastClass3? If Class1 and Class2 have 10-10 properties FatButFastClass3 will have 20 properties.

I don't fully understand either, but maybe it eliminates the overhead of repeatedly traversing a bunch of linked pointers.

ShenRen, 2020-01-06:
Same as html + css , Qt has style sheet syntax named qss, which is very easy to use.
[The Style Sheet Syntax | Qt Widgets 5.14.0](https://doc.qt.io/qt-5/stylesheet-syntax.html)
I hope it inspires you.

[6 more replies]

closed 2020-02-15 by kisvegabor