lvgl/lvgl #1313: Minor Fixes

xennex22, 2019-12-16:
lv_objx\lv_slider.c(36)
Parameters should be const
```
static void lv_slider_draw_knob(lv_obj_t * slider, lv_area_t * knob_area, lv_area_t * clip_area);
```
```
static void lv_slider_draw_knob(lv_obj_t * slider, const lv_area_t * knob_area, const lv_area_t * clip_area);
```
lv_objx\lv_kb.c(39)
Constant object is being stored in RW memory.
Three instances of this.  Unless the const char * char breaks some compilers.
```
static const char * default_kb_map_lc[] = {
```
```
static const char * const default_kb_map_lc[] = {
```

lv_objx\lv_kb.c(61) 
"Bksp" -> LV_SYMBOL_BACKSPACE

\lv_objx\lv_chart.c(191)
Repeated line
```
if(ser == NULL) return NULL;
```

lv_core\lv_indev.c(42)
Old prototype replaced by lv_indev_search_obj() ?
```
static lv_obj_t * indev_search_obj(const lv_indev_proc_t * proc, lv_obj_t * obj);
```

lv_core\lv_group.c(445)
Three instances.
```
if(!group) return false;
```
```
if(!group) return NULL;
```

lv_objx\lv_sw.c(538)
Not sure if this should be a max() or just repeated ?
```
knob_off_size = LV_MATH_MAX(knob_on_size, info.w / 2);
knob_off_size = LV_MATH_MAX(knob_on_size, info.h / 2);
```

lv_objx\lv_chart.c(934)
Repeated assignment
```
style_point.body.radius       = LV_RADIUS_CIRCLE;
style_point.body.opa          = ext->series.opa;
style_point.body.radius       = ext->series.width;
```

lv_misc\lv_ll.c(139)
The first NULL comparison is redundant.
```
void * lv_ll_ins_tail(lv_ll_t * ll_p)
{
    lv_ll_node_t * n_new;

    n_new = lv_mem_alloc(ll_p->n_size + LL_NODE_META_SIZE);
    if(n_new == NULL) return NULL;

    if(n_new != NULL) {
        node_set_next(ll_p, n_new, NULL);       /*No next after the new tail*/
        node_set_prev(ll_p, n_new, ll_p->tail); /*The prev. before new is tho old tail*/
        if(ll_p->tail != NULL) {                /*If there is old tail then the new comes after it*/
            node_set_next(ll_p, ll_p->tail, n_new);
        }

        ll_p->tail = n_new;      /*Set the new tail in the dsc.*/
        if(ll_p->head == NULL) { /*If there is no head (1. node) set the head too*/
            ll_p->head = n_new;
        }
    }

    return n_new;
}
```

Many thanks to those who contribute to this project.

kisvegabor, 2019-12-16:
Hi,

Thank you for the list! Can you send a Pull request? (To be sure anything won't be missed)

xennex22, 2019-12-16:
Sure.

This I could not do as I'm not sure of the correct code
lv_objx\lv_sw.c(538)
```
knob_off_size = LV_MATH_MAX(knob_on_size, info.w / 2);
knob_off_size = LV_MATH_MAX(knob_on_size, info.h / 2);
```

closed 2019-12-18 by xennex22

kisvegabor, 2019-12-19:
> knob_off_size = LV_MATH_MAX(knob_on_size, info.w / 2);
knob_off_size = LV_MATH_MAX(knob_on_size, info.h / 2);

I've fixed it here 5927de98

Thank you!