lvgl/lvgl #1316: lv_micropython: importing lvesp32 causes subsequent soft reboot to crash

spierepf, 2019-12-16:
First off, I want to say that I discovered this project a few days ago, and I think it is pretty awesome. I've been playing with it in the Arduino environment and it has worked well. However, I really like micropython, so I was thrilled to see that there were bindings.

I use rshell to interact with my DOIT ESP32 DEVKIT V1 board.
I believe rshell uses soft reboots before many of its on-board filesystem operations, and I've noticed that whenever I use the REPL to `import lvesp32` the subsequent soft reboot panics.

**To Reproduce**

```
$ python3 $HOME/.local/lib/python3.6/site-packages/rshell/main.py -p /dev/ttyUSB0 -b 115200 -e nano $@
Using buffer-size of 32
Connecting to /dev/ttyUSB0 (buffer-size 32)...
Trying to connect to REPL . connected
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /boot.py/
Setting time ... Dec 16, 2019 22:38:29
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
/home/build/lv_micropython> repl
Entering REPL. Use Control-X to exit.
>
MicroPython v1.9.4-2068-gbf7699eee on 2019-12-16; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> import lvgl as lv
>>> import lvesp32
>>> 
MPY: soft reboot
MicroPython v1.9.4-2068-gbf7699eee on 2019-12-16; ESP32 module with ESP32
Type "help()" for more information.
>>> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401e5f54  PS      : 0x00060330  A0      : 0x80104a39  A1      : 0x3ffcd080  
A2      : 0x3ffb388c  A3      : 0xc0006989  A4      : 0x3ffcd260  A5      : 0x00000001  
A6      : 0x0000000a  A7      : 0x3ffcd1e0  A8      : 0x00000018  A9      : 0x3ffcd070  
A10     : 0x3fffa599  A11     : 0x3f414018  A12     : 0x00000014  A13     : 0x3f4488ad  
A14     : 0x00000002  A15     : 0x3ffcd1c0  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xc0006989  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x401e5f54:0x3ffcd080 0x40104a36:0x3ffcd0b0 0x4014268f:0x3ffcd0d0 0x400e343a:0x3ffcd0f0 0x400df439:0x3ffcd110 0x400df47d:0x3ffcd130 0x400e035c:0x3ffcd160 0x400e0429:0x3ffcd1e0 0x400e0477:0x3ffcd200 0x400f4505:0x3ffcd220 0x40141d45:0x3ffcd240 0x401423cc:0x3ffcd260 0x400f43b0:0x3ffcd2a0 0x40090cc9:0x3ffcd2d0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4808
load:0x40078000,len:9652
load:0x40080400,len:6200
entry 0x400806f0
W (312) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
MicroPython v1.9.4-2068-gbf7699eee on 2019-12-16; ESP32 module with ESP32
Type "help()" for more information.
>>>  
```

**Expected behavior**

Soft reboots should not cause crashes, no matter what libraries are imported.

**Additional context**

The example `lvgl` scripts that I have tested so far have worked as expected. The buttons and labels come up exactly the way they are shown in the accompanying screen shots. But they appear to leave the board in a fragile state that would probably not be stable in production.

embeddedt, 2019-12-17:
@amirgon Can you look at this?

spierepf, 2019-12-17:
My first thought was limited memory size on the DOIT, so I tried to build for my LOLIN D32 PRO. I got essentially the same result Here is the script I used to build/deploy on the PRO in case thatis helpful:

```
#!/bin/bash

export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin

git clone --recurse-submodules https://github.com/littlevgl/lv_micropython.git
UPY_SOURCE=$HOME/lv_micropython

cd $UPY_SOURCE
ESP_IDF_HASH=`make -C ports/esp32/ | grep v3.3 | cut -d' ' -f5`

cd $HOME/esp-idf
git checkout $ESP_IDF_HASH
git submodule update --init --recursive

cd $UPY_SOURCE
cat <<EOF >ports/esp32/partitions-16MiB.csv
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x200000,
vfs,      data, fat,     0x210000, 0xdf0000,
EOF

make PYTHON=python3 ESPIDF=$HOME/esp-idf LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC_SPIRAM PART_SRC=partitions-16MiB.csv -C mpy-cross
make PYTHON=python3 ESPIDF=$HOME/esp-idf LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC_SPIRAM PART_SRC=partitions-16MiB.csv -C ports/esp32 erase
make PYTHON=python3 ESPIDF=$HOME/esp-idf LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC_SPIRAM PART_SRC=partitions-16MiB.csv -C ports/esp32 deploy
```

amirgon, 2019-12-17:
Thanks for reporting this @spierepf.

When micropython soft-resets, it deallocates all the ram that was allocated by gc.
lvgl is taking advantage of micropython's gc but today it is not aware of the soft reset, therefore after soft reset lvgl holds invalid pointers.

I can use a callback to trigger something upon soft reset (`MICROPY_PORT_DEINIT_FUNC`). After soft reset lvgl needs to be "de-initialized" in order to set `lv_initialized` to false.
This way, when calling `lv_init` again after soft reset, lvgl would allocate all memories again.

@kisvegabor - Do you agree to add such de-init function? I think it's important to do it on v6 to fix this bug, and it does not break backward compatibility.

spierepf, 2019-12-18:
@amirgon The fix doesn't sound too complicated. Assuming that @kisvegabor  considers this an appropriate fix, would you mind if I take a crack at it and submttted a PR? I'd like to contribute, even in a small way.

amirgon, 2019-12-18:
> Assuming that @kisvegabor considers this an appropriate fix, would you mind if I take a crack at it and submttted a PR? I'd like to contribute, even in a small way.

Sure, go for it!  

This is not enough, though. 
The following would need to be handled as well upon soft reset (a call to `MICROPY_PORT_DEINIT_FUNC`):
- lvesp32 module has a FreeRTOS timer that needs to be deleted 
- ili9341 driver needs to free the SPI bus

amirgon, 2019-12-18:
> This is not enough, though.

Looking at this again - it's a bit more complicated.
It's not enough to set `lv_initialized` to false, we also need to reset all gc root variables.
In order to state all gc roots only once, it would require some preprocessor trickery.

embeddedt, 2019-12-18:
Can't the processor just be restarted fully, like any other device? That's how most other applications would handle this. It also solves the problem of resetting LittlevGL cleanly.

amirgon, 2019-12-18:
> Can't the processor just be restarted fully, like any other device? That's how most other applications would handle this. It also solves the problem of resetting LittlevGL cleanly.

Of course - that is called "hard reset". It works well and this is what I usually do when I want to "restart".
However, this issue was opened on **soft reset**. Apparently there are some applications that require this instead of a hard reset (it's probably quicker)

amirgon, 2019-12-21:
Implemented on: 
- https://github.com/littlevgl/lvgl/pull/1319
- https://github.com/amirgon/lv_bindings/commit/461fcf42a135c7d4142527e1e0ff805225197cc7
- https://github.com/amirgon/lv_mpy/commit/25763e707e88f65d3b2d0da110e7075203e29f62

(After lvgl pull is accepted I'll push the others)

This adds support for soft reset on lvgl, lvesp32, ili9341 and xpt2046 (micropython drivers) .
Every driver needs to take soft reset into account in order to free resources (such as DMA memory, SPI bus etc).

There is one open issue, however.
When using ili9341 with double buffer, there is a heap corruption when trying to free the DMA memory. This only happens with double buffer, only when lvgl is actually using the buffer, so I suspect  there's some code in lvgl that corrupts the heap somewhere around the display buffer when using double buffer.  

In the meanwhile, for cases where soft reset is important, a workaround is to use a single buffer (I've added `double_buffer` named argument to `ili9341` so this could be controlled by the user).

embeddedt, 2019-12-21:
> This only happens with double buffer, only when lvgl is actually using the buffer, so I suspect there's some code in lvgl that corrupts the heap somewhere around the display buffer when using double buffer.

Do you know if this happens without MicroPython (i.e. with a C driver)?

amirgon, 2019-12-21:
> Do you know if this happens without MicroPython (i.e. with a C driver)?

No, actually I'm only working with Micropython and a micropython driver.  
On C (without micropython) there is no way to remove a driver and no point deallocating the buffer, so I wouldn't be surprised that no one ever tried this.  

btw, I've noticed [someone requested a deinit function on the forum](https://forum.littlevgl.com/t/turn-off-display-deinit-littlevgl/1406), so that might be useful for C as well. At least as a way to reduce RAM consumption, since power/performance can be reduced in other ways.

amirgon, 2019-12-21:
> (After lvgl PR is accepted I'll push the others)

Pushed to `lv_binding_micropython` and `lv_micropython`.
Please let me know if you see further problems.

Otherwise, I think this can be closed.  
Maybe a new issue could be opened on the heap corruption, but for now I haven't proved yet this is an lvgl problem (really hard to debug that one).

spierepf, 2019-12-22:
I really appreciate the effort. This is a very cool project. Unfortunately I cannot check out the code any longer:
```
$ git clone --recurse-submodules https://github.com/littlevgl/lv_micropython.git
Cloning into 'lv_micropython'...
remote: Enumerating objects: 76715, done.
remote: Total 76715 (delta 0), reused 0 (delta 0), pack-reused 76715
Receiving objects: 100% (76715/76715), 53.04 MiB | 6.06 MiB/s, done.
Resolving deltas: 100% (55415/55415), done.
Submodule 'lib/asf4' (https://github.com/adafruit/asf4) registered for path 'lib/asf4'
Submodule 'lib/axtls' (https://github.com/pfalcon/axtls) registered for path 'lib/axtls'
Submodule 'lib/berkeley-db-1.xx' (https://github.com/pfalcon/berkeley-db-1.xx) registered for path 'lib/berkeley-db-1.xx'
Submodule 'lib/libffi' (https://github.com/atgreen/libffi) registered for path 'lib/libffi'
Submodule 'lib/lv_bindings' (https://github.com/littlevgl/lv_bindings.git) registered for path 'lib/lv_bindings'
Submodule 'lib/lwip' (https://git.savannah.gnu.org/r/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'lib/mbedtls'
Submodule 'lib/mynewt-nimble' (https://github.com/apache/mynewt-nimble.git) registered for path 'lib/mynewt-nimble'
Submodule 'lib/nrfx' (https://github.com/NordicSemiconductor/nrfx.git) registered for path 'lib/nrfx'
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path 'lib/stm32lib'
Submodule 'lib/tinyusb' (https://github.com/hathach/tinyusb) registered for path 'lib/tinyusb'
Cloning into '/home/build/lv_micropython/lib/asf4'...
remote: Enumerating objects: 1520, done.        
remote: Total 1520 (delta 0), reused 0 (delta 0), pack-reused 1520        
Receiving objects: 100% (1520/1520), 1.75 MiB | 4.86 MiB/s, done.
Resolving deltas: 100% (928/928), done.
Cloning into '/home/build/lv_micropython/lib/axtls'...
remote: Enumerating objects: 1, done.        
remote: Counting objects: 100% (1/1), done.        
remote: Total 3191 (delta 0), reused 1 (delta 0), pack-reused 3190        
Receiving objects: 100% (3191/3191), 3.81 MiB | 5.76 MiB/s, done.
Resolving deltas: 100% (2342/2342), done.
Cloning into '/home/build/lv_micropython/lib/berkeley-db-1.xx'...
remote: Enumerating objects: 270, done.        
remote: Total 270 (delta 0), reused 0 (delta 0), pack-reused 270        
Receiving objects: 100% (270/270), 332.23 KiB | 1.94 MiB/s, done.
Resolving deltas: 100% (127/127), done.
Cloning into '/home/build/lv_micropython/lib/libffi'...
remote: Enumerating objects: 12491, done.        
remote: Total 12491 (delta 0), reused 0 (delta 0), pack-reused 12491        
Receiving objects: 100% (12491/12491), 7.37 MiB | 6.04 MiB/s, done.
Resolving deltas: 100% (7833/7833), done.
Cloning into '/home/build/lv_micropython/lib/lv_bindings'...
remote: Enumerating objects: 41, done.        
remote: Counting objects: 100% (41/41), done.        
remote: Compressing objects: 100% (31/31), done.        
remote: Total 1100 (delta 19), reused 26 (delta 10), pack-reused 1059        
Receiving objects: 100% (1100/1100), 1.14 MiB | 5.28 MiB/s, done.
Resolving deltas: 100% (649/649), done.
Cloning into '/home/build/lv_micropython/lib/lwip'...
remote: Counting objects: 53114, done.        
remote: Compressing objects: 100% (12606/12606), done.        
remote: Total 53114 (delta 40017), reused 52862 (delta 39850)        
Receiving objects: 100% (53114/53114), 10.17 MiB | 5.86 MiB/s, done.
Resolving deltas: 100% (40017/40017), done.
Cloning into '/home/build/lv_micropython/lib/mbedtls'...
remote: Enumerating objects: 11, done.        
remote: Counting objects: 100% (11/11), done.        
remote: Compressing objects: 100% (11/11), done.        
remote: Total 95218 (delta 2), reused 3 (delta 0), pack-reused 95207        
Receiving objects: 100% (95218/95218), 41.92 MiB | 6.25 MiB/s, done.
Resolving deltas: 100% (73631/73631), done.
Cloning into '/home/build/lv_micropython/lib/mynewt-nimble'...
remote: Enumerating objects: 124, done.        
remote: Counting objects: 100% (124/124), done.        
remote: Compressing objects: 100% (111/111), done.        
remote: Total 37081 (delta 70), reused 17 (delta 5), pack-reused 36957        
Receiving objects: 100% (37081/37081), 10.80 MiB | 6.25 MiB/s, done.
Resolving deltas: 100% (22269/22269), done.
Cloning into '/home/build/lv_micropython/lib/nrfx'...
remote: Enumerating objects: 1882, done.        
remote: Total 1882 (delta 0), reused 0 (delta 0), pack-reused 1882        
[... 80 lines cut ...]
remote: Enumerating objects: 2362, done.        
remote: Total 2362 (delta 0), reused 0 (delta 0), pack-reused 2362        
Receiving objects: 100% (2362/2362), 8.53 MiB | 6.06 MiB/s, done.
Resolving deltas: 100% (1853/1853), done.
Cloning into '/home/build/lv_micropython/lib/tinyusb/tools/uf2'...
remote: Enumerating objects: 12, done.        
remote: Counting objects: 100% (12/12), done.        
remote: Compressing objects: 100% (12/12), done.        
remote: Total 380 (delta 2), reused 9 (delta 0), pack-reused 368        
Receiving objects: 100% (380/380), 110.91 KiB | 1.56 MiB/s, done.
Resolving deltas: 100% (193/193), done.
Submodule path 'lib/tinyusb/hw/mcu/microchip/samd/asf4': checked out '82fe3aa05bc3b9f4956ee775c748a3885443f66b'
Submodule path 'lib/tinyusb/hw/mcu/nordic/nrfx': checked out '9d68726e41c321f1772c187bd12d82f5b13104f1'
[... cut ...]
Submodule 'hidapi' (https://github.com/signal11/hidapi) registered for path 'lib/tinyusb/tools/uf2/hidapi'
Cloning into '/home/build/lv_micropython/lib/tinyusb/tools/uf2/hidapi'...
remote: Enumerating objects: 2006, done.        
remote: Total 2006 (delta 0), reused 0 (delta 0), pack-reused 2006        
Receiving objects: 100% (2006/2006), 2.73 MiB | 5.26 MiB/s, done.
Resolving deltas: 100% (1172/1172), done.
Submodule path 'lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
Failed to recurse into submodule path 'lib/lv_bindings'
```

amirgon, 2019-12-22:
> Unfortunately I cannot check out the code any longer

@spierepf 
Fixed. Please try again.

spierepf, 2019-12-23:
It works great now! I'd love

closed 2019-12-23 by spierepf

embeddedt, 2020-04-22:
@amirgon The issue with ILI9341 and double buffering was reported again, just so you know: https://forum.littlevgl.com/t/double-buffer-heap-corruption/2123

amirgon, 2020-04-22:
> @amirgon The issue with ILI9341 and double buffering was reported again, just so you know: https://forum.littlevgl.com/t/double-buffer-heap-corruption/2123

@embeddedt Thanks for letting me know. I spent some time debugging this in the past, it looks like a nasty memory corruption.  
At the moment I'm a bit busy, hopefully I'll get back to this some time in the future.