lvgl/lvgl #1320: Improve the MicroPython simulator

embeddedt, 2019-12-22:
The [MicroPython simulator](https://sim.littlevgl.com/micropython/ports/javascript/lvgl_editor.html) is online and ready to use, but needs to be polished further.

Here is a list of improvements compiled by @amirgon:

- [ ] Monaco Editor
- [ ] Allow changing the lvgl screen size
- [ ] Try to improve Frame-Rate
- [ ] Host SnippetBin backend on LittlevGL server instead of Heroku
- [ ] Improve styling of the html, Bootstrap, change frame sizes by dragging etc.  

Another feature that could be useful:
- [ ] Support loading fonts on runtime on lvgl. This would allow sharing snippets with custom fonts

embeddedt, 2019-12-22:
> Monaco Editor

I experimented with it over the course of a few weekends. When used in Python mode, it's actually no better than our existing editor on its own, because it requires a language server running for anything beyond basic syntax coloring.

amirgon, 2019-12-22:
> it's actually no better than our existing editor on its own, because it requires a language server running for anything beyond basic syntax coloring.

@embeddedt have a look at this example from [here](https://microsoft.github.io/monaco-editor/index.html):
[image]

the same code on Ace editor:
[image]

What I like about Monaco is
- It looks cleaner, nicer color and fonts, feels more modern
- There is a "document map" on the right that can help scrolling

So it's not a big difference, but even without a language server I think Monaco is nicer.

puzrin, 2019-12-26:
> Host SnippetBin backend on LittlevGL server instead of Heroku

There are other alternatives:

- gists
- URL-encoded content, in anchor (up to 2K should work everywhere, don't remember upper limit)

embeddedt, 2019-12-26:
We already discussed those other alternatives, but they had a number of downsides. We concluded concluded that SnippetBin was the best choice.

amirgon, 2020-01-17:
> Allow changing the lvgl screen size

Micropython SDL driver now accepts parameters for screen width and height.  
So only need to dynamically set canvas size in order to implement this in the simulator.

See:
- PR: https://github.com/littlevgl/lv_binding_micropython/pull/65
- Usage example: https://gist.github.com/amirgon/f1f9e3a448ce089536ba725e8456a19d

embeddedt, 2020-01-17:
I think we need a hook for adjusting the window size after initialization as well.

amirgon, 2020-01-17:
> I think we need a hook for adjusting the window size after initialization as well.

To simplify things, what if you let the user adjust it by interactively dragging the window edges?  
If it's too small, then only part of the viewport would be visible.

kisvegabor, 2020-01-20:
> To simplify things, what if you let the user adjust it by interactively dragging the window edges? 

It'd be great, but I don't know if @embeddedt has time and interest to do it.

embeddedt, 2020-01-20:
I'm definitely interested in making it work, but finding the time is an issue. I definitely won't be able to start on this before February.

embeddedt, 2020-02-09:
I've started working on replacing the existing editor with Monaco. Hopefully I can get it done this weekend.

embeddedt, 2020-02-09:
Here's a teaser. :wink:

[image]

I'm working on ironing out the last few issues, then I'll push this to the `lvgl_javascript` branch.

embeddedt, 2020-02-09:
The changes are pushed and live on https://sim.littlevgl.com. Let me know if you think anything needs adjusting.

kisvegabor, 2020-02-10:
Look nice! 
What were the "Á" characters in the screenshot?

amirgon, 2020-02-10:
Very nice!
I'm currently traveling, I'll have a look after I'm back next week.

embeddedt, 2020-02-10:
@kisvegabor 

> What were the "Á" characters in the screenshot?

At the time, the page was set to use ISO-8859-1 for the character encoding (not UTF-8), so I think it was interpreting `&nbsp;` as `Á`, or something like that.

@amirgon 

> I'm currently traveling, I'll have a look after I'm back next week.

No rush; have a safe trip!

amirgon, 2020-02-14:
> The changes are pushed and live on https://sim.littlevgl.com. Let me know if you think anything needs adjusting.

@embeddedt it's not working for me.
I'm getting:
```
main.65321e8f.js:1097 Uncaught ReferenceError: mp_js_stdout is not defined
    at main.65321e8f.js:1097
    at dispatch (jquery-1.12.4.js:5226)
    at elemData.handle (jquery-1.12.4.js:4878)
```

I'm using latest Chrome (Version 78.0.3904.70 (Official Build) (64-bit)) on Linux.

embeddedt, 2020-02-14:
Hmm. It seemed to be working for me on Firefox when I wrote that message, but I've just tried it now in Chrome and it is in fact not working.

I'll take a look when I get a chance. In the meantime, my latest commit to https://github.com/littlevgl/sim should be reverted (I'll try and do that today).

embeddedt, 2020-02-15:
Fixed... it should start working once GitHub Pages deploys my latest commit.

kisvegabor, 2020-02-15:
I"ve just tried the new version and i works well in Firefox but the "Restart" button remains inactive in Chrome.

embeddedt, 2020-02-15:
@kisvegabor GitHub had issues with the new version for some reason (I got an error about one of the symlinks in the repository), so I think you're still seeing the old version. I'll fix it today.

embeddedt, 2020-02-15:
The new version is pushed and appears to work fine in Chrome now.

Note that you may need to press Ctrl+F5 if it doesn't work the first time, as Chrome may still be caching the old, broken version.

amirgon, 2020-02-15:
> The new version is pushed and appears to work fine in Chrome now.

It does not work when passing `script` parameter:  

https://sim.littlevgl.com/micropython/ports/javascript/bundle_out/index.html?script=https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/master/examples/advanced_demo.py&cache_buster=7

```
main.cb19d18d.js:formatted:119808 Uncaught ReferenceError: total is not defined
    at XMLHttpRequest.o.onreadystatechange (main.cb19d18d.js:formatted:119808)
```

Also, could you list the parameters that the simulator accepts? (`script`, `init_script`, `cache_buster` etc.)

embeddedt, 2020-02-16:
> It does not work when passing `script` parameter.

The fix is pushed.

> Also, could you list the parameters that the simulator accepts?

* `script` - main script loaded through URL
* `script_direct` - main script loaded through SnippetBin
* `script_startup` - startup script loaded through URL
* Any other parameters are ignored but will force GitHub to load a new version of the page faster.

> It seems like you are using an older micropython version.

I'll merge `master` into it when I work on it next.

amirgon, 2020-02-16:
Very nice!
Thanks for switching to Monaco. It looks so much better!

For the reference here is an example of using both `script_startup` and `script` and also importing `display_driver` from the internet:

https://sim.littlevgl.com/micropython/ports/javascript/bundle_out/index.html?script_startup=https://gist.githubusercontent.com/amirgon/9cccfb7f22a533b90ae544f22d6df726/raw/9a58687894b250b13c624ae578c7705f1c052d57/lv_sim_init.py&script=https://raw.githubusercontent.com/littlevgl/lv_examples/rework/src/lv_ex_btn/lv_ex_btn_1.py

Some issues I noticed:
- Revision history doesn't work. After saving a few versions I expected that scrolling history would change the source, but it doesn't do anything.
- When clicking `Link to online script`, the `script_direct` parameter is added as expected but I also expect the other parameters (`script` and `script_startup`) to be removed since the don't have any effect any more once `script_direct` is specified.

kisvegabor, 2020-02-17:
Works here too in Chrome. Thank you for the update!

embeddedt, 2020-05-01:
@amirgon I may have figured out why the frame rate is so low when the MicroPython simulator first starts. Because Emscripten uses JIT compilation for running the code, I think LittlevGL ends up running through several frames before being recognized as a "hot path" by the compiler, at which point the frame rate improves.

I've noticed a less severe variation of this on the C-to-Emscripten port, but in that case the frame rate isn't too bad. I'm guessing that this is because there is less code running, so the JIT compiler decides that LittlevGL is worth optimizing sooner.

I'm not sure if there's a way of flagging certain code as being a "hot path" manually; I'll look into that once the MicroPython binding has been updated.

kisvegabor, 2020-05-01:
@embeddedt 
I'm reading the [docs of Emscripten](https://emscripten.org/index.html) and can't find an option to configure the JIT.

However, I've found a section about WebGL. I suppose we can achieve some speed up if we integrated WebGL as GPU. What do you think?

puzrin, 2020-05-20:
> I may have figured out why the frame rate is so low when the MicroPython simulator first starts. Because Emscripten uses JIT compilation for running the code, I think LittlevGL ends up running [...]

Switch to WebAssembly backend and forget about warm up. You will get speed close to naive C. WebAssembly just works in all modern browsers.

BTW, lv_font_conv uses WebAssembly build of FontForge (via emscripten).

> However, I've found a section about WebGL. I suppose we can achieve some speed up if we integrated WebGL as GPU. What do you think?

I think no. WebGL is for advanced manipulations. I don't know details about lvgl web driver, but at first glance it should just write memory framebuffer and copy all framebuffer to canvas element. That's fast enough and nothing to optimize.

embeddedt, 2020-05-20:
> Switch to WebAssembly backend and forget about warm up. You will get speed close to naive C. WebAssembly just works in all modern browsers.

I don't know if the upstream MicroPython port supports WebAssembly. I'm trying to make minimal changes to it to ease merging.

amirgon, 2020-05-20:
@embeddedt - Do you think you could find some time to look into the [issues I raised above](https://github.com/lvgl/lvgl/issues/1320#issuecomment-586763221)?   
They are not critical, but a bit annoying.

puzrin, 2020-05-20:
@embeddedt That's not related to MicroPython. That's emscripten flag. If your project can be compiled, then you can select what output to use - JS, WASM or both. It's transparent. Make sure your result is "not js only"

embeddedt, 2020-05-20:
I will try flipping it to WASM mode, but I think there are probably incompatibilities in the runtimes which will need to be addressed. I'll take a look at the issues @amirgon mentioned at the same time.

embeddedt, 2020-05-21:
I got the REPL to work with WASM. Still working on getting LittlevGL itself to run (I'm running into some Emscripten errors which aren't very descriptive).

kisvegabor, 2020-05-21:
I compiled the widget demo with Emscripten and added to the new website: https://lvgl.io/demos

Interestingly, it runs smoothly on my mid-top category notebook but lags on my friend's top category PC in Chrome. It's smooth for him in Edge. It would be great if these lags also could be solved by switching to WASM.

kisvegabor, 2020-05-21:
@embeddedt 
I've noticed that the simulator page is broken because bootstrap is not available on littlevgl.com but on lvgl.io now. Can you update its path to `https://lvgl.io/public/assets/bootstrap/`?

embeddedt, 2020-05-21:
> It would be great if these lags also could be solved by switching to WASM.

Weird, because the C Emscripten port already uses WASM. Take a look at the Network section in Developer Tools and you should see it fetching `lvgl.wasm`.

> Can you update its path to `https://lvgl.io/public/assets/bootstrap/`?

I've already done this locally; is it okay if I push that fix along with everything else, or should I separate it out and push it now?

kisvegabor, 2020-05-21:
> Weird, because the C Emscripten port already uses WASM. Take a look at the Network section in Developer Tools and you should see it fetching lvgl.wasm.

Ah, I see. Thank you for the clarification. 

> I've already done this locally; is it okay if I push that fix along with everything else, or should I separate it out and push it now?

If it's not a big deal could you fix it to not have a broken site?

embeddedt, 2020-05-23:
I've pushed the fix.

kisvegabor, 2020-05-24:
@embeddedt 
It's still broken because bootstrap is not in the root but in `https://lvgl.io/public/assets/bootstrap/`

embeddedt, 2020-05-24:
It's because I forgot to update the `sim` repository, so it was still pointing to the older commit. It should update itself now.

amirgon, 2020-05-24:
> I've pushed the fix.

@embeddedt I think the [revision history and the script/script_direct](https://github.com/lvgl/lvgl/issues/1320#issuecomment-586763221) are still broken.

embeddedt, 2020-05-24:
I haven't changed any code yet; that was just a quick hotfix to correct the broken styles caused by the domain yet.

I'll fix the other things as soon as I can.

amirgon, 2020-07-16:
@embeddedt I believe the revision history is still broken.
Please have a look at [this link](https://sim.lvgl.io/v7/micropython/ports/javascript/bundle_out/index.html?script_direct=2f617292599f61d6c52844437638e9a0d989a624) (it's part of a discussion on the forum).   
When moving the history knob, the code is not changed as expected.  

In this case it could be useful for the user to see the fix I added in his code by switching the history between his version (before the last) and my fix (the last version)

embeddedt, 2020-07-16:
It looks like I had missed a function when fixing the revision history last time. I've reproduced and fixed the issue on my local copy; GitHub Pages should pick up on the change soon.

amirgon, 2020-07-30:
@embeddedt I noticed some people try to paste code into the simulator, it doesn't work for them because they don't know how to set up the display driver, and then they give up.

A simple solution could be to replace the `print("hello world")` code that appears on startup with code that actually uses lvgl and displays something.  

For example:

```python
# Initialize 

import imp, sys
sys.path.append('https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/master/lib')
import display_driver
import lvgl as lv

# Create a button with a label 

scr = lv.obj()
btn = lv.btn(scr)
btn.align(scr, lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text('Hello World!')
lv.scr_load(scr)
```

What do you think?

embeddedt, 2020-08-02:
I'll change it today.

amirgon, 2020-08-14:
> I'll change it today.

 https://github.com/lvgl/lv_micropython/commit/edfa84ce54788e2b4f4d565243663c3208c3435e - @embeddedt I think it's not deployed.
I still see `print("hello world")` when openning https://sim.lvgl.io/v7/micropython/ports/javascript/bundle_out/index.html
Was it added to both v6 and v7 simulators?

btw, when I update lv_micropython release/v_ branch, do you have some automatic way to build and deploy it?

embeddedt, 2020-08-15:
> I think it's not deployed.

Strange. GitHub Pages didn't pick up on the change, but I confirmed that I did deploy the new files. I've just updated the simulator to the `master` branch so hopefully it sees the change this time.

> Was it added to both v6 and v7 simulators?

I only added it to the v7 simulator; I wasn't planning to update the v6 one anymore unless it completely broke for some reason.

> btw, when I update lv_micropython release/v_ branch, do you have some automatic way to build and deploy it?

At the moment, I manually pull the branch, build the new script, and push the change. I haven't had time to look into automating it.

embeddedt, 2020-08-15:
@amirgon It looks like it deployed this time. I wish GitHub Pages was more reliable for this kind of thing...

amirgon, 2020-08-15:
> @amirgon It looks like it deployed this time. I wish GitHub Pages was more reliable for this kind of thing...

@embeddedt Yes it looks good now. Thanks.

kisvegabor, 2020-11-26:
@amirgon @embeddedt 
As there was no activity here for a while I suggest closing this issue. 
Of course, we can reopen it any time.

amirgon, 2020-11-26:
> As there was no activity here for a while I suggest closing this issue.
> Of course, we can reopen it at any time.

Ok.  
There are [number of goals](https://github.com/lvgl/lvgl/issues/1320#issue-541453899) that we didn't achieve such as allowing changing the screen size and improving styling of the html, but on the other hand the adoption of the simulator was lower than I anticipated, unfortunately.    

Hopefully, the simulator would be used more often with the [new Micropython examples](https://github.com/lvgl/lv_examples/pull/85) and the  ["Try it yourself" buttons](https://github.com/lvgl/lvgl/issues/1872#issuecomment-719813275).

embeddedt, 2020-11-26:
As @uraich pointed out, many people want to play around with and use MicroPython on a real device rather than a simulator, which might explain the lower adoption. The main use of the simulator even with the new examples would mostly be for demonstration purposes.

I also suspect that MicroPython users may be a relative minority at the moment, given the high number of C questions as opposed to MicroPython ones. Or it's possible that things are easier to figure out with MicroPython and thus less help is needed.

amirgon, 2020-11-26:
> I also suspect that MicroPython users may be a relative minority at the moment, given the high number of C questions as opposed to MicroPython ones.

Looking at the forum today, there are 917 "How to" topics and 105 "Micropython" topics (which are mostly how-to-in-micropython).  
I think that's an interesting statistic. 10% is a minority, but not too small.  

Regarding the polls, I suspect that most Micropython users didn't even see these because they are less interested in the "General" category.

embeddedt, 2020-11-27:
> 10% is a minority, but not too small.

I agree. I wasn't discounting the users, just pointing out the relative difference.

> I suspect that most Micropython users didn't even see these because they are less interested in the "General" category.

Could you explain? The polls were pinned to the top of the post list so they should have been visible to everyone right away.

amirgon, 2020-11-27:
> Could you explain? The polls were pinned to the top of the post list so they should have been visible to everyone right away.

When I enter the forum I usually switch to Micropython category without paying to much attention to other categories, or even directly open the forum on that category.  
Poll's were pinned but only on General category, so to see them you need to either browse all posts, or switch to General category.

kisvegabor, 2020-11-27:
If you agree I can create a banner post (large and visible for everyone) to ask for votes.

uraich, 2020-11-27:
I am quite convinced that, if the micropython documentation was at the same level as the C docs, that would shift the mpy acceptance drastically. 
As I said before, I had a number of difficulties getting the mpy examples to work which I could only solve by having a look at the original C code and the code generated by gen_mpy.
I hope that the example collection does help to go into the direction of better mpy acceptance.

amirgon, 2020-11-27:
> If you agree I can create a banner post (large and visible for everyone) to ask for votes.

Sure, but keep it for long enough period of time. I suspect that the rate of new Micropython users is much lower than C users.

kisvegabor, 2020-11-27:
> I hope that the example collection does help to go into the direction of better mpy acceptance.

I'm sure! Someone contacted me a few days ago and said he has a good Python background but the lack of LVGL-MPy examples made it difficult for him to really use LVGL with MPy. I showed him the current state of the examples and he found it quite useful. 

> Sure, but keep it for long enough period of time. I suspect that the rate of new Micropython users is much lower than C users.

Done. We can leave it for 1-2 weeks.

[172 more replies]

closed 2022-05-30 by kisvegabor