lvgl/lvgl #1321: Latest Master Commit appears to be broken

pete-pjb, 2019-12-23:
Hi,

This morning I merged the latest master branch with my code and there appears to be conflicts in lv_list.c which are breaking the build.  For now I have reverted back to the previous version.

Please can you take a look when you have time.

Thank you,

Kind Regards,

Pete

embeddedt, 2019-12-25:
As of today, the latest version of the `master` branch seems to work fine now. There was an issue a few days ago where @kisvegabor accidentally merged part of `dev-7.0` into it, but that should be fixed now.

Can you confirm that it works?

pete-pjb, 2019-12-30:
Hi @embeddedt,

I have got it working again thanks, I had to delete my local master branch and re-create it.  I think there were some bits of version 7.0 in the mix on the local copy as you suggested.

Kind Regards,

Pete

closed 2019-12-30 by pete-pjb