lvgl/lvgl #1336: New Symbol Distortion

pmortsdl, 2019-12-31:
# All enhancement requests or questions should be directed to the Forum.

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

pmortsdl, 2019-12-31:
Hello! 

I have been using lvgl for a project that requires the use of symbols that aren't included in the library. In following the directions on the website, I successfully got an image to appear, however it is clearly distorted from what it should be. I tried using multiple symbols and they were all distorted, at different bpp values and sizes. 

The site I pulled the Unicode value from https://fontawesome.com/icons/key?style=solid 

The Documentation I've been referring to https://docs.littlevgl.com/en/html/overview/font.html

Below is the code that I am trying to use to create the symbol.
/**********************
	 * Key Bar
	 ***********************/
	/* Create a new style */
	static lv_style_t style_key;                         /*Styles can't be local variables*/
	lv_style_copy(&style_key, &lv_style_pretty);         /*Copy a built-in style as a starting point*/
	style_key.body.main_color = LV_COLOR_WHITE;
	style_key.body.grad_color = LV_COLOR_WHITE;
	style_key.body.border.width = 0;
	style_key.text.font = &key_24;
	style_key.text.color = LV_COLOR_BLACK;
	style_key.text.letter_space = 0;

	/*Create a base object and apply the new style*/
	lv_obj_t * keyBar1;
	keyBar1 = lv_obj_create(scr, NULL);
	lv_obj_set_size(keyBar1, 150, 30);
	lv_obj_set_style(keyBar1, &style_key);
	lv_obj_align(keyBar1,NULL, LV_ALIGN_CENTER, screenOffsetX,screenOffsetY);

	/* Add a label to the object.
	 * Labels by default inherit the parent's style */
	lv_obj_t * keyLabel;
	keyLabel = lv_label_create(keyBar1, NULL);
	lv_label_set_text(keyLabel, LV_SYMBOL_KEY);
	lv_obj_set_style(keyLabel, &style_key);
	lv_obj_align(keyLabel, NULL, LV_ALIGN_CENTER, 0, 0);

Is there a setting I'm missing somewhere that adjusts the image scaling? It seems to me that the image is warped, not that it's the wrong image. 

Thanks in advance.

embeddedt, 2020-01-01:
Are you using the PC simulator to test this? If not, can you try it there? If it still happens there, can you send a screenshot of the settings you used to convert the `key_24` font?

kisvegabor, 2020-01-02:
Please also attach the generated file, and describe the symbols you wanted to convert.

pmortsdl, 2020-01-02:
I tried it in the simulator and got the same image I was looking at on my physical screen, so it's not an issue there. 
[image]

Here are my settings used to create my font file.

[image]

My output file, key_24.c is below
`#include "lvgl/lvgl.h"

/*******************************************************************************
 * Size: 24 px
 * Bpp: 4
 * Opts: 
 ******************************************************************************/

#ifndef KEY_24
#define KEY_24 1
#endif

#if KEY_24

/*-----------------
 *    BITMAPS
 *----------------*/

/*Store the image of the glyphs*/
static LV_ATTRIBUTE_LARGE_CONST const uint8_t gylph_bitmap[] = {
    /* U+F084 "" */
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x1, 0x35, 0x79, 0xab, 0xcc, 0xdd, 0xdc,
    0xba, 0x97, 0x42, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x14, 0x7b, 0xdf, 0xff, 0xff,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xd9,
    0x41, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x13, 0x7b, 0xef,
    0xff, 0xff, 0xff, 0xff, 0xfe, 0xdc, 0xcc, 0xde,
    0xff, 0xff, 0xff, 0xff, 0xfb, 0x61, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x26, 0xbe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xb7,
    0x31, 0x0, 0x0, 0x0, 0x25, 0xad, 0xff, 0xff,
    0xff, 0xc7, 0x20, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x38, 0xdf, 0xff, 0xff, 0xff,
    0xff, 0xff, 0xd7, 0x20, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x5, 0xbf, 0xff, 0xff, 0xfa, 0x50, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x27, 0xcf,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xb6, 0x10,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x3, 0x9e, 0xff,
    0xff, 0xfa, 0x50, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x3, 0x9e, 0xff, 0xff, 0xff, 0xec, 0x86,
    0x43, 0x33, 0x32, 0x10, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x38, 0xdf, 0xff, 0xff, 0xe8, 0x30, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x28, 0xdf, 0xff,
    0xff, 0xe9, 0x40, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x2, 0x33, 0x33, 0x45, 0x8b, 0xef, 0xff, 0xff,
    0xfe, 0x94, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x4a, 0xff, 0xff, 0xff, 0xa4, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x5, 0xaf, 0xff, 0xff,
    0xff, 0xff, 0xff, 0xff, 0xd8, 0x20, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x4a, 0xff, 0xff,
[... 92 lines cut ...]
    .kern_scale = 0,
    .cmap_num = 1,
    .bpp = 4,
    .kern_classes = 0,
    .bitmap_format = 0
};

/*-----------------
 *  PUBLIC FONT
 *----------------*/

/*Initialize a public general font descriptor*/
lv_font_t key_24 = {
    .get_glyph_dsc = lv_font_get_glyph_dsc_fmt_txt,    /*Function pointer to get glyph's data*/
    .get_glyph_bitmap = lv_font_get_bitmap_fmt_txt,    /*Function pointer to get glyph's bitmap*/
    .line_height = 22,          /*The maximum line height required by the font*/
    .base_line = 3,             /*Baseline measured from the bottom of the line*/
    .subpx = LV_FONT_SUBPX_HOR,
    .dsc = &font_dsc           /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
};

#endif /*#if KEY_24*/
`

In the code that I have been running I have removed the line `.subpx = LV_FONT_SUBPX_HOR,` because that parameter is not included in the version of the lv_font_t struct that I downloaded originally. Is that potentially causing this issue? There weren't any comments about what it is and or does so I'm not sure.

pmortsdl, 2020-01-02:
The symbol I am trying to convert is the key from this URL

https://fontawesome.com/icons/key?style=solid

[image]

From my perspective it looks like the output is cutting off part of the key and inverting it, or just distorting the character all together.

[image]

pmortsdl, 2020-01-02:
I should also note that I tried to use this file while I was debugging myself. I thought a grid would be a good way to determine if the image was distorted or if it was just random pixels being used. 

https://fontawesome.com/icons/chess-board?style=solid

[image]

This is the font file I created,
`#include "lvgl/lvgl.h"

/*******************************************************************************
 * Size: 50 px
 * Bpp: 4
 * Opts:
 ******************************************************************************/

#ifndef CHESS
#define CHESS 1
#endif

#if CHESS

/*-----------------
 *    BITMAPS
 *----------------*/

/*Store the image of the glyphs*/
static LV_ATTRIBUTE_LARGE_CONST const uint8_t gylph_bitmap[] = {
    /* U+F43C "" */
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x3, 0x68, 0x88, 0x88, 0x88, 0x88,
    0x88, 0x88, 0x88, 0x88, 0x53, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x3, 0x68, 0x88,
    0x88, 0x88, 0x88, 0x88, 0x88, 0x88, 0x85, 0x20,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x36, 0x88, 0x88, 0x88, 0x88, 0x88, 0x88, 0x88,
    0x88, 0x85, 0x30, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x36, 0x88, 0x88, 0x88, 0x88,
    0x88, 0x88, 0x88, 0x88, 0x74, 0x10, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x16, 0xcf, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xff, 0xff, 0xa5, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x16, 0xcf, 0xff, 0xff, 0xff,
    0xff, 0xff, 0xff, 0xff, 0xfa, 0x50, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x1, 0x6c, 0xff,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfa,
    0x50, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x1, 0x6c, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xff, 0xff, 0xd8, 0x20, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x16, 0xcf,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xa5, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x16, 0xcf, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xff, 0xff, 0xfa, 0x50, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x1, 0x6c, 0xff, 0xff, 0xff,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xfa, 0x50, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x1, 0x6c,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xd8, 0x20, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x16, 0xcf, 0xff, 0xff,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xa5, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x16,
    0xcf, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xfa, 0x50, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x1, 0x6c, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xff, 0xff, 0xff, 0xfa, 0x50, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x1, 0x6c, 0xff, 0xff,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xd8, 0x20,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x16, 0xcf, 0xff, 0xff, 0xff, 0xff,
    0xff, 0xff, 0xff, 0xff, 0xa5, 0x0, 0x0, 0x0,
[... 491 lines cut ...]
    .cmap_num = 1,
    .bpp = 4,
    .kern_classes = 0,
    .bitmap_format = 0
};

/*-----------------
 *  PUBLIC FONT
 *----------------*/

/*Initialize a public general font descriptor*/
lv_font_t chess = {
    .get_glyph_dsc = lv_font_get_glyph_dsc_fmt_txt,    /*Function pointer to get glyph's data*/
    .get_glyph_bitmap = lv_font_get_bitmap_fmt_txt,    /*Function pointer to get glyph's bitmap*/
    .line_height = 51,          /*The maximum line height required by the font*/
    .base_line = 7,             /*Baseline measured from the bottom of the line*/
    .dsc = &font_dsc           /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
};

#endif /*#if CHESS*/
`

and this is the output on my simulator

[image]

puzrin, 2020-01-02:
@pmortsdl please edit your messages to format code samples as code blocks, now it's very difficult to understand.

I'd suggest you to do couple if quick checks, prior to dive more deep:

- In CLI convertor, try `--format dump` and make sure generated images look as expected
- Try smaller size in emulator, 16-20px - check if problem is caused by size or something else.

pmortsdl, 2020-01-02:
apologies about the formatting, I forgot to verify that they were showing how I had intended. They should be legible now. 

As I have been continuing to debug, I updated the version of LVGL that I was using and determined that it is indeed the line I removed from the generated file that is causing the distortion. 
```
.subpx = LV_FONT_SUBPX_HOR,
```

I would suggest that comments be added to explain what this is in case future users make the mistake I did in removing that line. 

However the image is not the same as I was expecting

[image]

Is there a reason that it doesn't match the expected font awesome symbol? If necessary I will open a different issue and we can continue there.

puzrin, 2020-01-02:
Subpixel smoothing & compression were added  in 6.1.

> Is there a reason that it doesn't match the expected font awesome symbol? If necessary I will open a different issue and we can continue there.

I'm sure problem is not in lvgl or convertor. Open your source font in FontForge and check twice codepoint of desired glyph.

pmortsdl, 2020-01-02:
Thank you for your help! that resolves this issue for me.

closed 2020-01-02 by pmortsdl

embeddedt, 2020-01-02:
Enabling subpixel support results in a different bitmap being generated. LittlevGL 6.1+ use the `subpx` field to determine whether the bitmap includes subpixel information or not. LittlevGL 6.0 has no notion of subpixel hinting and will assume that there is no subpixel information inside.

I've [added a note](https://github.com/littlevgl/lv_font_conv/commit/302f830c58e6443aaebc34da5ae439686ace7) to the font converter UI explaining this; when the web version is next updated this change should be visible to everyone.