lvgl/lvgl #1338: new features for chart object

ali-rostami, 2020-01-02:
Hi
As the @kisvegabor [suggested](https://forum.littlevgl.com/t/add-track-cursor-to-chart-object/1425/7?u=ali_rostami), I opened this issue to talk about the features we want to add to the chart object.

here is the list until now and AFAIK:

-  Add track cursor to the `chart` object
-  support different series types on one chart
- X/Y mode
- maybe rework the existing chart types? Maybe some of them can be combined or dropped.
- [Assign the range to the series](https://forum.littlevgl.com/t/assign-the-range-to-the-series/1417)

kisvegabor, 2020-01-04:
@ali-rostami Thank you for opening this issue.

@embeddedt Can you show a screenshot/example/scratch about your vision of chart cursor?

embeddedt, 2020-01-05:
I'm thinking it would be similar to what @ali-rostami showed here: https://forum.littlevgl.com/t/add-track-cursor-to-chart-object/1425/3?u=embeddedt

The cursor labels will be drawn in a similar position to where the ticks/texts are currently drawn. Therefore, using `lv_chart_set_margin` will be required for the cursor information to be properly visible.

kisvegabor, 2020-01-06:
I see. The other typical approach is to show a box like this:
[image]

Do you like the values on the edges more?

embeddedt, 2020-01-06:
I don't really have a preference, aside from the fact that the box obscures more of the chart, which might be an issue on smaller screens. Which do you prefer?

ali-rostami, 2020-01-06:
Oh, It's nice.
You know, placing the box in a proper position, is dependent on the size and complexity of the chart.
I think for more flexibility It's better to let the user create a label and put it where ever he\she wants to. 
But for doing this, there should be a function that provides us with the proper information. Maybe a getter function for the position of the cursor. By using this function we can put the label below or above the cursor or on the axes.
Another thing is the way we want to show the cursor. I think by using this getter function, we can have the most flexibility for it too. I liked the idea of using two circles in the above picture. Or maybe someone wants just the cross of two lines or dotted lines.
So what do you think?

kisvegabor, 2020-01-07:
> You know, placing the box in a proper position, is dependent on the size and complexity of the chart.

My first idea was to align the box to the average of the max. and min. valoue of the series at the given x position. (But limit on the edges of the chart.)

> But for doing this, there should be a function that provides us with the proper information. Maybe a getter function for the position of the cursor.

I agree. We should have a `lv_chart_set_cursor position(chart. x = 5)` function and other functions to get the series value at the cursor:
- `lv_chart_get_cursor_value(chart, ser1)`: get the real value of the serie
- ``lv_chart_get_cursor_value_px(chart, ser1)`: get the y coordinate of the point

There could be a style for the cursor, and the cursor could be drawn automatically with a highlight on the points under it.

So the question is how the write out the values:
- on the y axis(es)?
- next to the cursor in a box?
- multiple small boxes for each series?  
- below the cursor in a box?

It seems too many possibilities and each has pros and cons. We can consider leaving it for the user to create box/texts as he wants. With a proper API and some examples, it should be easy.

What do you think?

ali-rostami, 2020-01-08:
> It seems too many possibilities and each has pros and cons. We can consider leaving it for the user to create box/texts as he wants. With a proper API and some examples, it should be easy.
> What do you think?
As I said, I think It's better to let the programmer decide.

kisvegabor, 2020-01-09:
Okay, then we are on the same page :slightly_smiling_face: 

What do you think about "support different series types on one chart"? Is it really required? If so can you mention a use-case?

ali-rostami, 2020-01-09:
> Okay, then we are on the same page 
So good 👍
I think it's rare that someone wants to use multiple types. And if there was such an application, we can use two  -or even more- chart over each other.

kisvegabor, 2020-01-11:
> I think it's rare that someone wants to use multiple types. And if there was such an application, we can use two -or even more- chart over each other.

I agree.

Regarding the X/Y mode, it needs additional data to be stored. Now we store only the Y values. It can be a new pointer in `lv_chart_series_t`, e.g. `lv_coord_t  * point_x`, which allocated only if required. So this part seems clear.

"Assign the range to the series" would mean some kind of special horizontal cursors, wouldn't it?

ali-rostami, 2020-01-11:
> Regarding the X/Y mode, it needs additional data to be stored. Now we store only the Y values. It can be a new pointer in `lv_chart_series_t`, e.g. `lv_coord_t * point_x`, which allocated only if required. So this part seems clear.

I have no idea what you are talking about :)

> "Assign the range to the series" would mean some kind of special horizontal cursors, wouldn't it?

I wanted to implement something like this:
[image]
(I took it from [here](https://forum.littlevgl.com/t/add-secondary-axis-support-for-chart-and-y-axis-inversion/424))
the trick I used for solving it was using [2 charts over each other](https://forum.littlevgl.com/t/assign-the-range-to-the-series/1417).
but maybe it would be better if I didn't need to do this. how?
If the range and color could be set for series. so then I could set one series of data blue and the other one red. and also one ranges from (0 - 100) and the other ranges from (150 - 250).
what do you think? Does it worth it, or it's better to use the trick I used?

kisvegabor, 2020-01-11:
> I have no idea what you are talking about :)

No problem, I can implement it easily. 

> If the range and color could be set for series. so then I could set one series of data blue and the other one red. and also one ranges from (0 - 100) and the other ranges from (150 - 250).
what do you think? Does it worth it, or it's better to use the trick I used?

I see. So the series needs to be assigned either to the left or right y-axis, and there should be two ranges and colors for the two y-axis too, right?

ali-rostami, 2020-01-12:
> I see. So the series needs to be assigned either to the left or right y-axis, and there should be two ranges and colors for the two y-axis too, right?

Yeah, so good. 👍 
And for the flexibility, if it doesn't cost many resources maybe it's better to assign the color to the series.
consider a situation that we want to display 4 different series on a chart. each of them should have their own unique color.
*****************************
Another feature I think would be good to have is scrolling the chart. like the chart in [pulse oximeter demo here.](https://www.embedded-wizard.de/demo)
The solution that comes to my mind is defining direction for the `lv_chart_set_next` .
Then in the callback of the `chart` after catching `LV_EVENT_DRAG_END` event we can call it.

kisvegabor, 2020-01-13:
> consider a situation that we want to display 4 different series on a chart. each of them should have their own unique color.

You can already do this. See [here](https://github.com/littlevgl/lvgl/blob/master/src/lv_objx/lv_chart.c#L176).

> Another feature I think would be good to have is scrolling the chart. like the chart in pulse oximeter demo here.

Do you mean something like the [Update modes](https://github.com/littlevgl/lvgl/blob/master/src/lv_objx/lv_chart.h#L56-L60)?

> Then in the callback of the chart after catching LV_EVENT_DRAG_END event we can call it. what do you think?

Scrolling the chart would require to store more data than we display to have something new to show we scroll, right?

ali-rostami, 2020-01-14:
> You can already do this. See [here](https://github.com/littlevgl/lvgl/blob/master/src/lv_objx/lv_chart.c#L176).

oops, my mistake. 😕

> Do you mean something like the [Update modes](https://github.com/littlevgl/lvgl/blob/master/src/lv_objx/lv_chart.h#L56-L60)?

No, It's different. That's the update mode, but the thing I mean is direction. AFAIK right now when we call `lv_chart_set_next` the new data will be added to the right side of the chart, but I think we need to be able to change it to the left side too.
For example, consider, the cursor is on `x = 0`, when we press the `left key`, we want to shift the whole chart data to the right and add new data from the left. exactly the opposite of what we have now. -when the update mode is `LV_CHART_UPDATE_MODE_SHIFT`and the `lv_chart_set_next` is called.-

> Scrolling the chart would require to store more data than we display to have something new to show we scroll, right?

yeah, you mean the solution I mentioned, is so slow? 
then what can we do if there is so much data, stored in the flash, and the user wants to see them when he scrolls the chart?

kisvegabor, 2020-01-15:
> No, It's different. That's the update mode, but the thing I mean is direction. AFAIK right now when we call lv_chart_set_next the new data will be added to the right side of the chart, but I think we need to be able to change it to the left side too.

I see. What about having `LV_CHART_UPDATE_MODE_SHIFT_LEFT` and `LV_CHART_UPDATE_MODE_SHIFT_RIGHT`?

> yeah, you mean the solution I mentioned, is so slow?
then what can we do if there is so much data, stored in the flash, and the user wants to see them when he scrolls the chart?

I meant the currently not visible data should in the chart to smoothly scroll to them. Imagine that the use scrolls (drags by finger) the chart and 10 more data need to be displayed on the right or left depending on the direction of the scroll. To get the new data we have two options:
1. ask the user the tell them (via event callback?)
2. simply read the new data from the series. It requires to store more data than displayed data.

ali-rostami, 2020-01-15:
> I see. What about having `LV_CHART_UPDATE_MODE_SHIFT_LEFT` and `LV_CHART_UPDATE_MODE_SHIFT_RIGHT`?

Great 👍 

> I meant the currently not visible data should in the chart to smoothly scroll to them. Imagine that the use scrolls (drags by finger) the chart and 10 more data need to be displayed on the right or [...]

Not sure, but I think option 1 is better. The hidden data of the chart might be in the ram or an external spi flash -in my case-. so the user can choose where he can read them from. I'm just afraid if it will be so slow to scroll smoothly, even when they are read from ram.
Even if we have option 2, I think we still need to ask the user the new data in case there are so much data. For example, in my case, I might have near 1MB data stored in the spi flash. It doesn't seem reasonable to assign them all to the chart, which in addition to the huge ram it takes, It takes so much time too.

kisvegabor, 2020-01-17:
> For example, in my case, I might have near 1MB data stored in the spi flash. It doesn't seem reasonable to assign them all to the chart, which in addition to the huge ram it takes, It takes so much time too.

It's really a possible use-case. Let me think about it a little. 

I made some progress with *"maybe rework the existing chart types? Maybe some of them can be combined or dropped"*. While reworking the styles I was able to reduce the types to columns and lines only. All the other types (area, point etc) can be set via styles.

---

So in summary:
- [ ] Add cursor option with an API to let the user draw boxes or texts for the selected values.
- ~Support different series types on one chart~
- [ ] X/Y mode: doable, I'll do it.
- [x] Rework the existing chart types
- [ ] Allow assigning the series to left or right y-axis which can have different ranges
- [ ] Add scroll feature and load new data on demand
- [ ] Add `LV_CHART_UPDATE_MODE_SHIFT_LEFT/RIGHT`

Have I missed something?

ali-rostami, 2020-01-17:
Thanks Gabor

> Have I missed something?

So great, no I don't think so.
Let me know, when I can test it or if I can help.

kisvegabor, 2020-01-17:
Let's wait until I merge the "new styles" update to `dev-7.0` to avoid conflicts.
It should be done in a 2 weeks.

embeddedt, 2020-02-07:
Not stale; work on the style system was delayed a bit as mentioned [here](https://github.com/littlevgl/lvgl/issues/1243#issuecomment-583383453).

kisvegabor, 2020-02-29:
Not stale

davidjade, 2020-04-04:
Something I would like to see is support for binding to externally stored data. In other words, have the chart object not store the data points itself but rather ask either via a callback or iterator for the data points. 

The issue I have is that I end up with two copies of the data, one from the logger part of my app and another copy in the chart UI object. Sometimes I do have to translate to point data types for display but any callback could do that on the fly as well. Whether the callback just asks for point by point or a small buffer of points would have to be decided. If a small buffer then it should also be provided that it's size can be set, again to control memory usage.

This would also perhaps fit in with the other suggestion to allow for charts to scroll (I would add zooming an axis scale as well to that feature suggestion).

kisvegabor, 2020-04-06:
@davidjade, thank you for sharing your thought!

As you mentioned, it seems it really could you use the API then than scroll/zoom. 

This API could work:
```c
lv_chart_set_ext_y_cb(chart, series, my_callback);
...
int32_t y = lv_chart_get_ext_y(chart, series, id);
```

What do you think?

davidjade, 2020-04-06:
@kisvegabor,

Yes, I think that would work - asking for the point data as needed. 

I do wonder however if something like this though might be useful:

```c
lv_chart_set_ext_y_cb_buffer(chart, lv_coord_t* buffer, int size);
lv_chart_set_ext_y_cb(chart, series, my_callback);
...
my_callback(chart, series, start, len);
```

The idea being, it may be more CPU efficient to request a slice of data at at time vs. a callback for every point for each series. So the lvgl chart code would request chunks at a time and the user code would fill the buffer. This is likely most useful when doing line or area charts where the callback feature could potentially need a lot of data points.

So first you'd tell the chart here's the one buffer to use for all callback requests, then the callback just notifies user code to fill the buffer with what the chart needs. The user code could then choose to have a buffer big enough for everything or only smaller chunks if that makes sense. The drawing code would have to request data as needed, potentially multiple chunks but far less frequently than per-point. The user code could then have some control over the CPU vs memory balance.

Maybe this is overkill but in my case, I intend to use a 800x480 display and on a wide Line chart with a few series and a lot of points, that could be a lot of callbacks, especially if I were trying to scroll/zoom.

kisvegabor, 2020-04-07:
Sounds reasonable! Let's do it this way!

kisvegabor, 2020-04-07:
Unfortunately, I need to say that I can deal with these features only in 7.1. We need to announce feature stops at some point for v7.0 else it'd be never released.

ali-rostami, 2020-04-07:
> Unfortunately, I need to say that I can deal with these features only in 7.1. We need to announce feature stops at some point for v7.0 else it'd be never released.

Oh, really. I've been waiting for these features for a long time. 🤕
Can't you do something for it?🙏🙏🙏

kisvegabor, 2020-04-07:
What would be the most important for you?

ali-rostami, 2020-04-07:
Almost all of them.
Which ones do you think are hard to implement?

embeddedt, 2020-04-07:
> maybe rework the existing chart types?

We've done some work on this already in 7.0.

> support different series types on one chart

The current workaround is to stack chart objects on top of each other. Modifying the chart to allow types to be specified for each series should be reasonably straightforward.

> X/Y mode

I can't remember what feature this was. @ali-rostami Do you know?

ali-rostami, 2020-04-07:
>I can't remember what feature this was. @ali-rostami Do you know?
As I mentioned above, I've no idea what it is, but Gabor said I can handle it.

kisvegabor, 2020-04-09:
> X/Y mode
I can't remember what feature this was. @ali-rostami Do you know?

It's like this (also known as "Scatter"):
[image]
Now the chart is divided horizontally into equal portions.  However, e.g. if you have time+temperature points (not necessarily measure with equal interval) it might be useful to set the X coordinate too. Or you measure Voltage and Current pairs and want to display Voltage on X and the related Current on the Y-axis.

---

In general, my concern is that we can do something quickly but it'd be better to have enough time to design these features carefully. 
I've collect my first thoughts about the mentioned features. 

Scroll:
- Besides the current `point_num` we should add `visible_point_num` too
- We should have an offset (in pixel unit) to keep track of the current scroll position.
- We need to update the offset based on the user's scroll action.
- Calculate the first point based on the offset
- Implement "on-demand data read"
- We should move the ticks too.
-It would be great to automatically calculate the axis labels on scroll. E.g. on X-axis now we have `0 10 20 30`. After a big scroll, it should be `60 70 80 90`. Now, these labels come from a static text. 

Zoom:
- It should change `visible_point_num` and `y_min`, `y_max` automatically.
- Ticks and axis labels should be automatically updated here too.
- It would be nice to avoid non-human-friendly zooms. E.g. somehow force x2, x5, x10, x20 ... values to keep the axis labels "normal". Or shall we allow arbitrary values too?

X/Y support:
- We need an `x_min` and `x_max` too.
- Here can be any number of point not only `point_num`

On-demand data load:
- The above-mentioned prototype should work
- We need to know the  number of available points for scroll and zoom
- Some internal functions would be great to hide how the point is read 

2 Y-axis:
- It seems straightforward. 
- Add `y_min_right`, `y_max_right`
- Add `lv_chart_set_series_y_axis(chart, ser, LV_CHART_Y_LEFT/RIGHT)`
 
`LV_CHART_UPDATE_MODE_SHIFT_LEFT/RIGHT`
- It seems also straightforward. 
- It could be added implemented in `lv_chart_set_next`

Let me know if you see them differently. We can prioritize them based on importance and complexity. Please also let me know if you can contribute to any of them.

ali-rostami, 2020-04-10:
So great details, thanks.
For the zoom part, I think the user should be able to enable horizontal, vertical or both kind of zoom.
And maybe there should be an array for the zoom values which is given by the user. For example for vertical zoom the values might be:
 float vertical_zoom_factors[5] = {1, 1.5, 2, 2.5, 3}; 
And with a function, it will be set.
What do you think?

kisvegabor, 2020-04-10:
Sounds good. Only a minor thing that we don't use float, but it's can be simply in 0.1 unit. So your example would look like this ` {10, 15, 20, 25, 30}`

kisvegabor, 2020-04-10:
One more thing about including these features in v7.0 or not: The new plans are that we will make more frequent releases and merge the new features more often to `master`. So even if these things will be delayed you don't have to wait a half year for them.

ali-rostami, 2020-04-10:
OK so great.
For the help, I'm not sure which one I can implement but I will check and see if I can do anything.
I haven't yet tested the ver 7 branch, I think at first I should set it up first. 
The above list you created is great. If any of them implemented please check them so then others put their time on other options.

kisvegabor, 2020-04-12:
> I haven't yet tested the ver 7 branch, I think at first I should set it up first.

I agree. Let us know if you stuck with something.

> The above list you created is great. If any of them implemented please check them so then others put their time on other options.

Okay, I will. Still there is a lot of things to do with v7. (examples, documentation, demos etc) but it'd be great to finish the majority of things in April to have a release candidate for 1 May.

pete-pjb, 2020-06-11:
Hi @kisvegabor,

Hope you are well!

I have a need to modify the 'lv_chart' object to accommodate my application and I believe what I am trying to achieve may fit in with some of the plans here.

I have a 24 hour rolling graph which goes from 00:00 Hrs to 23:59 hours with 1440 points and I need to update a point every minute.  Currently there is no API method to set the starting point in the series so for it to work I would effectively have to wait for the first occurrence of midnight to start the chart after start-up if that makes sense?
The data points I am plotting are stored in an existing memory buffer in RAM and updated elsewhere in my application.

I see a couple of options for how to deal with this scenario:

### 1) Add an API method which allows the use of an external buffer pointer for the data.
 (This will also reduce my memory usage and efficiency for some other charts in my system which are currently using RAM buffers updated by DMA and then the points are copied to the chart every 500mS.)   
It is similar to what @davidjade mentioned above, but much simpler.  If the data source is updated by DMA or a logger or what ever and the chart points are attached directly to the data source via a pointer the data source processing can trigger or call directly the ```lv_chart_refresh();``` function after it has completed it's update of the buffer.  This method is very simple and keeps the complexity and overheads of selective buffer updates out of LVGL and negates the need for callbacks,  I am not sure it should be LVGL's responsibility to request data for a chart, in my opinion the chart should be 'dumb' and display the data in the buffer it is pointing at and the application should be responsible for delivering the data in anyway it feels fit to the buffer.

I propose we would modify the function:
```c
void lv_chart_set_point_count(lv_obj_t * chart, uint16_t point_cnt);
```
To be:
```c
 void lv_chart_set_point_count(lv_obj_t * chart, uint16_t point_cnt, lv_coord_t *ext_points );
```
We would then set ext_points to NULL if we wanted to use the chart as it is now. Or provide a external allocated buffer which will be used instead.  With out delving to deeply into the existing code right now, there is likely  to be the need for a bit flag to indicate an external buffer has been defined also.

### 2) Add an API method to set the Y start point in the data to something other than point 0.

For this I propose a function similar to this:
```c
void lv_chart_set_y_start_point(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t y);
```
I think this is self explanatory.

I need to create a solution to this as soon as possible, so I am volunteering to take on the required changes,  but I would prefer to do things in a way that fits with the bigger plan.  Feel free to comment and bring forth any suggestions.  Maybe both of these options are viable features?

I look forward to any feedback, feel free to shoot me down if I am way of base here! 😃 

Kind Regards,

Pete

embeddedt, 2020-06-11:
I'd prefer the first option (it sounds more flexible), however, the second option also sounds useful.

kisvegabor, 2020-06-11:
Hi Pete,

> 1) ...This method is very simple and keeps the complexity and overheads of selective buffer updates out of LVGL and negates the need for callbacks, I am not sure it should be LVGL's responsibility to request data for a chart, in my opinion the chart should be 'dumb' and display the data in the buffer it is pointing to and the application should be responsible for delivering the data in anyway it feels fit to the buffer.

The callback can especially useful for the zoom and scroll features. Let's say you store 1 M samples in an external flash or SD card. But only a small range is displayed and you scroll the chart horizontally. The chart needs to get the data on the fly. 
If the chart is zoomed out, probably some points are discarded, hence it's not trivial which points should be displayed. 
The callback also allows making some formatting/scaling on the points on the fly. 

> void lv_chart_set_y_start_point(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t y);

To set the first point horizontally, it should be rather 
`void lv_chart_set_x_start_point(lv_obj_t * chart, lv_chart_series_t * ser, uint32_t id);`, isn't it?

If so it's quite easy to implement as we already have a `start_point` field in the series. 

Besides, we can also add a function to use an external array `lv_point_t` array but I think a dedicated function would be clearer for it.  E.g. `lv_chart_set_ext_array(chart, ser, array)`

I've just found that an `lv_chart_set_point_id(chart, ser, value, id)` function to directly set a given point in a series still would be reasonable to have.

pete-pjb, 2020-06-11:
Hi @kisvegabor ,

Thank you for responding quickly.  

> The callback can especially useful for the zoom and scroll features. Let's say you store 1 M samples in an external flash or SD card. But only a small range is displayed and you scroll the chart [...]

Okay I can see your point here to some degree here.
 I have actually implemented a limited zoom on one of my charts, it only zooms in on the centre of the x-axis as that is the point of interest for my application( RF Spectral Plot). I do it externally to the chart object by reformatting the x-axis on the fly and changing the number of points in the chart, this is selected with a +/- button overlaid on the corner of the chart.  
I see for scrolling and gesture zooming from the touch screen you might want to do things the way you have suggested.  I also acknowledge the potential need to pull data from an external storage medium, but I would still have fears about the performance of pulling data directly from flash memory. I expect it could create a noticeable glitch in the GUI.  If I were doing this type of thing I would have the data loaded up into RAM ahead of time, with that said though in a RAM limited system this may not be possible.  I guess If the callbacks are in place then it gives the user a chance to tackle the problem how ever they feel fit.  I wont be able to spend time on this part right now but if it is still waiting at a later date I may be able to get involved if need be.

> To set the first point horizontally, it should be rather
> `void lv_chart_set_x_start_point(lv_obj_t * chart, lv_chart_series_t * ser, uint32_t id);`, isn't it?

Oops, 😊 Yes you are correct that was my mistake... Doh!

Note: I believe the indices for points in the charts are uint16_t.

So in summary based on your reply I am happy to go ahead and add the following functions:

```c
void lv_chart_set_x_start_point(lv_obj_t * chart, lv_chart_series_t * ser, uint16_t id);

void lv_chart_set_ext_array(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t array[] );

void lv_chart_set_point_id(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t value, uint16_t id);
```

Would you like me to work on it and create a pull request to the dev branch when I am done?

Kind Regards,

Pete

davidjade, 2020-06-11:
I still prefer the callback as I suggested. As other have noted, it gives quite a lot of flexibility and supports things such as scrolling and zooming. My main goal initially however was to be able to remove duplication of data that LVGL buffers. If I am logging data then I am already storing and managing it somewhere and that might be in a format other than what LVGL needs (for instance floating point). Better to have LVGL ask for the data it needs in the format it needs rather than storing it a second time in another buffer just for charts. 

Any method that requires setting a buffer of a fixed data type for LVGL to use potentially runs into this scenario any time the native data has to be translated for LVGL's use. I think it best to translate on the fly as needed rather than duplicate in multiple data formats. And with the callback method the programmer can still make the memory vs. CPU trade off if they have the memory to spare as they can still double buffer in LVGL data types if they choose to.

kisvegabor, 2020-06-12:
@pete-pjb 
> So in summary based on your reply I am happy to go ahead and add the following functions:

Looks good! A PR would be welcome :slightly_smiling_face: 

@davidjade 
I agree, the callback would give great flexibility. If you read from an internal buffer in RAM the performance drop should be negligible. 

--- 

Let's reiterate to see what are the ideas/task here:
- Easily doable, no careful planning is required:
  - A different range for y-axises and assign a series to the primary or secondary y-axis
  - X/Y mode (scatter) support
- Not that trivial but already see the light
  - Cursor management: Get the coordinates and value of the nearest data point from a touched point to draw an infobox there
  - Add callback support. IMO for simplicity, at first, reading only one data at once could be enough
  - Dynamic tick label generation e.g. from start value and step. (required for zoom and scroll).
- Tough topics:
  - Scroll: Scroll points, ticks, tick labels, division lines
  - zoom: Scroll points, ticks, tick labels, division lines, policy to drop points if they are crowded

I can do some from the beginning of the list next week. 
Do you feel linking to implement one or more?

pete-pjb, 2020-06-12:
Hi @kisvegabor,

No problem I will get started on this today....

Thanks,

Pete

pete-pjb, 2020-06-12:
Hi @kisvegabor,

Is there any reason why we have to use an array to assign an external buffer to the series, can it just be a lv_coord_t pointer and we set the length parameter with the existing ```lv_chart_set_point_count(lv_obj_t * chart, uint16_t point_cnt) ``` and I'll add a flag to the ```lv_chart_series_t``` structure to indicate we are using an external buffer so the existing code can be adapted to conditional carry out the correct operations depending on the state of the flag which will be set when an external buffer is assigned.

So the function becomes:
```c
void lv_chart_set_ext_buffer(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t *buffer );
```
The reason I ask is, the only advantage I see in using the array declaration is you can get the size of it, but what I am proposing is to make it the users responsibility to set the point count which they have to do currently anyway...

Or we could make the function:
```c
void lv_chart_set_ext_buffer(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t *buffer, 
           uint16_t buf_size );
```
with this approach we could save the user a call to ```lv_chart_set_point_count(lv_obj_t * chart, uint16_t point_cnt) ``` as I could take care of it inside the library.

In my own application I do not declare arrays at all and for example when I do my external zoom I would just change the pointer to a different section of the same data buffer and set the point count to a lesser value. If that makes sense?

I don't know if @embeddedt might want to comment here also? :smiley:

What do you think?

Kind Regards,

Pete

embeddedt, 2020-06-12:
You can't get the size from an array declaration anyways; it just decays into a pointer inside the function. This "feature" is one of the confusing things about C.  That means you'd need a size parameter anyways (although I'd suggest length instead, i.e. the number of `lv_coord_t` values).

pete-pjb, 2020-06-12:
Hmmm, thanks for the heads up on the arrays @embeddedt I didn't realise that, good to know.  I really don't use them very much at all myself.  
Perhaps the function parameter should be named better for clarity as you have indicated, what about: 

```c
void lv_chart_set_ext_buffer(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t *buffer, 
           uint16_t point_count );
```
Do you think that works?

embeddedt, 2020-06-12:
That looks fine to me.

embeddedt, 2020-06-12:
One thing: I forget what the convention is for MicroPython. Even though using `lv_coord_t []` vs `lv_coord_t *` in a C function declaration are identical, the MicroPython binding script takes advantage of that to distinguish whether the value is a pointer to one element or an array. Perhaps @kisvegabor remembers which needs to be used. We can always change it after the function is implemented, as it shouldn't affect the code at all.

pete-pjb, 2020-06-12:
Okay thanks @embeddedt

kisvegabor, 2020-06-15:
> remembers which needs to be used. We can always change it after the function is implemented, as it shouldn't affect the code at all.

As here we want to pass an array I think `lv_coord_t []` should be used.

pete-pjb, 2020-06-15:
Noted, thanks @kisvegabor

kisvegabor, 2020-06-26:
I've added the feature of assigning series to the primary or secondary y-axis.

It can be used like this:
```c
    lv_chart_set_series_axis(chart, ser, LV_CHART_AXIS_SECONDARY_Y);
    lv_chart_set_y_range(chart, LV_CHART_AXIS_SECONDARY_Y, -1000, 500);
```

---

I've updated my first comment in this issue with a summary of the discussed ideas.

To keep the list of issues clean I close this for now. If someone is interested in contributing to any of them feel free to comment here or open a dedicated issue.

closed 2020-06-26 by kisvegabor

ali-rostami, 2020-08-18:
Hi
so can we have the scroll feature now?
@kisvegabor would you please update this ideas list. thanks.

> Summary of ideas:
> 
> * Easily doable, no careful planning is required:
>   
>   * [x]  A different range for y-axises and assign a series to the primary or secondary y-axis
>   * [x]  Series's data from an external array
>   * [ ]  X/Y mode (scatter) support
>   * [ ]  LV_CHART_UPDATE_MODE_SHIFT_LEFT
> * Not that trivial but already see the light
>   
>   * [ ]  Cursor management: Get the coordinates and value of the nearest data point from a touched point to draw an infobox there
>   * [ ]  Add callback support. IMO for simplicity, at first, reading only one data at once could be enough
>   * [ ]  Dynamic tick label generation e.g. from start value and step. (required for zoom and scroll).
> * Tough topics:
>   
>   * [ ]  Scroll: Scroll points, ticks, tick labels, division lines
>   * [ ]  zoom: Scroll points, ticks, tick labels, division lines, policy to drop points if they are crowded
> 
> If you are interested in contributing any of them feel free to comment here or open an issue for it.

kisvegabor, 2020-08-20:
@ali-rostami Update with what? Scrolling is already there.

ali-rostami, 2020-08-20:
updating the ticks, which features are now implemented.

> Scrolling is already there.
Oh, really. I couldn't find anything about it in documents. would please help me with it, or I can open an issue in forum if you prefer.

kisvegabor, 2020-08-24:
@ali-rostami 

Sorry, I meant the "scrolling" is already in the list. But it's still not implemented... :slightly_frowning_face:

ali-rostami, 2020-08-24:
O.K
So the list is still valid. right?
I thought that the update @pete-pjb implemented could help to this.
Anyway, I hope, I can work on cursor soon.
best regard

kisvegabor, 2020-08-25:
> I thought that the update @pete-pjb implemented could help to this.

Unfortunately, it was another topic. :( 

> Anyway, I hope, I can work on cursor soon. If there is any idea about it, let me know. thanks

Great, please open a new issue for it if you have time to deal with it.

[1 more reply]