lvgl/lvgl #1340: img_flower_icon not working properly with 16-bit color depth

embeddedt, 2020-01-04:
**Describe the bug**

`img_flower_icon` appears to be broken when using an `LV_COLOR_DEPTH` of 16. I haven't tried any other images, but I would expect they have the same problem.

**To Reproduce**

Create an image object on the PC simulator configured with an `LV_COLOR_DEPTH` of 16. Set the image object's source to `&img_flower_icon`.

`lv_test_img_1` also reproduces the problem effectively.

**Expected behavior**

The flower icon draws properly.

**Screenshots**

[image]

**Additional context**

The image draws normally if I instead choose an `LV_COLOR_DEPTH` of 32.

This happens with both 6.1 and dev-7.0.

embeddedt, 2020-01-04:
I think I must have forgotten something basic. :slightly_smiling_face: I was doing some rudimentary bisecting and I found the same problem in 6.0.0. There's no way no one else has run into this problem in the last 6 months! Trying to think of what I've forgotten.

embeddedt, 2020-01-04:
I think I figured it out. I don't think it's a LittlevGL issue after all.

If I reconvert the image with the same settings, it seems to work properly on both color depths. I think that `img_flower_icon` may just have been outdated, so I will push a newly converted copy to `master` on `lv_examples`.

Whew!

closed 2020-01-04 by embeddedt

kisvegabor, 2020-01-04:
Great! 
Although, I wonder how can it be outdated as the indexed color format hasn't changed for a while. 
But nevermind, probably it's not worth the effort to investigate what has been changed.