lvgl/lvgl #1343: Some unfixed [-Wnarrowing] warnings

puzrin, 2020-01-06:
**Describe the bug**

I see some warnings from `lvgl.h` after switch to latest master HEAD. Note, i did not changed project settings. Only updated dependency.

```
In file included from .pio/libdeps/hardware_usb/lvgl/src/lv_hal/lv_hal_disp.h:21:0,
                 from .pio/libdeps/hardware_usb/lvgl/src/lv_hal/lv_hal.h:16,
                 from .pio/libdeps/hardware_usb/lvgl/lvgl.h:24,
                 from src/app.h:7,
                 from src/screen_dose.cpp:1:
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h: In function 'lv_color_t lv_color_make(uint8_t, uint8_t, uint8_t)':
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h:474:60: warning: narrowing conversion of '(((unsigned int)(((int)g) >> 5)) & 7)' from 'unsigned int' to 'uint16_t {aka short unsigned int}' inside { } [-Wnarrowing]
 #define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{(g8 >> 5) & 0x7U, (r8 >> 3) & 0x1FU, (b8 >> 3) & 0x1FU, (g8 >> 2) & 0x7U}})
                                                  ~~~~~~~~~~^~~
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h:482:12: note: in expansion of macro 'LV_COLOR_MAKE'
     return LV_COLOR_MAKE(r, g, b);
            ^~~~~~~~~~~~~
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h:474:78: warning: narrowing conversion of '(((unsigned int)(((int)r) >> 3)) & 31)' from 'unsigned int' to 'uint16_t {aka short unsigned int}' inside { } [-Wnarrowing]
 #define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{(g8 >> 5) & 0x7U, (r8 >> 3) & 0x1FU, (b8 >> 3) & 0x1FU, (g8 >> 2) & 0x7U}})
                                                                    ~~~~~~~~~~^~~
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h:482:12: note: in expansion of macro 'LV_COLOR_MAKE'
     return LV_COLOR_MAKE(r, g, b);
            ^~~~~~~~~~~~~
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h:474:97: warning: narrowing conversion of '(((unsigned int)(((int)b) >> 3)) & 31)' from 'unsigned int' to 'uint16_t {aka short unsigned int}' inside { } [-Wnarrowing]
 #define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{(g8 >> 5) & 0x7U, (r8 >> 3) & 0x1FU, (b8 >> 3) & 0x1FU, (g8 >> 2) & 0x7U}})
                                                                                       ~~~~~~~~~~^~~
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h:482:12: note: in expansion of macro 'LV_COLOR_MAKE'
     return LV_COLOR_MAKE(r, g, b);
            ^~~~~~~~~~~~~
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h:474:116: warning: narrowing conversion of '(((unsigned int)(((int)g) >> 2)) & 7)' from 'unsigned int' to 'uint16_t {aka short unsigned int}' inside { } [-Wnarrowing]
 #define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{(g8 >> 5) & 0x7U, (r8 >> 3) & 0x1FU, (b8 >> 3) & 0x1FU, (g8 >> 2) & 0x7U}})
                                                                                                          ~~~~~~~~~~^~~
.pio/libdeps/hardware_usb/lvgl/src/lv_hal/../lv_misc/lv_color.h:482:12: note: in expansion of macro 'LV_COLOR_MAKE'
     return LV_COLOR_MAKE(r, g, b);
            ^~~~~~~~~~~~~
```

**To Reproduce**

Use `-Wnarrowing` build option.

**Expected behavior**

`lvgl.h` should hot cause warnings

puzrin, 2020-01-06:
BTW: i think it would be nice to test that. Create a test with dummy file, including header only, and build with max warnings enabled and fail on any.

embeddedt, 2020-01-06:
Please try again with the latest `master`.

puzrin, 2020-01-06:
Fixed, thanks.

closed 2020-01-06 by puzrin