lvgl/lvgl #1344: Memory leak - lv_img

nwsetec, 2020-01-06:
**Describe the bug**
When repeatedly creating a lv_img object, rendering it and lv_obj_clean() the default screen on a button click a memory leak seems to be happening and eventually memory cannot be allocated.

**To Reproduce**
I've created a branch with a commit to reproduce the issue here:
https://github.com/nwsetec/lv_sim_eclipse_sdl.git

**Expected behavior**
If the button is clicked repeatedly eventually no memory can be allocated.

**Screenshots**
[image]

**Additional context**
LittlevGL is configured for 1bit pixel depth.
Note that if the object isn't rendered then the memory leak doesn't happen, image cache issue?

kisvegabor, 2020-01-06:
Hi,

I've found the issue. It was related to indexed images. A part of the palette was not freed. 
It should be fixed in f31ba6cf

nwsetec, 2020-01-07:
Thanks Gabor!  This issue is now fixed for me.

closed 2020-01-07 by nwsetec

kisvegabor, 2020-01-09:
Great! :slightly_smiling_face: