lvgl/lvgl #1345: PNG issues for Online Image Converter

imosts, 2020-01-07:
# All enhancement requests or questions should be directed to the Forum.

Hi, I use the Online Image Converter to convert a PNG image with alpha channel, the transparent part of the image is not transparent after conversion, but black. 
The color format I chose is **True color with alpha**, and output format I chose is **C array**.
I don't know if there is a problem with this PNG file format or with the Online Image Converter.

This is the PNG image: [https://drive.google.com/file/d/14G0EYui4DUurA0jlo1o05xR1zc86BOxg/view?usp=sharing](url)

This is the C array:
[https://drive.google.com/file/d/1DEHbeNHWeu-oLflvxPtgMyTZrt_YYPmh/view?usp=sharing](url)

Thanks,
imosts

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

kisvegabor, 2020-01-09:
Hi,

The image had indexed mode. I've opened it with GIMP and in Image/Mode I've changed it RGB. Now it works for me.

Here is the corrected image: 
[image]

imosts, 2020-01-10:
> Hi,
> 
> The image had indexed mode. I've opened it with GIMP and in Image/Mode I've changed it RGB. Now it works for me.
> 
> Here is the corrected image:
> [page_indicator.png.zip](https://github.com/littlevgl/lvgl/files/4041542/page_indicator.png.zip)

That great! Thank you very much!

closed 2020-01-10 by imosts