lvgl/lvgl #1346: CMake build for the library and examples repository

ValentiWorkLearning, 2020-01-07:
Hi
As the @kisvegabor suggested, I opened this issue to talk about the CMake-based build for the library or/and simulator repository.
Here is the list of questions:
- CMakeLists for lvgl library

- CMakeLists for examples

- CMakeLists for simulator( this achieves the possibility of using IDE's that support CMakeLists such as Qt Creator, CLion and other )

I've provided the sample of CMake-based bulild of Simulator in my repository:
https://github.com/ValentiWorkLearning/GradWork/tree/dev/lvgl_implementation/Simulator
https://github.com/ValentiWorkLearning/GradWork/blob/dev/lvgl_implementation/Simulator/graphics/CMakeLists.txt

So, if these options are acceptable, I can start from implementation of Simulator CMake-build.

puzrin, 2020-01-09:
Cmake is instrument, not goal. Could you explain goal you wish to acheive?

ValentiWorkLearning, 2020-01-09:
@puzrin I just want to achieve cross-platform build of LVGL without looking for any IDE-dependent and OS - dependent solutions. Also, it'll achieve possibility of using lvgl as subtarget of CMake-based projects.

ValentiWorkLearning, 2020-01-09:
@puzrin also, CMake target for lvgl achieve auto-rebuild, integration with syntax navigators in popular IDE's and other cool things that can be achieved with single consistency of project build hierarchy.

kisvegabor, 2020-01-09:
@ValentiWorkLearning 
We already have [CMakeList.txt](https://github.com/littlevgl/lv_sim_eclipse_sdl/blob/master/CMakeLists.txt) in [lv_sim_eclipse_sdl](https://github.com/littlevgl/lv_sim_eclipse_sdl) project. 

The current CMake file seems simpler because it doesn't require to list all the files. However, I'm not too familiar with CMake so I'm not aware of the advantages and disadvantages of the solutions. 

Instead of creating a new project we should add it to one of the existing simulators. `lv_sim_eclipse_sdl` is the most popular one, so I vote for this. Is it ok for you too?

ValentiWorkLearning, 2020-01-09:
@kisvegabor Yes, I agree with you, that CMakeLists should be added for one of existing simultors. Eclipse is acceptable for me. I'll try to fork and implement it.

kisvegabor, 2020-01-10:
Great! :+1:

ValentiWorkLearning, 2020-01-13:
@kisvegabor it's better to implement CMake for lvgl 7.0 dev branch or for 6.1? Suppose it'll be several PR's - to LVGL main repository, to driver's repository, to example's and to Eclipse simulator.

embeddedt, 2020-01-13:
Please send it to dev-7.0. 6.1 only receives bugfixes.

ValentiWorkLearning, 2020-01-14:
@kisvegabor  @embeddedt sterted implementation here:
https://github.com/ValentiWorkLearning/lvgl/commit/eeef10cefec31cc0414864664a4f9a163d42bd32
probably change to separate CMakeLists for each subfolder( lv_core, lv_drawing,lv_font,lv_msic).
Commits for example will be as soon as possible.

kisvegabor, 2020-01-15:
@ValentiWorkLearning 
Thanks for the update!

Would it be possible to automatically search the source files instead of directly enumerate them?

ValentiWorkLearning, 2020-01-15:
@kisvegabor suppose it's possible. I'll try to use the  ```aux_source_directory``` cmake command. That's a right way to supply the library.

kisvegabor, 2020-01-17:
Great, thank you!

closed 2020-02-14 by stale[bot]