lvgl/lvgl #1347: Lots of errors after including lvgl/lvgl.h

JimmyTheNerd, 2020-01-07:
**Describe the bug**

I'm trying to compile a very basic example, but get errors (see below) when I include the library into my file and run `gcc gui.c`. Thanks in advance.

**To Reproduce**

Folder structure:
```
project/
    lvgl/    
    gui.c
    lv_conf.h    
```
gui.c:
```
#include "lvgl/lvgl.h"
 
int main()
{	 
    return 0;
}
```

OS: Raspbian

**Expected behavior**

No output, no errors. 

**Additional context**

Log from the terminal:
```
$ gcc gui.c 
In file included from lvgl/lvgl.h:19:0,
                 from gui.c:5:
lvgl/src/lv_misc/lv_log.h:29:28: error: expected declaration specifiers or ‘...’ before numeric constant
 #define LV_LOG_LEVEL_TRACE 0 /**< A lot of logs to give detailed information*/
                            ^
lvgl/src/lv_misc/lv_log.h:36:21: note: in expansion of macro ‘LV_LOG_LEVEL_TRACE’
 LV_EXPORT_CONST_INT(LV_LOG_LEVEL_TRACE);
                     ^~~~~~~~~~~~~~~~~~
lvgl/src/lv_misc/lv_log.h:30:27: error: expected declaration specifiers or ‘...’ before numeric constant
 #define LV_LOG_LEVEL_INFO 1  /**< Log important events*/
                           ^
lvgl/src/lv_misc/lv_log.h:37:21: note: in expansion of macro ‘LV_LOG_LEVEL_INFO’
 LV_EXPORT_CONST_INT(LV_LOG_LEVEL_INFO);
                     ^~~~~~~~~~~~~~~~~
lvgl/src/lv_misc/lv_log.h:31:27: error: expected declaration specifiers or ‘...’ before numeric constant
 #define LV_LOG_LEVEL_WARN 2  /**< Log if something unwanted happened but didn't caused problem*/
                           ^
lvgl/src/lv_misc/lv_log.h:38:21: note: in expansion of macro ‘LV_LOG_LEVEL_WARN’
 LV_EXPORT_CONST_INT(LV_LOG_LEVEL_WARN);
                     ^~~~~~~~~~~~~~~~~
lvgl/src/lv_misc/lv_log.h:32:28: error: expected declaration specifiers or ‘...’ before numeric constant
 #define LV_LOG_LEVEL_ERROR 3 /**< Only critical issue, when the system may fail*/
                            ^
lvgl/src/lv_misc/lv_log.h:39:21: note: in expansion of macro ‘LV_LOG_LEVEL_ERROR’
 LV_EXPORT_CONST_INT(LV_LOG_LEVEL_ERROR);
                     ^~~~~~~~~~~~~~~~~~
lvgl/src/lv_misc/lv_log.h:33:27: error: expected declaration specifiers or ‘...’ before numeric constant
 #define LV_LOG_LEVEL_NONE 4 /**< Do not log anything*/
                           ^
lvgl/src/lv_misc/lv_log.h:40:21: note: in expansion of macro ‘LV_LOG_LEVEL_NONE’
 LV_EXPORT_CONST_INT(LV_LOG_LEVEL_NONE);
                     ^~~~~~~~~~~~~~~~~
In file included from lvgl/src/lv_hal/lv_hal_disp.h:21:0,
                 from lvgl/src/lv_hal/lv_hal.h:16,
                 from lvgl/lvgl.h:24,
                 from gui.c:5:
[... 341 lines cut ...]
lvgl/src/lv_draw/lv_img_decoder.h:174:5: error: unknown type name ‘lv_img_decoder_read_line_f_t’
     lv_img_decoder_read_line_f_t read_line_cb;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
lvgl/src/lv_draw/lv_img_decoder.h:260:63: error: unknown type name ‘lv_coord_t’
 lv_res_t lv_img_decoder_read_line(lv_img_decoder_dsc_t * dsc, lv_coord_t x, lv_coord_t y, lv_coord_t len,
                                                               ^~~~~~~~~~
lvgl/src/lv_draw/lv_img_decoder.h:260:77: error: unknown type name ‘lv_coord_t’
 lv_res_t lv_img_decoder_read_line(lv_img_decoder_dsc_t * dsc, lv_coord_t x, lv_coord_t y, lv_coord_t len,
                                                                             ^~~~~~~~~~
lvgl/src/lv_draw/lv_img_decoder.h:260:91: error: unknown type name ‘lv_coord_t’
 lv_res_t lv_img_decoder_read_line(lv_img_decoder_dsc_t * dsc, lv_coord_t x, lv_coord_t y, lv_coord_t len,
                                                                                           ^~~~~~~~~~
lvgl/src/lv_draw/lv_img_decoder.h:300:66: error: unknown type name ‘lv_img_decoder_read_line_f_t’
 void lv_img_decoder_set_read_line_cb(lv_img_decoder_t * decoder, lv_img_decoder_read_line_f_t read_line_cb);
                                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
lvgl/src/lv_draw/lv_img_decoder.h:339:100: error: unknown type name ‘lv_coord_t’
 lv_res_t lv_img_decoder_built_in_read_line(lv_img_decoder_t * decoder, lv_img_decoder_dsc_t * dsc, lv_coord_t x,
                                                                                                    ^~~~~~~~~~
lvgl/src/lv_draw/lv_img_decoder.h:340:51: error: unknown type name ‘lv_coord_t’
                                                   lv_coord_t y, lv_coord_t len, uint8_t * buf);
                                                   ^~~~~~~~~~
lvgl/src/lv_draw/lv_img_decoder.h:340:65: error: unknown type name ‘lv_coord_t’
                                                   lv_coord_t y, lv_coord_t len, uint8_t * buf);

```

embeddedt, 2020-01-07:
What are the contents of your `lv_conf.h`? It appears that one of the issues is that the configurable `LV_EXPORT_CONST_INT` macro is not defined properly.

kisvegabor, 2020-01-07:
Try to add `#error "lv_conf.h compile test"` to the beginning of `lv_conf.h`to see it's really included into the build.

JimmyTheNerd, 2020-01-07:
My bad: I set `#if 1` in the wrong file. The errors disappeared. 

Thank you very much for your fast responses.

closed 2020-01-07 by JimmyTheNerd

kisvegabor, 2020-01-07:
No problem! :wink:

David-OConnor, 2020-04-17:
Is there a way to make the error more useful? Would be nice if instead of that (Which is the longest I've seen in quite a while, if not ever). Ie it could read something to the effect of *Unable to start lvgl. Did you change `#if 0`  to `#if  1` in `lv_conf.h` ?

embeddedt, 2020-04-17:
> Could also add this to the Add LittlevGL to your project section in the Readme.

I've added a note about it here: https://github.com/littlevgl/lvgl/commit/9bd86963c9a4f4471bf470d35e28d3873ac8b42d