lvgl/lvgl #1348: [Dev-7.0] Clang compile error

xennex22, 2020-01-09:
Clang is fussy about types in initializer lists.  
```
[bcc32c Error] lv_color.h(547): non-constant-expression cannot be narrowed from type 'unsigned int' to 'std::uint16_t' (aka 'unsigned short') in initializer list
  lv_color.h(547): insert an explicit cast to silence this issue
```
As it suggests, a cast fixes it.
```
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{(b8 >> 3) & 0x1FU, (g8 >> 2) & 0x3FU, (r8 >> 3) & 0x1FU}})
```
```
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{(uint16_t)((b8 >> 3) & 0x1FU), (uint16_t)((g8 >> 2) & 0x3FU), (uint16_t)((r8 >> 3) & 0x1FU)}})
```

embeddedt, 2020-01-09:
I had already patched this in my local copy of the tree; I will push the patch when I work on LittlevGL next.

xennex22, 2020-01-09:
OK, great.

closed 2020-01-09 by xennex22

embeddedt, 2020-01-10:
Patch was applied in https://github.com/littlevgl/lvgl/commit/8f9fb4578c0ca06a9d9f8ed40308146fd58f840e.