lvgl/lvgl #1349: [Dev-7.0] lv_disp_drv_register uninitialized memory

xennex22, 2020-01-09:
In lv_disp_drv_register() the display refresh task is used before it is initialized.

Half way through the function, before disp->refr_task is allocated:
lv_obj_set_style() ->
lv_obj_refresh_style() ->
lv_obj_invalidate() ->
lv_inv_area() ->
lv_task_set_prio(disp->refr_task, LV_REFR_TASK_PRIO);

Moving the refresh task create before the lv_obj_set_style() call fixes it, but I have not looked to see if this breaks anything else.

last_activity_time is uninitialized too but I don't think it matters.

embeddedt, 2020-01-09:
I've applied a patch [here](https://github.com/littlevgl/lvgl/commit/d3d8205b2fb59c43830eb47c9c2c5af821853b7d). Can you confirm that it fixes the problem?

xennex22, 2020-01-09:
Yes, that is exactly what I did.

xennex22, 2020-01-09:
Ok, got another one.

lv_obj adv_hittest is not initialized in lv_obj_create when copying from another object.

Needs this added
```
new_obj->adv_hittest  = 0;
```

kisvegabor, 2020-01-09:
I've fixed it here: bcd4e224
Thank you!

closed 2020-01-09 by xennex22