lvgl/lvgl #1350: [dev-7.0] List alignment is broken?

embeddedt, 2020-01-10:
**Describe the bug**

With dev-7.0, lists don't get centered when you call `lv_obj_align(list, LV_ALIGN_CENTER, 0, 0)`.

**To Reproduce**

```c
lv_obj_t * list = lv_list_create(lv_scr_act(), NULL);
lv_list_add_btn(list, LV_SYMBOL_FILE, "New");
lv_list_add_btn(list, LV_SYMBOL_DIRECTORY, "Open");
lv_obj_align(list, NULL, LV_ALIGN_CENTER, 0, 0);
```

**Expected behavior**

The list should have been centered properly.

**Screenshots**

[image]

kisvegabor, 2020-01-13:
It happens because the scrollable part of the list is transparent by default.

You can make it visible by applying:  `lv_list_set_style(list, LV_LIST_STYLE_BG, &lv_style_plain_color);`
[image]

You can also apply `fit` vertically:
```c
    lv_cont_set_fit2(list, LV_FIT_NONE, LV_FIT_TIGHT);
    lv_obj_align(list, NULL, LV_ALIGN_CENTER, 0, 0);
```
[image]

Or simply make the scrollable transparent instead of the background:
```c
    lv_list_set_style(list, LV_LIST_STYLE_BG, &lv_style_pretty);
    lv_list_set_style(list, LV_LIST_STYLE_SCRL, &lv_style_transp_fit);
```
[image]

embeddedt, 2020-01-13:
> You can also apply fit vertically

Is there any reason why the vertical fit isn't set that way by default? It seems weird that the scrollable is taller than its contents.

kisvegabor, 2020-01-13:
I've chosen this way to make the list looks as small as possible with few elements (similarly to "Right-click menu") but add a scrollbar if there are more elements. 

Using vertical fit by default would prevent to show scrollbars.

Making the background visible and the scrollable transparent would add lots of extra space. But at least it's clearer what happens.

The current solution is quite flexible IMO, but really not so intuitive.

What do you think about which way should we choose?

embeddedt, 2020-01-13:
I think we should make the extra space visible. Otherwise, `LV_ALIGN_CENTER` doesn't visually align the object the way one would expect.

The extra space might be annoying, but if someone doesn't want it they can always call `lv_obj_set_height`, right?

kisvegabor, 2020-01-14:
> The extra space might be annoying, but if someone doesn't want it they can always call lv_obj_set_height, right?

Yes, that true.

Finally, I agree. I'll set the reworked styles as you suggested.

kisvegabor, 2020-02-04:
In `feat-new_style` it works like this.
I can show something from it in a week.

closed 2020-02-04 by kisvegabor

embeddedt, 2020-02-04:
Great; thanks for taking a look at it!