lvgl/lvgl #1351: Crash after touch event

gian72, 2020-01-10:
Hi I'm using micropython porting for ESP32 with ili9341, XPT2046.

**Describe the bug**

After few touches on a text area, system crash

**To Reproduce**
Here's the code, taken from sample on https://docs.littlevgl.com/en/html/object-types/ta.html:
`import lvgl as lv
from ili9341 import ili9341
disp = ili9341(miso=19, mosi=23, clk=18, cs=5, dc=4, rst=22, power=14, backlight=15, mhz=40, factor=4,
               height=240, width=320,rot=ili9341.LANDSCAPE, hybrid=True)
import xpt2046
touch = xpt2046.xpt2046(cs=14,transpose = False,cal_y1 = 3808, cal_x1 = 3664, cal_y0 = 336, cal_x0 = 256)

def event_handler(obj, event):
    print(event)
    if event == lv.EVENT.VALUE_CHANGED:
        print("Value: %s" % obj.get_text())
    elif event == lv.EVENT.LONG_PRESSED_REPEAT:
        # For simple test: Long press the Text are to add the text below
        ta1.add_text("\n\nYou can scroll it if the text is long enough.\n")

ta1 = lv.ta(lv.scr_act())
ta1.set_size(200, 100)
ta1.align(None, lv.ALIGN.CENTER, 0, 0)
ta1.set_cursor_type(lv.CURSOR.BLOCK)
ta1.set_text("A text in a Text Area")     # Set an initial text
ta1.set_event_cb(event_handler)
`

**Expected behavior**
-

**Screenshots**
Following the console output along with crash report:
`>>> %Run -c $EDITOR_CONTENT
ILI9341 initialization completed
Enable backlight
Double buffer
>>> Event received: 0
Event received: 1
Event received: 1
Event received: 1
Event received: 1
Event received: 1
Event received: 3
Event received: 6
Event received: 7
Event received: 12
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x40104a19  PS      : 0x00060530  A0      : 0x8011464a  A1      : 0x3ffccf10  
A2      : 0x400014bf  A3      : 0x3fff14c0  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffccee0  
A10     : 0x00000000  A11     : 0x3ffb00d4  A12     : 0x3f4312d4  A13     : 0x00000023  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x0000001e  EXCCAUSE: 0x00000003  
EXCVADDR: 0x400014bf  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40104a19:0x3ffccf10 0x40114647:0x3ffccf70 0x4011c9c1:0x3ffccfd0 0x4011e123:0x3ffcd020 0x400f94c3:0x3ffcd040 0x400f9d21:0x3ffcd080 0x40105bda:0x3ffcd0d0 0x40105c90:0x3ffcd0f0 0x4014293b:0x3ffcd110 0x400e306e:0x3ffcd130 0x400df06d:0x3ffcd150 0x400df0b1:0x3ffcd170 0x400dff90:0x3ffcd1a0 0x400e005d:0x3ffcd220 0x400e00ab:0x3ffcd240 0x400f4141:0x3ffcd260 0x401425e2:0x3ffcd280 0x400f3fb0:0x3ffcd2b0 0x4008e26d:0x3ffcd2e0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4876
load:0x40078000,len:9652
load:0x40080400,len:6200
entry 0x400806f0
W (314) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
MicroPython v1.9.4-2072-gf89c8c48c-dirty on 2020-01-04; ESP32 module with ESP32
Type "help()" for more information.
>>> 
`

embeddedt, 2020-01-11:
@amirgon

amirgon, 2020-01-11:
Could be related to #1316?
Are you using rshell ? There's a known issue where soft rest causes crashes on lvgl, and rshell is resetting the board for some operations.
If this is the case, try using the REPL directly, or try using single buffer instead of double buffer on ili9341.

Another question: what is printing "Event received: 1"? I didn't see this in your code, and I'm not familiar with such printing on our side.

gian72, 2020-01-11:
Thanks for the replies.
I've tried your suggestions, but crash happens again, that printing is just  to see what event has been reveived:
```
0: PRESSED
1: PRESSING
2: PRESS_LOST
3: SHORT_CLICKED
4: LONG_PRESSED
5: LONG_PRESSED_REPEAT
6: CLICKED
7: RELEASED
8: DRAG_BEGIN
9: DRAG_END
10: DRAG_THROW_BEGIN
11: KEY
12: FOCUSED
13: DEFOCUSED
14: VALUE_CHANGED
15: INSERT
16: REFRESH
17: APPLY
18: CANCEL
19: DELETE
```

amirgon, 2020-01-11:
@gian72 - I tried running your code on my ESP32 device, and it runs without any problem.  

I configured the ili9341 and xpt2046 drivers a bit different on my side, to match my hardware (most parameters are the default values declared on [ili9341](https://github.com/littlevgl/lv_binding_micropython/blob/0d60a7197dd69824fbab860ff53b980590ade786/driver/esp32/ili9341.py#L48-L51) and [xpt2046](https://github.com/littlevgl/lv_binding_micropython/blob/0d60a7197dd69824fbab860ff53b980590ade786/driver/esp32/xpt2046.py#L19) drivers):

```
disp = ili9341(dc=32, cs=33, power=-1, backlight=-1)
touch = xpt2046.xpt2046()
```

Since I cannot reproduce the problems you see, I can only guess that your problem is related either to your configuration (driver parameters, wiring etc) or to your hardware.  

Next step for you is to try to debug it.
- Build micropython in DEBUG configuration
- Parse the Backtrace line with [addr2line](https://sourceware.org/binutils/docs/binutils/addr2line.html). The backtrace can be very helpful
- Debug your device with JTAG. [Here is how I'm doing it](https://github.com/amirgon/ESP32-JTAG).

gian72, 2020-01-11:
Hi Amirgon, thanks for the useful advice, I'll surely try them :-)

gian72, 2020-01-12:
Sorry for the stupid question: how can I build in DEBUG mode? 
I've  tried:
make **DEBUG=1** LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC
or 
make LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1 **-DDEBUG**" BOARD=GENERIC

But I'm not sure the above options change somthing in the produced firmware (size is still the same)

amirgon, 2020-01-12:
> make **DEBUG=1** LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC

That one.  
Make sure you `make clean` before you build in DEBUG mode, because incremental build does not take it into account.

gian72, 2020-01-12:
Here's the stack trace:
```
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_misc/lv_bidi.c:195 (discriminator 1)
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_objx/lv_label.c:732
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_objx/lv_ta.c:1879
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_objx/lv_ta.c:1528
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_core/lv_indev.c:764
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_core/lv_indev.c:668
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_misc/lv_task.c:370 (discriminator 1)
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/lvgl/src/lv_misc/lv_task.c:135
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/lv_bindings/driver/esp32/modlvesp32.c:33
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../py/objfun.c:70
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../py/runtime.c:642
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../py/runtime.c:620
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../py/runtime_utils.c:33
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../py/scheduler.c:72
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../py/scheduler.c:59
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/mphalport.c:68
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/../../lib/utils/pyexec.c:369
/home/gianpaolo_fresia/esp32/lv_micropython/ports/esp32/main.c:127
```

gian72, 2020-01-12:
Ok, at that line (lv_bidi.c:195), probably there's a typo:
```
if (str_out) str_out[len] = '\0';
```
I've tried to comment that out and that type of crash seems disappered :-)
Even if after some more touch interaction I have this other crash with this message:
```
MemoryError: memory allocation failed, allocating 65535 bytes
```

But probably is anouter issue, let me try with other your suggestions :-)

amirgon, 2020-01-12:
That's interesting.
It crashes on this line:
https://github.com/littlevgl/lvgl/blob/248868fef1f4925fada1aa7591b19e3bc837db51/src/lv_misc/lv_bidi.c#L195

In function:
```c
void lv_bidi_process_paragraph(const char * str_in, char * str_out, uint32_t len, lv_bidi_dir_t base_dir, uint16_t *pos_conv_out, uint16_t pos_conv_len)
```

called from here:
https://github.com/littlevgl/lvgl/blob/248868fef1f4925fada1aa7591b19e3bc837db51/src/lv_objx/lv_label.c#L729-L732

There is some memory allocation involved here: `lv_draw_get_buf` should allocate enough ram for `bidi_txt`, but something goes wrong there.
Perhaps you are out of ram?

Try to print out the return value of `lv_draw_get_buf`, or print something in `LV_DEBUG_ASSERT` so you could see if the memory allocation is successful or not.

gian72, 2020-01-13:
I've tried to use lv_mem_monitor to retrieve memory available (before and after lv_draw_get_buf), but I can't get valid information (lv_mem_monitor_t  stuct is always all 0)

BTW, disabling bidi (LV_USE_BIDI at 0), the issue disappers.

amirgon, 2020-01-13:
> I've tried to use lv_mem_monitor to retrieve memory available

You can't use it with micropython. When running lvgl with micropython, lvgl uses micropython's gc. 
 Try `gc.mem_free()`.  
Checking the return value of `lv_draw_get_buf` would be best.

> BTW, disabling bidi (LV_USE_BIDI at 0), the issue disappers.

I'm not convinced the problem is related to bidi, even though the crash was in there.  
If it's not, it could randomly appear elsewhere.

kisvegabor, 2020-01-13:
@amirgon 
While reworking the styles I also faced a crash due to bidi that I just wanted to report.
I see two problems [here](https://github.com/littlevgl/lvgl/blob/master/src/lv_misc/lv_bidi.c#L151-L153):
1.  `uint16_t *pos_conv_buf = (uint16_t*) ((char*)buf + len);` can be an odd address. It might a problem for some architectures to read `uint16_t` from an unaligned address.
2. `len` should be `len+1` becasue [lv_bidi_process_paragraph](https://github.com/littlevgl/lvgl/blob/master/src/lv_misc/lv_bidi.c#L195) set the trailing zero immediatelly. 

I've fixed these [like this](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_misc/lv_bidi.c#L191-L192).

What do you think?

amirgon, 2020-01-13:
@kisvegabor 

> 1. `uint16_t *pos_conv_buf = (uint16_t*) ((char*)buf + len);` can be an odd address. It might a problem for some architectures to read `uint16_t` from an unaligned address.

It's true for some architectures, I'm actually not sure about ESP32.  
I agree it should be fixed, to be on the safe side.  
However, it's not related to the problem above since `lv_bidi_process_paragraph` was not called from `lv_bidi_get_logical_pos` in that case. Even if it did, it would cause `UNALIGNED ERROR` error and not `LOADSTORE ERROR` (xtensa architecture has different codes for these erros).

> 2\. `len` should be `len+1` becasue [lv_bidi_process_paragraph](https://github.com/littlevgl/lvgl/blob/master/src/lv_misc/lv_bidi.c#L195) set the trailing zero immediatelly.

I think you are right, but I don't see how this could cause the crash above. When `str_out[len]` is written to, it mistakenly writes to the first element of `pos_conv_buf` which comes right after it, but it should not cause a `LoadStoreError` by itself.

> I've fixed these [like this](https://github.com/littlevgl/lvgl/blob/feat-new-style/src/lv_misc/lv_bidi.c#L191-L192).

With multiple temporary buffers, it makes sense allocating two buffers instead of one.
But multiple temporary buffers are not available yet on v.6 so we would need a different fix for v.6.

As I mentioned, I'm not sure that this fix would also fix the problem above, at least I haven't seen how the problems you pointed at could cause a `LoadStoreError` when executing `str_out[len] = '\0'`, especially when it was called from `lv_label_get_letter_on` where `pos_conv_buf` is not used.

Do you think this is related?
Do you have any idea what else could cause this problem?

kisvegabor, 2020-01-14:
What I can imagine is the `\0` is overwritten by `pos_conv_buf` and therefore the string looks longer. I'm not sure what are the real consequences (how could it cause LoadStoreError) but it looks evil. 

Anyway, IMO it's worth a try to see if it fixes the crash.

kisvegabor, 2020-01-20:
I've added fix here: a53fca8c

Now it seems stable to me.

@amirgon Could you update the binding?

amirgon, 2020-01-20:
> Could you update the binding?

@kisvegabor Updated. 
In the future we can consider automating this (Azure Pipelines?)

kisvegabor, 2020-01-21:
Thanks!

I've just seen that there is a release pipeline too where (I suppose) we can run any scripts to automate release to be sure we don't miss anything. 
Even hardware and simulator projects could be updated on minor versions automatically.

embeddedt, 2020-02-07:
It seems that the original problem has been fixed so I'll close this issue.

commit a53fca8c (2020-01-20): fix buffer out of bound error in lv_bidi_get_visual/logicl_pos(). Fixes #1351
closed 2020-02-07 by embeddedt