lvgl/lvgl #1352: Web server is down?

amirgon, 2020-01-13:
https://forum.littlevgl.com/ and https://littlevgl.com/ are unreachable now....

closed 2020-01-13 by amirgon

kisvegabor, 2020-01-13:
Probably just a temporal issue.

jTecRepos, 2024-08-21:
nope - website is completely dead

kisvegabor, 2024-08-21:
Do you mean lvgl.io? t's working here. Can you try from mobile too?

jTecRepos, 2024-08-22:
no i mean littlevgl.com (see original comment) 
i think littlevgl.com was bought out and replaced with something else to collect site visits.
its a little bit offtopic here but a lot of older posts recommend littlevgl to generate fonts, so I wanted to leave that information here. 

for fonts generation use: [https://lvgl.io/tools/fontconverter](https://lvgl.io/tools/fontconverter)

kisvegabor, 2024-08-23:
Ah, when we have change the LittleVGL name to LVGL I kept the littlevgl.com domain for a few years, but I wasn't playing for it anymore. It seems it was really taken by someone.