lvgl/lvgl #1353: Button matrix crash: Problem with the stability of  "lv_btnm_get_active_btn_text" on esp32.

justbarran, 2020-01-13:
Good Day, 

When I used "lv_btnm_get_active_btn_text" in the code as in the example https://docs.littlevgl.com/en/html/object-types/btnm.html
The esp32 would crash if you drag your finger across multiple buttons. What happens is the output string starts to concatenate and go outside an array limits if specified. To fix this I had to use:

static void event_handler(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_VALUE_CHANGED) {
        uint16_t keypress = lv_btnm_get_active_btn(obj);
        if(keypress < 15){ //Remove any outide values due to screen drag causing value to jump to65K
        printf("%s was pressed\n", lv_btnm_get_btn_text(obj,keypress););
        }
    }
}

kisvegabor, 2020-01-13:
Hi,

Take look at the implementation of [lv_btnm_get_active_btn_text](https://github.com/littlevgl/lvgl/blob/master/src/lv_objx/lv_btnm.c#L518-L528). It returns `NULL` if there is no active button. It seems to me that a more appropriate fix would be to check if the return vaéues of `lv_btnm_get_active_btn_text` is `NULL` or not.

kisvegabor, 2020-02-04:
I close this issue in lack of activity. 

Feel free to comment here if the suggested method doesn't work.

closed 2020-02-04 by kisvegabor