lvgl/lvgl #1355: Tslib Support

j-clay, 2020-01-15:
Hello!

I have a device using xf86-input-tslib as the touchscreen input driver. Is this currently supported or some temporary work around to use it?

I've only been getting x:0,y:0 from the indev x,y input anytime I touch the screen. I've tried enabling evdev but not libinput config yet (i currently have an issue compiling libinput which i posted about in the lv_drivers repo). 

Thanks

embeddedt, 2020-01-15:
We only support full-on X11 (through the SDL drivers) or direct evdev access (through the evdev driver). Which driver are you using?

If you're using `evdev`, have you tried running `evtest <your input device>` on the command line to verify that the right events are being sent by Linux?

embeddedt, 2020-01-15:
Also, please note that the forum is really the proper place to post questions like these. It's fine for this time, but please post there in the future. :slightly_smiling_face:

j-clay, 2020-01-15:
Sorry about posting here! Will do that next time :)

I ran evtest and it is /dev/input/event1 for the ADS7846 Touchscreen I'm using.  However, I've tried to use evdev configuration in lv_conf.h and no such luck. 

The target system is using xf86-input-tslib x.org input driver

embeddedt, 2020-01-15:
Did you try adding a printf statement above [this line](https://github.com/littlevgl/lv_drivers/blob/1f13e4cc31f20a9afcf4309ad00927d0f2eec5e0/indev/evdev.c#L103) to see if it picks up any events from the device?

j-clay, 2020-01-15:
Just tried that and it doesn't hit that line when i touch the touchscreen. 

I also have this output in file lvgl/src/lv_core/lv_indev.c and I receive x:0 y:0 all the time, even when pressing the touch screen.

```
void lv_indev_read_task(lv_task_t * task)
{
    LV_LOG_TRACE("indev read task started");

    lv_indev_data_t data;

    indev_act = task->user_data;

    /*Read and process all indevs*/
    if(indev_act->driver.disp == NULL) return; /*Not assigned to any displays*/

    /*Handle reset query before processing the point*/
    indev_proc_reset_query_handler(indev_act);

    if(indev_act->proc.disabled) return;
    bool more_to_read;
    do {
        /*Read the data*/
        more_to_read = lv_indev_read(indev_act, &data);
        printf("reading in data x %d y %d\n", data.point.x, data.point.y);
        /*The active object might deleted even in the read function*/
        indev_proc_reset_query_handler(indev_act);
        indev_obj_act = NULL;
```

embeddedt, 2020-01-15:
That means it's failing to read from `/dev/input/event1`. Did you call `evdev_init` and register the input device?

j-clay, 2020-01-15:
Here's my code:

`
#include "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"
#include "lv_drivers/indev/evdev.h"
#include "lv_examples/lv_examples.h"
#include <unistd.h>
#include <stdio.h>

#define DISP_BUF_SIZE (80*LV_HOR_RES_MAX)

/**********************
 *  STATIC PROTOTYPES
 **********************/
/**
 * Called when a button is released
 * @param btn pointer to the released button
 * @param event the triggering event
 * @return LV_RES_OK because the object is not deleted in this function
 */
static void btn_event_cb(lv_obj_t * btn, lv_event_t event)
{
//    if(event == LV_EVENT_RELEASED) {
    	printf("Button clicked\n");
	/*Increase the button width*/
        lv_coord_t width = lv_obj_get_width(btn);
        lv_obj_set_width(btn, width + 20);

  //  }
}

bool my_touchpad_read(lv_indev_drv_t * indev_driver, lv_indev_data_t * data)
{
    static lv_coord_t last_x = 0;
    static lv_coord_t last_y = 0;   
#if 0
    last_x = last_x + 1;
    
    if(last_x == 640) {
    	last_y = last_y + 1;
        last_x = 0;
    }
   
    if(last_y == 480)
	last_y = 0;

    data->point.x=last_x;
    data->point.y=last_y;
#endif 
    return false; /*Return `false` because we are not buffering and no more data to read*/
}

int main(void)
{
    /*LittlevGL init*/
    lv_init();

    /*Linux frame buffer device init*/
    fbdev_init();

    evdev_init(); 

    /*A small buffer for LittlevGL to draw the screen's content*/
    static lv_color_t buf[DISP_BUF_SIZE];

    /*Initialize a descriptor for the buffer*/
    static lv_disp_buf_t disp_buf;
    lv_disp_buf_init(&disp_buf, buf, NULL, DISP_BUF_SIZE);

    /*Initialize and register a display driver*/
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.buffer = &disp_buf;
    disp_drv.flush_cb = fbdev_flush;
    lv_disp_drv_register(&disp_drv); 

    lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);             /*Descriptor of a input device driver*/
    indev_drv.type = LV_INDEV_TYPE_POINTER;    /*Touch pad is a pointer-like device*/
    indev_drv.read_cb = my_touchpad_read;      /*Set your driver function*/
    lv_indev_drv_register(&indev_drv);         /*Finally register the driver*/

    //lv_tutorial_objects();

    /*Create a "Hello world!" label*/
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    
    /* CREATE A SCREEN
     *******************/
    /* Create a new screen and load it
     * Screen can be created from any type object type
     * Now a Page is used which is an objects with scrollable content*/
    lv_obj_t * scr = lv_page_create(NULL, NULL);
    lv_disp_load_scr(scr);
    
    /****************
     * ADD A TITLE
     ****************/
    lv_obj_t * title = lv_label_create(scr, NULL); /*First parameters (scr) is the parent*/
    lv_label_set_text(title, "Object usage demo");  /*Set the text*/
    lv_obj_set_x(title, 50);                        /*Set the x coordinate*/

    /***********************
     * CREATE TWO BUTTONS
     ***********************/
    /*Create a button*/
    lv_obj_t * btn1 = lv_btn_create(lv_disp_get_scr_act(NULL), NULL);         /*Create a button on the currently loaded screen*/
    lv_obj_set_event_cb(btn1, btn_event_cb);                                  /*Set function to be called when the button is released*/
    label = lv_label_create(btn1, NULL);
    lv_label_set_text(label, "Button 1");
    lv_obj_align(btn1, label, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 20); 

    /*Copy the previous button*/
    lv_obj_t * btn2 = lv_btn_create(scr, btn1);                 /*Second parameter is an object to copy*/
    lv_obj_align(btn2, btn1, LV_ALIGN_OUT_RIGHT_MID, 50, 0);    /*Align next to the prev. button.*/

    /*Create a label on the button*/
    label = lv_label_create(btn2, NULL);
    lv_label_set_text(label, "Button 2");

    /*Handle LitlevGL tasks (tickless mode)*/
    while(1) {
        lv_tick_inc(5);
        lv_task_handler();
        usleep(5000);
    }

    return 0;
}

`

embeddedt, 2020-01-15:
I think `my_touchpad_read` should be replaced with `evdev_read`.

j-clay, 2020-01-15:
That did the trick!

I did not find that on any tutorial or guide that is posted for this library after lots of searching. Maybe this is relatively unique to my setup, but this might be helpful to mention this somewhere in the guide.

I appreciate the quick response and help @embeddedt . Thank you very much!

embeddedt, 2020-01-15:
The idea is that for most applications you would have to create your own touchscreen driver. In your case, you can use one of the premade drivers from the lv_drivers repository, but you have to make sure that the premade function is actually the one being used.

Can this issue be closed now?

j-clay, 2020-01-15:
Gotcha.

closed 2020-01-15 by j-clay