lvgl/lvgl #1358: [BUG] Calendar header button signal wrong behavior.

turoksama, 2020-01-18:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

A clear and concise description of what the bug is.
Hello there,
As mention above.
The action of pressing the header of Calendar will signal an user event callback.

**To Reproduce**

Please provide a small, independent code sample that can be used to reproduce the issue. Ideally this should work in the PC simulator unless the problem is specific to one platform.
```
static const char *day_names_char[]={"Su","Mo","Tu","We","Th","Fr","Sa",	""};
static const char *month_names_char[]=
{"Jan","Feb","Mar","Apr","May","Jun","Jul","Aug","Sep","Oct","Nov","Dec",""};

static void mal_calendar_cb(lv_obj_t * obj, lv_event_t event)
{
		switch (event)
		{
			case LV_EVENT_PRESSED:
			{
				//Select
				lv_calendar_date_t *date_sel = lv_calendar_get_pressed_date(calendar);
				lv_calendar_set_today_date(calendar, date_sel);
			}break;
			case LV_EVENT_SHORT_CLICKED:
			{
			}break;
			case LV_EVENT_CLICKED:break;
			case LV_EVENT_LONG_PRESSED:break;
			case LV_EVENT_RELEASED:break;
			default:break;
		}
}

lv_obj_t *calendar_create(lv_obj_t * par)
{
	lv_obj_t *ca = calendar = lv_calendar_create(par, NULL);

	lv_calendar_set_day_names(ca, day_names_char);//
	lv_calendar_set_month_names(ca, month_names_char);//

	lv_obj_set_size(ca,350,380);
	lv_obj_align(ca, par, LV_ALIGN_CENTER, 0, 0);
	lv_obj_set_top(ca, true);

	lv_obj_set_event_cb(ca, mal_calendar_cb);

        return ca;
}
```

**Expected behavior**

A clear and concise description of what you expected to happen.

I think this kind of action is internal behavior of the object. 
Generate an EVENT may crash. In my case, it did.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Add any other context about the problem here.

A hack to solve this problem:

```
static void mal_calendar_cb(lv_obj_t * obj, lv_event_t event)
{
//detect point is not on header area
    lv_calendar_ext_t * ext = lv_obj_get_ext_attr(calendar);
    lv_area_t header_area;
    lv_area_copy(&header_area, &calendar->coords);

    lv_coord_t h = lv_font_get_line_height(ext->style_header->text.font) + ext->style_header->body.padding.top +
           ext->style_header->body.padding.bottom;

    header_area.y2 = header_area.y1 + h;

    lv_indev_t * indev = lv_indev_get_act();
    lv_point_t p;
    lv_indev_get_point(indev, &p);

    /*If the header is pressed mark an arrow as pressed*/
    if(!lv_area_is_point_on(&header_area, &p))
    {
                //no, point is not on header area, let's do our job.
		switch (event)
		{
			case LV_EVENT_PRESSED:
			{
				//Select
				lv_calendar_date_t *date_sel = lv_calendar_get_pressed_date(calendar);
				lv_calendar_set_today_date(calendar, date_sel);
			}break;
			case LV_EVENT_SHORT_CLICKED:
			{
			}break;
			case LV_EVENT_CLICKED:break;
			case LV_EVENT_LONG_PRESSED:break;
			case LV_EVENT_RELEASED:break;
			default:break;
		}
    }
}
```

embeddedt, 2020-01-18:
The basic input events are always triggered on the whole surface of an object (regardless of its type) so I don't think that is specifically the problem. They come directly from the input system, not from the object's signal handler.

> lv_calendar_get_pressed_date(calendar); will result NULL if not a date select condition

That makes sense to me. If you didn't touch a date, what would you expect it to return?

For your case, can't you just check whether the return value was NULL or not? That seems like the cleanest solution to me, and doesn't seem very odd. Or have I missed something?

turoksama, 2020-01-18:
Well, it's a little bit different from version 5.3,fair enough,I figured it out. I realized that it is clean and logical.Thanks for your support.Happy days are back again. @embeddedt

closed 2020-01-18 by turoksama

embeddedt, 2020-01-18:
No problem! :slightly_smiling_face: