lvgl/lvgl #1360: Invisible textareas cause driver redraws

fpw, 2020-01-18:
**Describe the bug**

I've upgrade my project to lvgl 6.1.1 and observed that #230 is happening again: A text area in an invisible container causes periodic calls to the driver's flush_cb due to the blinking cursor.

**To Reproduce**

See in the discussion of #230

**Expected behavior**

Text areas in invisible containers don't cause screen updates.

**Screenshots**
Not applicable.

**Additional context**
I believe the problem is that `cursor_blink_anim` calls `lv_inv_area` nowadays. It used to call `lv_obj_invalidate` in older versions and that function checked if the object is visible before invalidating the area.

embeddedt, 2020-01-18:
I recall this being done to speed up the drawing of the text area as a whole, as drawing the whole object when only the cursor area had changed didn't make sense.

Perhaps we should add a new function similar to `lv_inv_area` that also checks whether the area is actually visible?

fpw, 2020-01-18:
Thanks for the explanation, `lv_obj_invalidate` would indeed redraw the whole textarea. But visibility for an area cannot be determined since the area itself is covered with another object that is visible.

An alternate solution would be to simply check the visibility of the text area inside `cursor_blink_anim`, i.e. adding the visibility to the existing if statement. This would have the same effect as `lv_obj_invalidate` but still only invalidate the area of the cursor.

embeddedt, 2020-01-19:
> check the visibility of the text area inside `cursor_blink_anim`, i.e. adding the visibility to the existing if statement.

Exactly; I just thought abstracting it into a function that could be shared with another object made more sense. Otherwise, the check may have to be duplicated for another, similar object.

kisvegabor, 2020-01-20:
(Sorry I closed it accidentally)

I've added `lv_obj_invalidate_area()` here: 9b283981

fpw, 2020-01-21:
Can confirm the commit fixes the redrawing problem, thanks!

closed 2020-01-21 by fpw

kisvegabor, 2020-01-23:
Thanks for the feedback!

commit 9b283981 (2020-01-20): add lv_obj_invalidate_area to replace lv_inv_area in objects. Fixes #1360