lvgl/lvgl #1364: LV_LABEL_LONG_DOT: write `'\0'` out of the bounds of the text buffer.

apapkovskiy, 2020-01-29:
Hi. My question is regarding the following line of code:
[ext->text[byte_id_ori + i] = '.';](https://github.com/littlevgl/lvgl/blob/10b9c9b2f5344e7b2f5cc00a19ed86ed56ae9866/src/lv_objx/lv_label.c#L1318)

```
            if(lv_label_set_dot_tmp(label, &ext->text[byte_id_ori], len)) {
                for(i = 0; i < LV_LABEL_DOT_NUM; i++) {
                    ext->text[byte_id_ori + i] = '.';
                }
                ext->text[byte_id_ori + LV_LABEL_DOT_NUM] = '\0';
                ext->dot_end                              = letter_id + LV_LABEL_DOT_NUM;
            }
```
How we can be sure that there is space for this dots. For example, take a string "iiiiWWW". I think it is possible the space occupied by the last two W is enough for 3 dots. In this particular case  `byte_id_ori` points to the second W and it is 5 and we write `'\0'` out of the bounds of the text buffer. What do you think?

kisvegabor, 2020-01-30:
Hi,

Great catch!
I've just fixed in `master`.

kisvegabor, 2020-02-20:
I close this issue in lack of activity.

Feel free to comment here if you see further related issues.

closed 2020-02-20 by kisvegabor