lvgl/lvgl #1366: lv_obj_move_foreground takes long time

zhoufs, 2020-01-29:
I created two screens( each screen includes one parent and multiple children) and user function "lv_obj_move_foreground" to switch two screens. In embedded system(STM32L4R5I), it takes at least half seconds  between function lv_obj_move_foreground and LCD flush. Any clue about this issue?

embeddedt, 2020-01-29:
If you use `lv_scr_load` instead (which is the usual way of doing it) do you see the same issue?

zhoufs, 2020-01-29:
I use the function to change part of screen,I use the top/bottom part for status bar. so lv_scr_load might not be the best one. I also noticed that long delay comes from function lv_refr_areas called by lv_disp_ref_task  in lv_refr.c

embeddedt, 2020-01-29:
That means that it's taking too long to refresh the display. The problem is not with `lv_obj_move_foreground`.

zhoufs, 2020-01-29:
I agree, what can I try to reduce the delay?

embeddedt, 2020-01-29:
Try increasing your display buffer size (if you can). That will reduce the number of calls that need to be made to your `flush_cb` function.

zhoufs, 2020-01-30:
My display is a Sharp memory display, 320x240 monochrome, SPI interface. The display buffer size is set as frame size(320x240/8). I tried two screen buffer size, and it does not help. As I said, the delay happens from function  lv_refr_areas, before flush_cb is called.

kisvegabor, 2020-01-30:
@zhoufs 
I've some questions to get more info about your system:
- What type of MCU do you use? (especially how fast it is?)
- Do you use `set_px_cb`?

zhoufs, 2020-01-30:
The CPU is STM32L4RIZ, system clock was set at 4Mhz to reduce power. I increased it to 48Mhz, it helps a lot . I believe set_px_cb is not used, does it help? I try to find balance between performance and power consumption.

embeddedt, 2020-01-30:
> The CPU is STM32L4RIZ, system clock was set at 4Mhz to reduce power. I increased it to 48Mhz, it helps a lot.

I'm impressed that it ran well; I guess we write reasonably efficient code. :wink: 

Perhaps dynamically adjusting  the processor speed is an option. That way, the processor only runs at high speed when absolutely necessary. You mentioned you were using STM32; I assume you use the `WFI` instruction to enter sleep mode whenever `lv_task_handler` isn't running?

> I believe set_px_cb is not used, does it help?

No; it almost always causes a drop in performance. It's mainly meant for small systems with monochrome displays where memory consumption is an issue and you want to pack pixels into a buffer as tightly as possible.

zhoufs, 2020-01-30:
Yes, that is my next step(it seems I have to), I will implement dynamic power management to optimize overall current consumption. 
The GUI is superb, I like it very much! Thanks again for answering my question.

closed 2020-01-30 by zhoufs