lvgl/lvgl #1368: lv_label_set_text_fmt on stm32

imprego, 2020-02-03:
I want set text area's text with lv_label_set_text_fmt(lv_ta_get_label(smth), ...) and into function lv_label_set_text_fmt "uint32_t len = lv_vsnprintf(NULL, 0, fmt, ap);" always set len == 0 and text area stay empty (on stm32), but in codeblocks_sim text set correctly.
All of this is equivalent for "label" object.
keil - mVision v5.28.0.0

imprego, 2020-02-03:
The ussue is seen with microlib.

embeddedt, 2020-02-03:
What do you have `LV_SPRINTF_CUSTOM` set to in `lv_conf.h`?

imprego, 2020-02-04:
I used:
#define LV_SPRINTF_CUSTOM   1
#if LV_SPRINTF_CUSTOM
#  define LV_SPRINTF_INCLUDE <stdio.h>
#  define lv_snprintf     snprintf
#  define lv_vsnprintf    vsnprintf
#endif  /*LV_SPRINTF_CUSTOM*/

Then I've redefined snprintf and vsnprintf
#  define lv_snprintf     my_snprintf
#  define lv_vsnprintf    my_vsnprintf

int my_vsnprintf(char *s, size_t n, const char *fmt, va_list args)
{
	char tmpChr = '\0';
	if(s == NULL || n == 0)
		return vsnprintf(&tmpChr, 1, fmt, args);
	return vsnprintf(s, n, fmt, args); 
}

Problem with vsnprintf(NULL, 0, fmt, args) === 0 seen only with microlib.

kisvegabor, 2020-02-04:
> Problem with vsnprintf(NULL, 0, fmt, args) === 0 seen only with microlib.

So is it working with  `LV_SPRINTF_CUSTOM 0`?

imprego, 2020-02-04:
With LV_SPRINTF_CUSTOM 0 I get .\Obj\IP80.axf: Error: L6218E: Undefined symbol lv_vsnprintf (referred from lv_label.o).
I set redefined functions as written above.

embeddedt, 2020-02-04:
> .\Obj\IP80.axf: Error: L6218E: Undefined symbol lv_vsnprintf

You probably aren't including `lv_misc/lv_printf.c` in your list of files to compile.

I think your C library's `printf` is not standards-compliant enough to work properly with LittlevGL, so I recommend sticking with the built-in one.

imprego, 2020-02-04:
I've added `lv_misc/lv_printf.c` in my project on stm32 and set `LV_SPRINTF_CUSTOM 0`, and `lv_label_set_text_fmt` work normal. I have no idea, why this file wasn't added earlier. Thanks for answers.

closed 2020-02-04 by imprego