lvgl/lvgl #1369: lv_ta_set_one_line reset text align?

imprego, 2020-02-04:
I've create text area, and want add button on left side of this and set text align to right.
If I write 
```
lv_ta_set_one_line(ta, true);
lv_ta_set_text_align(ta, LV_LABEL_ALIGN_RIGHT);
```
all ok, but if
```
lv_ta_set_text_align(ta, LV_LABEL_ALIGN_RIGHT);
lv_ta_set_one_line(ta, true);
```
text align in text area aligned left.

embeddedt, 2020-02-04:
I think this should fix it: https://github.com/littlevgl/lvgl/pull/1370

Can you try it and confirm that it works for you?

imprego, 2020-02-05:
I can't build project now:
C:\Users\dudenkov\Desktop\all\pc_simulator_win_codeblocks\lvgl\src\lv_objx\lv_ta.c|736|undefined reference to `lv_ta_get_text_align'|

embeddedt, 2020-02-05:
Sorry about that. Try it again; it should work this time.

imprego, 2020-02-06:
Yes, now text alignment set to `LV_LABEL_ALIGN_RIGHT` independently from lines position.

kisvegabor, 2020-02-07:
The PR is merged yo I close this issue.
@embeddedt  thank you for the fix!

closed 2020-02-07 by kisvegabor