lvgl/lvgl #1372: Null pointer de-referencing

sicrisembay, 2020-02-10:
Target: iMXRT1052
LVGL version: 6.1.1

I encountered a crash when deleting and creating screens.  Long story short, I've tracked the root cause and it was due to null pointer de-reference.  This is exactly the same issue as issue #1095.  My fix is shown in the attached patch.

Attached is the patch on lv_obj.c
[lv_obj.patch.txt](https://github.com/littlevgl/lvgl/files/4179871/lv_obj.patch.txt)

embeddedt, 2020-02-10:
As far as I can tell, this can only happen if you call `lv_obj_set_pos` on a screen, which is not supported (same conclusion in #1095). Can you explain the difference you see?

sicrisembay, 2020-02-11:
It is the same with #1095 and I also agree with the conclusion.

The difference may be on triggering this issue.  My issue is triggered when I create a new screen (NULL parent) and copy (properties) from another screen. 

`me->objScreen = lv_obj_create(NULL, me->pBaseScreen);`

This may be an invalid use-case of lv_obj_create ( I may missed reading some API documentation).

I have 1 request:
Request#1: In lv_obj_set_pos(), can we add LV_ASSERT_OBJ() statement to check that obj->par is not NULL.
Rationale: IMHO, it helps on catching misuse of API via LV_DEBUG_ASSERT.

In my case, side-effect was a hardfault error and it took me some time to figure this out.  I wish a simple LV_DEBUG_ASSERT to catch this was available.

embeddedt, 2020-02-11:
>  I create a new screen (NULL parent) and copy (properties) from another screen.

That's supposed to work, as far as I can tell. Notice [here](https://github.com/littlevgl/lvgl/blob/2cd3a5ac09f53a8eacdfc7f522dc20d874e299ac/src/lv_core/lv_obj.c#L388-L393) that it checks for a parent object.

Do you know where it crashes inside `lv_obj_create`?

> In lv_obj_set_pos(), can we add LV_ASSERT_OBJ() statement to check that obj->par is not NULL.

Sure; I've [added a warning in `dev-7.0`](https://github.com/littlevgl/lvgl/commit/b9697df3a1fac8feffcb15b86f6aecd6d5a04e55).

kisvegabor, 2020-02-11:
> Sure; I've added a warning in dev-7.0.

Thank you!

sicrisembay, 2020-02-12:
> Do you know where it crashes inside `lv_obj_create`?

line392 of code snippet below.
https://github.com/littlevgl/lvgl/blob/2cd3a5ac09f53a8eacdfc7f522dc20d874e299ac/src/lv_core/lv_obj.c#L388-L393

In my case, new_obj->par is NULL and in lv_set_pos() line 700, par is assigned NULL and de-referenced in line 702 and 703.  Hardfault side-effect is random and depends on the invalid dereferenced values (par->coords.x1 and par->coords.y1).  If x1 and y1 is zero, there'll be no hardfault as it simply returns on line713.  At some point, this is not zero and hardfault will be thrown when executing function pointer par->signal_cb in line733.

https://github.com/littlevgl/lvgl/blob/2cd3a5ac09f53a8eacdfc7f522dc20d874e299ac/src/lv_core/lv_obj.c#L700-L703

> Sure; I've [added a warning in `dev-7.0`](https://github.com/littlevgl/lvgl/commit/b9697df3a1fac8feffcb15b86f6aecd6d5a04e55).

Thank you very much!  This should be enough to catch this behavior.

closed 2020-02-12 by sicrisembay