lvgl/lvgl #1373: Unable to resolve out of memory error

GnUfTw, 2020-02-10:
**Description**

I am working on an LVGL application that has 8 tabs and displays a number of fields and controls in each of those tabs. I am trying to add a label to one of the tabs but after doing so, I get the following error when launching the application in the simulator:
```
Warn: Couldn't allocate memory  (lvgl/lv_misc/lv_mem.c #145)
Error: Out of memory!   (lvgl/lv_core/lv_obj.c #187)
```
This has happened to me before and I usually just increase the size of `LV_MEM_SIZE` to eliminate the issue. `LV_MEM_SIZE` is currently defined as `(1024U * 1024U)`. If I increase the size of `LV_MEM_SIZE` definition to `(2048U * 1024U)`, there is no difference in behavior and the application still throws the error shown above upon launch.

If I remove the code that adds the label and pushes the application above the memory limit, the application runs fine and the memory monitor reports the following:
```
used:  55000 (  6 %), frag:   1 %, biggest free: 993568
```

If I am understanding the memory monitor output correctly, I am no where near taking up the 1MB of memory being allocated to LVGL. How is adding a single label pushing it over the edge?

**To Reproduce**

The best way to reproduce this issue would be to attach the application source to this ticket but I will not be able to do this as it is closed source. 

**Expected behavior**

- I expected the out of memory issue to go away after increasing LV_MEM_SIZE
- I expected the memory monitor to report a much higher _used_ value after removing the label that pushes LVGL over the edge

**Screenshots**

NA

**Additional context**
- LVGL version: 5.2.0
- OS: Ubuntu 18.04 x64

embeddedt, 2020-02-10:
LittlevGL 5.2 is very old, and isn't really supported anymore. Can you update to 5.3, at the very least? Even 5.3 is outdated now but I assume this is for an existing product and you can't afford a major upgrade at the moment.

The behavior you're describing is very weird. Creating a label shouldn't use anywhere near 1MB of memory. I'm guessing that there's memory corruption somewhere. Are you using a custom memory allocator or the built-in one? What does the memory monitor report after the allocation fails? (You might have to insert the memory monitor code in `lv_obj.c`` for debugging purposes.)

embeddedt, 2020-02-10:
Some other things to try:

* Does this happen if you create a different object type (like a button)?
* You mentioned that you're using the simulator. Does this issue also happen on the target device (if one exists for testing)?

GnUfTw, 2020-02-10:
- Updating to 5.3 should be possible as the app has not been released yet, I will report back after I attempt to upgrade.
- Using the built in memory allocator, below is relevant section of lv_conf.h:
```
#define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#define LV_MEM_SIZE    (1024U * 1024U)      /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#define LV_MEM_ADR          0               /*Set an address for memory pool instead of allocation it as an array. Can be in external SRAM too.*/
#define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/
```
- The memory monitor does not report anything when the application fails. The simulator window that shows up but it is blank. The only output I see is the output reported in my first post. I believe this is because the entire UI is constructed immediately after LVGL is initialized and before the memory monitor is given a chance to execute.
- Adding a button instead of a label does not cause the issue to occur, application runs fine
- I do not have access to target device at the moment but give it a shot once I do

embeddedt, 2020-02-10:
Can you try moving the memory monitor statements into [this `if` statement](https://github.com/littlevgl/lvgl/blob/2cd3a5ac09f53a8eacdfc7f522dc20d874e299ac/src/lv_misc/lv_mem.c#L182)? (The line numbers probably won't match because that's the 6.1 file.)

> Adding a button instead of a label does not cause the issue to occur, application runs fine

Very strange. Can you try adding a label as a child of the button?

embeddedt, 2020-02-10:
By the way, I believe commit https://github.com/littlevgl/lvgl/commit/683545f5f083f8fcc5b580b2a47853cf6a322686 is the last commit before 6.0 got merged into `master`. (We weren't tagging patch releases at that point in time, so this commit should have every fix since 5.3's release.)

GnUfTw, 2020-02-10:
- I have moved memory monitor statements into the if statement requested, with a slight hitch. Attempting to pass `mon.frag_pct` to `printf` results in `Floating point exception (core dumped)`. Removing this arg and its corresponding formatting to `printf` results in no exception. With that being said, here is the updated output:
```
used: 1048576 (100 %), biggest free:      0
Warn: Couldn't allocate memory  (lvgl/lv_misc/lv_mem.c #157)
Error: Out of memory!   (lvgl/lv_core/lv_obj.c #187)
```
- I actually added a label as a child of the button when I tested adding a button instead of a label, I should have mentioned this, but it worked without issue
- Thanks for the 683545f commit reference, I will change to it and see if the issue gets fixed

GnUfTw, 2020-02-10:
I am experiencing difficulty updating lvgl to commit 683545f. Between 5.2.0 and 683545f the source file **lv_core/lv_vdb.h** got removed. The following files from **lv_drivers** reference this header:
- fbdev.c
- monitor.c
- R61581.c
- SSD1963.c
- ST7565.c

This results in compilation failure. Not sure what the right fix is for this. I tried updating lv_drivers source to commit 060bb6f (last commit before 6.0 tag) to fix the issue but the references to **lv_vdb.h** are still there. Any suggestions?

embeddedt, 2020-02-10:
Replacing them with `lvgl/lvgl.h` should work, although it seems that you just fixed that problem. :slightly_smiling_face:

GnUfTw, 2020-02-10:
I am going to have to hold off on updating lvgl to commit 683545f. There are some breaking changes related to container styling and fitting and I presume these will significantly effect the layout. Before I move further with the update I need to investigate alternative solutions to the issue.

GnUfTw, 2020-02-10:
I found the issue and it was an issue in application code. Issue was due to calling `lv_cont_create` where I should have been using `lv_label_create`.

closed 2020-02-10 by GnUfTw

embeddedt, 2020-02-10:
Glad to hear it!

Just so you know, LittlevGL 6.1 includes additional built-in assertions to lessen the likelihood of that bug going unnoticed.