lvgl/lvgl #1374: Solve the memory leak of user data

ShenRen, 2020-02-10:
Please forgive me, my network can't connect to lvgl's forum, so I'm tired of putting questions here.

```c++
/*Declare the type of the user data of object (can be e.g. `void *`, `int`, `struct`)*/
typedef void * lv_obj_user_data_t;

typedef struct _lv_obj_t
{
    //Other members
#if LV_USE_USER_DATA
    lv_obj_user_data_t user_data; /**< Custom user data for object. */
#endif

} lv_obj_t;
```
Object has members that manage user data. It is very useful for function expansion.
Although we can determine the type of data, we do not have a proper way to release the user data when the object is deleted. This requires us to spend extra resources to record and manage these user data to avoid memory leakage. Especially when objects nesting level is very deep, the problem is more troublesome.
I think we need to provide an interface function to improve this problem.

```c++

//In lv_conf.h file //////////////////////////////////////////////////

/*1: Add a `user_data` to drivers and objects*/
#define LV_USE_USER_DATA        1

#if LV_USE_USER_DATA
/*Declare the type of the user data of object (can be e.g. `void *`, `int`, `struct`)*/
typedef void * lv_obj_user_data_t;
#define LV_USER_DATA_FREE_INCLUDE  "something.h"       /*Header for user data free  function*/
#define LV_USER_DATA_FREE  (user_data_free)     /*Invoking for user data free  function */
#endif

//in something.c file ////////////////////////////////////////////////
void user_data_free(lv_obj_user_data_t & user_data)
{
// free data
}

//in lv_obj.c file //////////////////////////////////////////////////////

#if LV_USE_USER_DATA
#include LV_USER_DATA_FREE_INCLUDE  
#endif 

/**
 * Delete 'obj' and all of its children
 * @param obj pointer to an object to delete
 * @return LV_RES_INV because the object is deleted
 */
lv_res_t lv_obj_del(lv_obj_t * obj)
{
//Other code
#if LV_USE_USER_DATA
    LV_USER_DATA_FREE(obj->user_data);
#endif 
}
```
If there is no problem in this way, I can create a pull request.

embeddedt, 2020-02-10:
I think this can already be done by using an event callback and freeing user data on `LV_EVENT_DELETE`.

ShenRen, 2020-02-11:
> I think this can already be done by using an event callback and freeing user data on `LV_EVENT_DELETE`.

Yes, event processing can release user data. But you need every object that uses user data change event handler one by one. At the same time, you have to record the object's original event handler.

ShenRen, 2020-02-11:
I always wanted to  wrap lvgl as a C + + Library. In this process, I need to extend some members and methods by using user data. But memory release of user data is always a problem. Using event handling will destroy the original design and increase memory costs. Not so elegant and direct.
I pursue as few code changes as possible.Changing lvgl's original code is the last thing I want to do.
But if lvgl can provide this kind of function interface, my work will be very smooth.

embeddedt, 2020-02-11:
@kisvegabor What do you think about this?

kisvegabor, 2020-02-11:
I see the issue and it's really an ugly situation. Creating event functions for a lot of objects only for `LV_EVENT_DELTE` is really tedious and unpleasant. 

I can't imagine a better solution and your suggestion seems simple and elegant.

Only a minor thing: let's pass `obj` to `LV_USER_DATA_FREE` instead of `obj->user_data` because it's easy to get `user_data` in the function too but `obj` might mean a useful context.

So if @embeddedt also agrees, please send a PR.

@amirgon could this custom "user data free" function be useful for Micropython?

amirgon, 2020-02-11:
> @amirgon could this custom "user data free" function be useful for Micropython?

The suggestion sounds good and useful for C++ and perhaps other bindings.  
In Micropython, however, memory is deallocated automatically using the garbage collector, so there is no need to explicitly deallocated memory after an object is deleted.  

One thing I can think of though, is using this callback to mark the Micropython wrapping object as invalid.  
Today, accessing an object after it was deleted causes a crash because the wrapping Micropython object points to the original lvgl object and is not aware that it was already freed.

embeddedt, 2020-02-11:
I agree with @kisvegabor's recommendation.

@ShenRen Please send a PR. It should ideally go to `dev-7.0`.

ShenRen, 2020-02-12:
> Only a minor thing: let's pass `obj` to `LV_USER_DATA_FREE` instead of `obj->user_data` because it's easy to get `user_data` in the function too but `obj` might mean a useful context.

> @ShenRen Please send a PR. It should ideally go to `dev-7.0`.

```c++

// In lv_conf.h ////////////////////////////////////

/*1: Add a `user_data` to drivers and objects*/
#define LV_USE_USER_DATA        0

#if LV_USE_USER_DATA
/*Declare the type of the user data of object (can be e.g. `void *`, `int`, `struct`)*/
typedef void * lv_obj_user_data_t;
/*Provide a function to free user data*/
#define LV_USE_USER_DATA_FREE 0
#if LV_USE_USER_DATA_FREE
#  define LV_USER_DATA_FREE_INCLUDE  "something.h"  /*Header for user data free function*/
/* Function prototype : void user_data_free(lv_obj_t * obj); */
#  define LV_USER_DATA_FREE  (user_data_free)       /*Invoking for user data free function*/
#endif
#endif

// In lv_obj.c //////////////////////////////////////////////////

#if defined(LV_USER_DATA_FREE_INCLUDE)
#include LV_USER_DATA_FREE_INCLUDE
#endif /* LV_USE_USER_DATA_FREE */

lv_res_t lv_obj_del(lv_obj_t * obj)
{
#if LV_USE_USER_DATA_FREE
    /*Delete the user data*/
    LV_USER_DATA_FREE(obj);
#endif
}

```

I have submitted it as required. Thank you.
Now I can continue to write my code. I'm so happy

puzrin, 2020-02-12:
Can it be useful to have flag in each object, should it free user data or not? For example, one can have common list, binded to 2 views.

IMO choice "always keep" / "always free" is not flexible.

kisvegabor, 2020-02-13:
> Can it be useful to have flag in each object, should it free user data or not? 

The user data can be a structure with other dynamically allocated elements. So a simple auto-free flag can work in all cases. 

> For example, one can have common list, binded to 2 views.

Can you explain it further?

puzrin, 2020-02-13:
If i destroy 2 views (widgets) with common data, each will try to free the same data. First will succeed, second can cause undefined behaviour.

That's only first example i could imagine. I can't explain why, but as architect i feel suggested change too restrictive. It fixes something, but not as much as it could.

ShenRen, 2020-02-13:
> If i destroy 2 views (widgets) with common data, each will try to free the same data. First will succeed, second can cause undefined behaviour.
>
> That's only first example i could imagine. I can't [...]

In this case, your user data needs a reference count. It can ensure that the data will not be accidentally deleted when there are other references.

puzrin, 2020-02-13:
> In this case, your user data needs a reference count. It can ensure that the data will not be accidentally deleted when there are other references.

Or each object should know if it has to manage user data or not.

I think, in terms of OOP we a inventing "object destructor". IMO that's not "static" thing.

kisvegabor, 2020-02-15:
> I think, in terms of OOP we a inventing "object destructor". IMO that's not "static" thing.

As I see it's not that complicated and not for the very object level. It's rather a system-level option that can be used if you use lvgl in a binding (e.g. C++ as mentioned).

While it's easy to implement (although limited to some use cases), a destructor kind of thing would be quite complicated.

closed 2020-02-17 by ShenRen

FHenriques, 2020-12-31:
Why not just a function pointer inside the lv_obj_t... 

___void freefoo(lv_obj_t*){
//free data
}
 lv_obj_add_userdata_freefoo(freefoo);_

Why the hell should I create a file and declare it elsewhere, why should the framework even call some user/application files?

In framework i could be called:
__if(lv_obj->user_free_foo)
   lv_obj->user_free_foo(lv_obj);_

I really like your work, and it rally is a nice framework but sometimes I just think that some things have been unnecessarily complicated just because.

I understand that the library rely heavily in MACROS to reduce footprint, but in this case the function 
 _lv_obj_add_userdata_freefoo(freefoo);_
could be hidden with the #define LV_USE_USER_DATA to.

At this point, having local functions related to each screen and objects in multiple files, I have to add all of the headers to the macro LV_USER_DATA_FREE_INCLUDE. Another disadvantage is that I can't even use private/static functions on .c file scope because I need to put the prototype in the header and add it to LV_USER_DATA_FREE_INCLUDE on lv_conf.h.

kisvegabor, 2021-01-04:
I agree it's not ideal but rather a quick fix to have something.

IMO the proper solution is the free the user data in `LV_EVENT_DELETE`. It needs you to write `event_cb` for every object with custom user data but
- if you already have an `even_cb` for that object it's easy  to add another `if(e == LV_EVENT_DELETE)`
- if you don't have `event_cb` you can add a `my_free_event_cb`.