lvgl/lvgl #1375: Label inside of Page not displayed if text is too much

Maldus512, 2020-02-11:
**Describe the bug**

When using a label inside of a page, text exceeding a certain number of characters makes it display as empty. I'm not sure how to pinpoint the exact location of the problem, but it seems to depend on how much memory I'm currently using (the size of the string needed to achieve the bug varies depending on what other objects are displayed on screen).
As far as I can tell, the allocation of the label and the corresponding text works fine

**To Reproduce**

I'm using the PC simulator and creating a single label with mock text inside of a page for scrolling:

```c
    char test[1000000];

    lv_obj_t *page  = lv_page_create(lv_scr_act(), NULL);
    lv_obj_t *label = lv_label_create(page, NULL);
    lv_obj_set_size(page, LV_HOR_RES, LV_VER_RES);

    // If the limit is 1500 the label is displayed correctly
    for (int i = 0; i < 2000; i++) {
        strcat(test, "Lorem ipsum dolor sit amet, consectetur adipiscing elit. Sed et sagittis diam, in gravida "
                     "dui metus.\n");
    }
    lv_label_set_text(label, test);
```

**Expected behavior**
I would expect the label to correctly display the text regardless of its size, or fail because of allocation problems.

**Screenshots**
The resulting screen is empty:
[image]

**Additional context**
I'm using the PC simulator with `malloc` and `free` as memory managing functions. For reference, this is my `lv_conf.h` file:

[lv_conf.txt](https://github.com/littlevgl/lvgl/files/4184562/lv_conf.txt)

embeddedt, 2020-02-11:
Can you enable logging and see if anything useful gets printed?

Maldus512, 2020-02-11:
Nothing useful it would seem. Whether the label ends up displayed or not, the output is the same:

```
Info: lv_init ready     (components/lvgl/src/lv_core/lv_obj.c #121)
Info: Screen create ready       (components/lvgl/src/lv_core/lv_obj.c #222)
Info: Screen create ready       (components/lvgl/src/lv_core/lv_obj.c #222)
Info: Screen create ready       (components/lvgl/src/lv_core/lv_obj.c #222)
Info: container created         (components/lvgl/src/lv_objx/lv_cont.c #116)
Info: container created         (components/lvgl/src/lv_objx/lv_cont.c #116)
Info: page created      (components/lvgl/src/lv_objx/lv_page.c #170)
Info: label created     (components/lvgl/src/lv_objx/lv_label.c #161)
```
(LV_LOG_LEVEL is info).

embeddedt, 2020-02-11:
Interesting. That means that you aren't running out of memory to allocate.

When I get a chance, I will try and track down the issue.

One thing to try (although this probably isn't the issue): can you call `memset` on the `test` buffer before you use it? At the start of the program you can't guarantee that it's initialized to zero, and it needs to be that way for `strcat` to work properly.

Maldus512, 2020-02-11:
> At the start of the program you can't guarantee that it's initialized to zero, and it needs to be that way for strcat to work properly.
You're correct; unfortunately I've tried and there is no difference.

kisvegabor, 2020-02-11:
What if you change `typedef int16_t lv_coord_t;` to `typedef int32_t lv_coord_t;` in `lv_conf.h`?

2000 lines of text could have ~25px * 2,000 lines = 50,000 px height which overflows the default 32k limit.

Maldus512, 2020-02-12:
> What if you change typedef int16_t lv_coord_t; to typedef int32_t lv_coord_t; in lv_conf.h?
> 2000 lines of text could have ~25px * 2,000 lines = 50,000 px height which overflows the default 32k limit.

Bingo! Now it works. Thanks a lot!

As a clarification, using a bigger type for coordinates should have no side effect besides having to handle bigger data types, right?

embeddedt, 2020-02-12:
> As a clarification, using a bigger type for coordinates should have no side effect besides having to handle bigger data types, right?

Correct. We use `int16_t` by default to save RAM, because small displays with low amounts of text don't need the extra range.

It might be useful to implement an assertion that checked for coordinate overflow in the text drawing code.

Maldus512, 2020-02-12:
> It might be useful to implement an assertion that checked for coordinate overflow in the text drawing code.
I might take a look at it, but no promises. In the meantime, case closed.

closed 2020-02-12 by Maldus512

kisvegabor, 2020-02-13:
FYI, a good place for the assert could be  `lv_txt_get_size()` in `lv_txt.c`.