lvgl/lvgl #1378: DEV-7.0 has broken tileview animation

barbiani, 2020-02-14:
Lastest dev-7.0 tileview is not responding to LV_EVENT_RELEASED/LV_EVENT_DRAG_END.

Tile rests where I leave it instead of moving back to the parent (0, 0) coordinate.

Tileview drag_end_handler() seems to not be called when mouse is released in the visual studio simulator.

rabbitsaviola, 2020-02-14:
Is it because "break" is missing in the [for loop](https://github.com/littlevgl/lvgl/blob/40e2da1905a20437aa66ca439ceb332841808eae/src/lv_objx/lv_tileview.c#L225) of function `lv_tileview_set_tile_act`  ?

xennex22, 2020-02-14:
> 
> 
> Is it because "break" is missing in the [for loop](https://github.com/littlevgl/lvgl/blob/40e2da1905a20437aa66ca439ceb332841808eae/src/lv_objx/lv_tileview.c#L225) of function `lv_tileview_set_tile_act` ?

No. That is non optimal, but not a bug.

rabbitsaviola, 2020-02-14:
> > Is it because "break" is missing in the [for loop](https://github.com/littlevgl/lvgl/blob/40e2da1905a20437aa66ca439ceb332841808eae/src/lv_objx/lv_tileview.c#L225) of function `lv_tileview_set_tile_act` ?
> 
> No. That is non optimal, but not a bug.

maybe it's not relevant to this issue, but without the `break` the `tile_id` passed to `lv_event_send` would be wrong?

embeddedt, 2020-02-14:
@rabbitsaviola Good catch. I don't think that would fix this issue in particular, but it's definitely a bug.

Fixed here: https://github.com/littlevgl/lvgl/commit/36a4307930a858887030e79aecf5c9536c46a6c2

rabbitsaviola, 2020-02-14:
It seems `set_drag_throw` should be set to true for tileview animation?
https://github.com/littlevgl/lvgl/blob/9d664b27ec9ce9e502755a577788a930689bf13d/src/lv_objx/lv_tileview.c#L116

barbiani, 2020-02-14:
Tileview was working fine last week.

xennex22, 2020-02-14:
> maybe it's not relevant to this issue, but without the `break` the `tile_id` passed to `lv_event_send` would be wrong?

Yes, sorry, I missed that.

kisvegabor, 2020-02-14:
Hi,

I've just fixed it. Somehow these [two lines](https://github.com/littlevgl/lvgl/commit/c00750d19f36f1df416712092e049944b23aac6d) have disappeared. (Probably during merging when solving conflicts)

barbiani, 2020-02-14:
I can confirm that drag end restores the tile correctly.

closed 2020-02-14 by barbiani