lvgl/lvgl #1379: angle mask is not removed after drawing

rabbitsaviola, 2020-02-14:
**Describe the bug**

Wrong mask parameters present in mask_list causes exception.

**To Reproduce**

create arc object with start_angle=0, and end_angle=360

**Expected behavior**

No exception happens

**Screenshots**
`lv_draw_mask_remove_id(mask_angle_id);`
should be added before return at following position?

https://github.com/littlevgl/lvgl/blob/9d664b27ec9ce9e502755a577788a930689bf13d/src/lv_draw/lv_draw_arc.c#L97

**Additional context**

Add any other context about the problem here.

rabbitsaviola, 2020-02-14:
BTW, if start_angle=0, end_angle=360，the full ring has a gap as the screenshot shows.
[image]

kisvegabor, 2020-02-15:
Thank you for pointing this out!

I've just merged a [huge update](https://github.com/littlevgl/lvgl/issues/1243#issuecomment-586579395) into `dev-7.0` and it was already [fixed](https://github.com/littlevgl/lvgl/blob/332124f1f43136a362e55de8f7e02387c65906da/src/lv_draw/lv_draw_arc.c#L93-L101) in it.

closed 2020-02-15 by kisvegabor