lvgl/lvgl #1380: Message box displays properly in simulator but not on target

GnUfTw, 2020-02-14:
**Description**

I am developing a LVGL application that uses message boxes as confirmation dialogs and error dialogs. One of the confirmation dialogs is pictured below:

[image]

When the user selects the button labeled 'Yes' in the message box pictured above, the application attempts to mount a USB drive. If this mount operation fails, I show another message box communicating the error to the user. This second message box is pictured below:

[image]

A snippet of the handler for the _first_ dialog presented to the user is shown below. You can see that we are creating the 2nd message box from the 1st message box handler.
```c
static lv_res_t sys_conf_dialog_action(lv_obj_t* mbox, const char* lbl)
{
    lv_obj_del(lv_obj_get_parent(mbox));
    if (strcmp("No", lbl) == 0)
    {
        // User canceled update process, hide modal and do nothing.
        lv_obj_set_hidden(modalContainer, true);
        return LV_RES_OK;
    }

    // Attempt to mount USB
    if (!mfr_swu_init())
    {
        // Mounting the USB failed.
        lv_obj_set_hidden(modalContainer, false);

        lv_obj_t* errorDialog = lv_mbox_create(modalContainer, NULL);
        lv_mbox_set_text(errorDialog, "Mounting USB failed");

        static const char* btns[] = {"\221Ok", ""};
        lv_mbox_add_btns(errorDialog, btns, NULL);
        lv_obj_set_width(errorDialog, 250);
        lv_obj_align(errorDialog, NULL, LV_ALIGN_CENTER, 0, 0);
        lv_mbox_set_action(errorDialog, sys_err_dialog_action);

        return LV_RES_OK;
    }

    /* ... Snip ... */
}
```

This all works excellent and exactly as I would expect but _only when running against the simulator_. When running on the target (embedded linux arm SOM) and using real hardware, the 2nd message box is displayed in an unexpected fashion. The unexpected result is pictured below:

[image]

What we see here is strange, the message boxes button matrix is being displayed incorrectly. We can briefly see the **Ok** label shown below the **Mounting USB failed** text but there is no button matrix to be found anywhere.

**Steps To Reproduce**

Unfortunately, I don't think I can construct steps to reproduce this issue since I am only seeing the issue on target hardware and not in the simulator.

**Expected behavior**

I would expect the **Mounting USB failed** message box to display the same on the target hardware as it does in the simulator. It is not showing properly on target hardware.

This is the first time I have observed differing behavior between the simulator and target hardware. Any ideas what might be going on here?

**Screenshots**
See above.

**Additional context**

- LVGL version: 5.2.0
- Works fine in simulator under linux and windows
- Happy valentines day

GnUfTw, 2020-02-14:
I have found a solution to the problem. Motivation for the fix came from the following snippet in the documentation:
```c
static lv_res_t mbox_apply_action(lv_obj_t* mbox, const char* txt)
{
    printf("Mbox button: %s\n", txt);
    return LV_RES_OK;    /* Return OK if the message box is not deleted */
}
```
The solution is to change all `return LV_RES_OK;` statements to `return LV_RES_INV;`

closed 2020-02-14 by GnUfTw

embeddedt, 2020-02-14:
For the record, this is no longer an issue in 6.0, which automatically tracks object deletion.